lumapu/ahoy #332: Feature Request : "Receive fail" verbessern

ntfrnd, 2022-10-06:
Aktuell ist es wohl so, dass der "receive fail"-Zähler bei der Statistics-Anzeige hoch zählt, wenn die Antwort nicht in Ordnung ist. Dabei spielt es offenbar keine Rolle, ob die Antwort fehlerhaft ist oder gar keine Antwort kommt (z.B. nach Sonnenuntergang). Da steigt die "receive fail"-Anzeige enorm schnell an.
Daher ist diese Info m.E. wenig hilfreich.

Besser wäre es doch, den "receive fail"-Zähler tatsächlich nur für fehlerhafte Antworten zu verwenden und zusätzlich einen Zähler für "receive no answer" einzuführen. Der würde dann über Nacht oder wenn sich der Inverter außerhalb der Reichweite befindet, hochzählen.

Im Gegensatz dazu könnte man  den "receive fail" nach einigen Tagen prüfen, ob die Kommunikation sauber verläuft.

Ich würde mich freuen, wenn mein Wunsch umgesetzt werden würde.

homeautomation2022, 2022-10-06:
In der Nacht kein Funkverkehr ist bereits in Arbeit. Die Funktion dazu habe ich schon fertig und im Discord als Patch geteilt, fehlt nur noch jemand der Lust auf das Programmieren des Webinterfaces und EEPROM hat, für die Koordinaten und zum deaktivieren der Funktion für Akkubetrieb.

lumapu, 2022-10-06:
trotzdem wird es Überschneidungen geben und der receive fail hochzählen.
Daher finde ich die Idee von @ntfrnd zwei separate Zähler zu haben sehr gut.

homeautomation2022, 2022-10-07:
Das stimmt, ganz perfekt wird man Sonnenauf- und untergang nicht treffen, dennoch muss er Nachts nicht permanent versuchen den HM zu erreichen. Du meinst also einen CRC fehlerhaft Zähler und nicht die Retransmitts?

ntfrnd, 2022-10-07:
Nur zwischen Sonnenauf- und -untergang zu senden ist auch ne gute Idee.
Darüber hinaus die Antworten aufteilen in 1) keine Antwort und 2) Antwort mit CRC-Error wäre doch perfekt, oder?

deep470, 2022-10-07:
Ahoy zeigt ja schon den Zeitpunkt der letzten erfolgreichen Verbindung an. Den keine Antwort-Zähler könnte man automatisch leeren wenn wieder eine erfolgreiche Verbindung stattgefunden hat. Oder nach 5 Minuten stabiler Verbindung.

homeautomation2022, 2022-10-07:
Ich versuche gerade herauszufinden, was Nachts überhaupt passiert und habe einfach mal eine falsche SN konfiguriert:
Tx kommt, Rx nichts aber es zählt auch wie gewollt kein Rx failed hoch. Vielleicht muss aber auch einmal die Verbindung da gewesen sein???

ntfrnd, 2022-10-07:
Aktuell sieht es so aus:
[image]

Also bei mir zählt der receive-fail-Zähler munter hoch, wenn die Sonne weg ist.
Ich gebe dir Recht, prinzipiell kann er das zählen. Aber für mich macht es einen Unterschied ob er nicht oder fehlerhaft geantwortet hat. Deshalb der Wunsch nach zwei Zählern.

closed 2022-10-11 by lumapu