lumapu/ahoy #334: No reception of data

mkoxyz, 2022-10-08:
After a very long frustrating time to find the right setup of both chips I managed to make a stable connection to my inverter. 
Well now I face the issue that the status page says my inverter ist available - but not producing. Well this is not correct as both my Shelly AND S-Miles app say that currently around 30W are "flowing". 
Has anyone an idea what is wrong with my setup?
I'm using HM-600 and the 0.5.17 precompiled bin image.

DanielR92, 2022-10-08:
Please tell us how you connected the components together. Which inverter (with SNr.) is used, how the settings are set in the setup area.

Only if we know all the information can we see through this error more precisely.

mkoxyz, 2022-10-08:
> Please tell us how you connected the components together. Which inverter (with SNr.) is used, how the settings are set in the setup area.
> 
> 
> 
> Only if we know all the information can we see through this error more precisely.

The connection between the components is exactly as described in the readme section. The only settings I changed from default  is the sn of my HM-600 and it's name. I'm not sure how the sn can help to solve the problem but it is exactly the same as on the label and in the native app from Hoymiles.

stefan123t, 2022-10-08:
Do you consider to turn off the S-Miles DTU lite as they may collide when both are talking to the inverter. Which nRf24 modules (with/without external antenna) are you using and at which Power Level? How long is the distance and which obstacles are between the AhoyDTU and the Inverter.
We need the first 4+1+2=7 digits of your Inverter to determine the model and build period. The remaining 5 digits are the running serial of your inverter and you can keep them for yourself.
Do you habe serial Log enabled and what. An you see via USB Serial port ?

mkoxyz, 2022-10-08:
> Do you consider to turn off the S-Miles DTU lite as they may collide when both are talking to the inverter. Which nRf24 modules (with/without external antenna) are you using and at which Power [...]

This was also my first idea to switch off the original DTU - so I did it today in the morning no change.
[image]
As you can see on the picture no data has been received. 
The serial number starts with the 1141810..... and this ist the only variable aside of the name I changed.
[image]
The amplifier level ist set by default to high and I did not change it. 
The HW is at the same place as the original DTU and this is like approx. 5 meters to the inverter and there is just one wall between. The NRF is without the external antenna.

Archibald-Harry-Tuttle, 2022-10-08:
Not sure if max power must be filled in the setup to get the inverter functional at all?
0 means in my opinion you have no power at all on your DC inverter entry.
My panels have 460wp and i put these values in.
In operation the measured wp power by inverter and the max exspected wp is compared and shown as percentage of max.100% is showing the momentary illumination on each panel.

mkoxyz, 2022-10-08:
> Not sure if max module power must be filled in the setup to get the inverter functional at all?
> 
> 0 means in my opinion you have no power at all on your DC inverter entry.
> 
> My panels have 460wp and i put these values in.
> 
> In operation the measured wp power by inverter and the max exspected wp is compared and shown as percentage of max.100% is showing the momentary illumination on each panel.

OK is a possible solution. Well I changed the value but I assume that now it will not work at all as the sun is down.
What is the usual display after sunset?

mkoxyz, 2022-10-09:
> OK is a possible solution. Well I changed the value but I assume that now it will not work at all as the sun is down.
> What is the usual display after sunset?

Well sun is out now and there is no change - ahoy says that the inverter is available but not producing. 

Any other ideas?

fila612, 2022-10-09:
hmmm, maybe you got no data from your inverter because you have no time. in the screenshot there is the current time not available. can you check your NTP settings?

mkoxyz, 2022-10-09:
> hmmm, maybe you got no data from your inverter because you have no time. in the screenshot there is the current time not available. can you check your NTP settings?

I think this is not the problem. The reason why there is no time is just that there is no internet connection as I did not add the HW into my WLAN. In the past I did it and the current time was shown but the issue was the same.

fila612, 2022-10-09:
So, of the Packages have no valid timestamp, you trink its not an Problem to interact with the inverter?
You asks for other ideas and don‘t give them a try….

stefan123t, 2022-10-10:
@mkoxyz 
The AhoyDTU will not query the inverter as long as it has not acquired the current NTP time via your network. So setting up your SSID and WiFi Pass is mandatory.
You may want to check the AhoyDTU serial console when connected to your PC via USB in order to see what it actually does. There is a checkbox on the setup screen too to enable serial logging.
From dusk till dawn your inverter will be unreachable and currently the failed receive count should rise quickly to some thousand per night. As yours is still 0 this means AhoyDTU does not query the inverter because it has no NTP time set.

mkoxyz, 2022-10-11:
> @mkoxyz
> The AhoyDTU will not query the inverter as long as it has not acquired the current NTP time via your network. So setting up your SSID and WiFi Pass is mandatory.
> You may want to check the [...]

Thank you very much for the clarification.
I just set up the AhoyDTU to connect to my network. NTP time is also set now.
Let's see what happens when the sun rises... 
[image]

stefan123t, 2022-10-11:
@lumapu @DanielR92 we need a warning on the Status / Statistics in case the NTP time has not been set.
Just giving out `now: n/a` is not enough IMHO. This should tell something like: 
`NTP timeserver unreachable. Please connect to WiFi in order to receive proper data from your inverter.`

mkoxyz, 2022-10-12:
Well thank you so much for all your help. Now it works quite stable.
Lessons learned:
- WIFI setup is mandatory and not at option as NTP must be reachable
- Setting the power of panel(s) is also mandatory otherwise you do not get valid data

commit 38bb3fd7 (2022-10-16): fix #334 added button to index if ntp timeserver was not reachable
closed 2022-10-16 by lumapu