lumapu/ahoy #335: [Git SHA: g67f99ab :: 0.5.18] Exception nach Update, ständig reboots

fila612, 2022-10-08:
Hi,

ich habe mit der neuesten Dev-Version auf meinem Testdevice dauernd Execptions und reboots.
Mit der 0.5.17. lief es stabil.

hier der Auszug aus der Konsole.

16:55:30.914 > I: Welcome to AHOY!
16:55:30.921 > I: 
16:55:30.921 > point your browser to http://ip-ip-ip-ip
16:55:30.921 > I: to configure your device
16:55:30.923 > I: ----------------------------------------
16:55:30.930 > 
16:55:30.942 > I: RF24 Amp Pwr: RF24_PA_I: HIGH
16:55:30.942 > I: Radio Config:
16:55:30.942 > SPI Frequency            = 1 Mhz
16:55:30.942 > Channel                  = 3 (~ 2403 MHz)
16:55:30.942 > Model                    = nRF24L01+
16:55:30.942 > RF Data Rate             = 250 KBPS
16:55:30.942 > RF Power Amplifier       = PA_HIGH
16:55:30.944 > RF Low Noise Amplifier   = Enabled
16:55:30.947 > CRC Length               = 16 bits
16:55:30.950 > Address Length           = 5 bytes
16:55:30.952 > Static Payload Length    = 32 bytes
16:55:30.955 > Auto Retry Delay = 250 microseconds
16:55:30.958 > Auto Retry Attempts      = 0 maximum
16:55:30.961 > Packets lost on
16:55:30.963 >     current channel      = 0
16:55:30.963 > Retry attempts made for
16:55:30.966 >     last transmission    = 15
16:55:30.969 > Multicast                = Disabled
16:55:30.972 > Custom ACK Payload       = Disabled
16:55:30.974 > Dynamic Payloads = Enabled
16:55:30.974 > Auto Acknowledgment      = Disabled
16:55:30.977 > Primary Mode             = RX
16:55:30.980 > TX address               = 0xdeadbeef01
16:55:30.983 > pipe 0 (closed) bound    = 0xdeadbeef01
16:55:30.986 > pipe 1 ( open ) bound    = 0x9825768701
16:55:30.989 > pipe 2 (closed) bound    = 0xc3
16:55:30.992 > pipe 3 (closed) bound    = 0xc4
16:55:30.994 > pipe 4 (closed) bound    = 0xc5
16:55:30.997 > pipe 5 (closed) bound    = 0xc6
16:55:37.032 > I: [NTP]: 2022-10-08 16:55:31
16:55:37.032 > I: 
16:55:37.034 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
16:55:37.040 > 
16:55:37.040 > Exception (28):
16:55:37.040 > epc1=0x4020247d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
16:55:37.049 > 
16:55:37.049 > >>>stack>>>
16:55:37.049 > 
16:55:37.049 > ctx: cont
16:55:37.054 > sp: 3ffffd00 end: 3fffffc0 offset: 0190
16:55:37.059 > 3ffffe90:  000032e2 00000000 3ffef220 4020247d  
16:55:37.060 > 3ffffea0:  00000000 3fff1b70 00000000 000007e6  
16:55:37.063 > 3ffffeb0:  3ffef920 000003e8 3ffef8e4 40201dc0  
16:55:37.069 > 3ffffec0:  00000000 3ffef320 3ffef220 40207163  
16:55:37.072 > 3ffffed0:  00000037 0000001f 00000020 40227da4  
16:55:37.077 > 3ffffee0:  00000000 001a001f 00000000 35353a36  
16:55:37.080 > 3ffffef0:  0031333a 00000000 3fff35cc 4021e0dc  
16:55:37.086 > 3fffff00:  00000000 001a001f 00000000 40209395  
16:55:37.089 > 3fffff10:  00000000 00000000 00000000 00000000  
16:55:37.094 > 3fffff20:  3fff4604 00000000 40207274 40207430  
16:55:37.100 > 3fffff30:  3fff4744 3ffef2c4 3fff35cc 40100c3c  
16:55:37.103 > 3fffff40:  d50a0a0a feefeffe 0000082c 0000001e  
16:55:37.109 > 3fffff50:  3fff35cc 3ffef2c4 3ffef220 3ffefb30  
16:55:37.112 > 3fffff60:  4021baf8 00000000 00000000 00000037  
16:55:37.118 > 3fffff70:  3fff1b54 00000001 feefeffe 00000000  
16:55:37.121 > 3fffff80:  3fffdad0 00000000 3ffef220 3ffefb30  
16:55:37.126 > 3fffff90:  3fffdad0 00000000 3ffefb1c 40207234  
16:55:37.132 > 3fffffa0:  feefeffe feefeffe 3ffefb1c 4021ae48  
16:55:37.135 > 3fffffb0:  feefeffe feefeffe 3ffe85d8 40100f99  
16:55:37.141 > <<<stack<<<
16:55:37.141 > 
16:55:37.141 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
16:55:37.158 > 
16:55:37.159 >  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
16:55:37.161 > 
16:55:37.161 > load 0x4010f000, len 3460, room 16 
16:55:37.168 > tail 4
16:55:37.168 > chksum 0xcc
16:55:37.168 > load 0x3fff20b8, len 40, room 4 
16:55:37.172 > tail 4
16:55:37.172 > chksum 0xc9
16:55:37.174 > csum 0xc9
16:55:37.174 > v000658d0
16:55:37.177 > ~ld
16:55:37.241 > I: resetPayload: id: 0
16:55:37.241 > I: resetPayload: id: 0
16:55:37.244 > I: resetPayload: id: 0
16:55:37.246 > I: resetPayload: id: 0

burn79, 2022-10-09:
bei mir genau dasselbe. Nach dem Holen der aktuellen Zeit vom NTP-Server erfolgt sofort der Reboot.
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007f760
~ld

SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
D: start: 0, length: 113
D: CRC RD: f448 CRC CALC: f448
D: start: 115, length: 562
D: CRC RD: 2931 CRC CALC: 2931
I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
fpm close 1
mode : sta(ac:0b:fb:dc:e3:d3)
add if0
wifi evt: 8
wifi evt: 2
I: connect to network 'heimnetz-huehn' ...
........scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

connected with xxxxxxxxxxxxx, channel 6
dhcp client start...
wifi evt: 0
.ip:192.168.10.197,mask:255.255.255.0,gw:192.168.10.3
wifi evt: 3
.
.
I:

----------------------------------------
I: Welcome to AHOY!
I:
point your browser to http://192.168.10.197
I: to configure your device
I: ----------------------------------------

I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency           = 1 Mhz
Channel                 = 255 (~ 2655 MHz)
Model                   = nRF24L01+
RF Data Rate            = 1 MBPS
RF Power Amplifier      = PA_MAX
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 4000 microseconds
Auto Retry Attempts     = 15 maximum
Packets lost on
    current channel     = 15
Retry attempts made for
    last transmission   = 15
[... 162 lines cut ...]
3ffffe70:  4020f4e4 00000000 4020f4e4 00000000
3ffffe80:  4020f4e4 00000000 3fff3ad4 3fff3234
3ffffe90:  4026f02c 4020eec0 40233a9c 4020f550
3ffffea0:  00000000 4020f550 00000000 4020f550
3ffffeb0:  00000000 3fff3ad4 3fff3234 40100a64
3ffffec0:  4020eec0 40233a9c 4020f288 00000000
3ffffed0:  4020f288 00000000 4020f288 00000000
3ffffee0:  3fff3ad4 3fff3234 00000024 00000000
3ffffef0:  001a001f 00000000 00000000 001a001f
3fffff00:  00000000 00000000 00000000 00000000
3fffff10:  32323032 2d30312d 32203830 37333a32
3fffff20:  0031323a 00000000 4020f3b8 00000000
3fffff30:  4020f3b8 00000000 3fff3ad4 3fff3234
3fffff40:  3ffef6f3 4020eec0 40233a9c 4020f450
3fffff50:  40230c58 00000002 00000000 40230bfa
3fffff60:  00000000 0000000a 00000008 00000016
3fffff70:  4020eec0 40233a9c 3fff4fb4 3ffefec8
3fffff80:  3fffdad0 00000000 3ffef5b0 3ffefec8
3fffff90:  3fffdad0 00000000 3ffe85d8 4020ed38
3fffffa0:  feefeffe feefeffe 3ffe85d8 4022fdaa
3fffffb0:  feefeffe feefeffe feefeffe 40101a71
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

stefan123t, 2022-10-10:
Thanks for providing the debug output. Please note that exception stack trace can only be decoded on your VScode using the current firmware bin file. Please search the net for esp8266/32 exception decoder and vscode.
Usually this tells you the stack when the issue occurred and several libraries on the stack.
Is your problem on ESP8266 or ESP32 ?

Also did you try to build the normal release because @burn79 definetly runs the Debug build which spits out a lot more devug information also for core libraries as the ESP WiFi stack etc. it usually is better to run the release build.

fila612, 2022-10-11:
thanks Stefan,

i had these issue on a ES8266 and runs on a normal release - not an a debug build.

burn79, 2022-10-11:
@stefan123t the problem is on ESP8266. Not only with my own builded bin-file, also with the bin's builded by the git actions.
Today I copiled the release-build. Same problem whith this build. Direct after getting time from NTP server the ESP restarts with exception.
The Exception Decoder outputs this:
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40202e85
EXCVADDR: 0x00000000

Decoding stack results
0x40226f45: _fclose_r at /workdir/repo/newlib/newlib/libc/stdio/fclose.c line 67
0x40226f45: _fclose_r at /workdir/repo/newlib/newlib/libc/stdio/fclose.c line 67
0x40230364: icmp_input at core/ipv4/icmp.c line 239
0x40230306: icmp_input at core/ipv4/icmp.c line 215
0x4022f4c2: ip4_input at core/ipv4/ip4.c line 1262
```

burn79, 2022-10-11:
Now I tried Git SHA: gc2b5555 commited an hour ago, and the exception is gone.
The ESP8266 build now works. Both, the self compiled and the bin file from git actions.

stefan123t, 2022-10-11:
@fila612 i noticed the following line in your original output:
```
16:55:30.921 > point your browser to http://ip-ip-ip-ip/
```
@lumapu @DanielR92 do you know what could have caused this ?

@burn79
though you are using the build for the same platform I can also see the ip4.c exceptions from core.
This would hint at some issue with the core libraries in the build you used back then.
But now you mentioned that the latest commit c2b5555 fixes this. So we can probably disregard the back issue.

@fila612 could you retry with the c2b5555 or later version to reproduce your original problem ?

@lumapu @DanielR92 it would be good to have the build hash number also in the preamble of the serial output in order to pin point and attribute issues more easily. Similar to what the ESP core libraries tell us:
```
SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635
...
----------------------------------------
I: Welcome to AHOY!
I: point your browser to http://192.168.10.197
I: to configure your device
I: ----------------------------------------
I: v0.5.18 http://github.com/lumapu/ahoy/commit/c2b5555
```

fila612, 2022-10-12:
Hi @stefan123t 
the part with "ip-ip-ip" was manually edited by myself - sorry that I forgot to point that out.
normally there is the correct ip address written.

I tried to reproduce the behaviour with the latest dev build (0.5.19 g6bf6e35) but currently its not reproducible.
seems to be good for the build - bad for analysis :(
i will try some more tests, if its not reproducible we should close this. 
maybe @burn79  can check on his environment

burn79, 2022-10-12:
I just did the OTA update to the current 0.5.19 [gdbdc2ad](https://github.com/lumapu/ahoy/commit/dbdc2ad5ab446e25b2f427f4a9323322391e6b66). Looks good to me.

stefan123t, 2022-10-13:
confirmed working by @fila612 and @burn79
issue will be closed as cannot reproduce

closed 2022-10-13 by stefan123t