lumapu/ahoy #336: Feature Request: Manuelle Zeit einstellen

fila612, 2022-10-09:
Hallo,

ich hatte letztens einen längeren Internetausfall (über 24h). Leider habe ich so erst gemerkt, dass ohne aktuelle Zeit ahoy keine Daten abrufen kann.
Da ich keine Möglichkeit gesehen habe die Zeit auch manuell zu setzen, wäre meine Frage ob dies noch in zukünftigen Releases eine Variante sein kann.
Sollte aus meiner Sicht im Setup neben dem NTP nur die Möglichkeit sein die Zeit manuell zu setzen.

DanielR92, 2022-10-09:
Ja könnte man mal einkippen. Die Zeit könnte man aber zur Not via Fritbox bekommen, oder einmalig beim Setup über den PC wenn man darauf zugreift. 

var today = new Date();
var time = today.getHours() + ":" + today.getMinutes() + ":" + today.getSeconds();

Wäre eine möglichkeit... dies müsste jedoch einmalig passieren.

fila612, 2022-10-09:
Bzgl Zeit vom Router: dachte ich auch, aber ohne netzt ist der auch ohne Zeit :(

stefan123t, 2022-10-09:
Man könnte auch den Zeitstempel vom Client-Browser übernehmen, der wird ja in der Regel im Request-Header übergeben.
Ich glaube Thomas macht das in OpenDTU auch so mit dem SYNC Button oder wird da immer der NTP request getriggert ?
Sonst könnte man das ja als Fallback machen:
1. NTP versuchen.
2. Zeit aus dem HTTP Request-Header vom Client übernehmen.

3. Alternativ ein paar Input Felder für Datum, Uhrzeit und ggf. Zeitzone damit man es ggf. wirklich manuell eintragen kann.
@DanielR92 ich habe keine Ahnung wie man das macht, daß es auf dem Mobiltelefon sinnvoll auswählbar ist ?

lumapu, 2022-10-11:
@fila612 bitte mit dem neuesten Dev-Build nochmal probieren, ich habe zwei Buttons eingebaut.
**Achtung** das EEprom Layout hat sich geändert, daher gehen bis auf WiFi alle Settings verloren.
Wenn alles passt den Issue bitte closen.

fila612, 2022-10-12:
Danke dir @lumapu ! :)
Habs grad aktualisiert und sieht gut aus, ich mach hier zu, sollte es zu Fehlern kommen, mach ich ein neues Issue auf.

stefan123t, 2022-11-18:
Es gab Abfragen wie man AhoyDTU Off-Grid bzw Off-Network betreibt ? Hier wäre es sinnvoll dass der WR den Zeitstempel zumindest bei jedem speichern des Setups mit in die JSON Config schreibt. Noch besser wäre natürlich die Anbindung von nem RTC Modul ?

closed 2023-02-07 by lumapu

DB1BMN, 2023-04-29:
Hi, sorry fürs Kapern... möchte es nur verstehen.

Für was genau benötigt es denn überhaupt die exakte Zeit/Datum?
OK, Sonnenauf- und Untergang, aber sonst?

Da es ja aktuell schon eine gute Unterstützung für die Displays gibt, kommt auch immer mehr ein Stand-Allone-Betrieb in Betracht, d.h. ohne externes WLAN/Internet.

Ergo: Modul per USB-Versorgen, per eigenem Access-Point die Seriennummer des Wechselrichters konfigurieren, Display einrichten, evtl. Default-Passwort ändern und so stehen lassen und sich an den Display-Daten erfreuen.

OK, falls der WR Datum/Uhrzeit braucht, damit er weiß, wann "today" ist, dann müsste man das evtl. manuell einstellen können. Den Button "Sync from Browser" gibts ja schon.
Bei Stromausfall müsste  man dann doch wieder noch eine (batteriegepufferte) RTC haben...

Beste Grüße, Mar(e)k

lumapu, 2023-05-03:
Man braucht es, für die Alarmmessages des Wechselrichters. Diese kommen mit der Zeit zurück, zu der sie aufgetreten sind. Bisher ist es nicht bekannt was mit dem YieldDay / YieldTotal passiert, wenn eine falsche Zeit verwendet wird.

DB1BMN, 2023-05-10:
Danke für die Info!
Jetzt habe ich es verstanden.
Habe auch in der seriellen Konsole gesehen, dass beim Senden ganz viele Nullen gesendet werden, solange die Uhrzeit nicht gesetzt und erst dann durch irgendwelche Hex-Werte ersetzt werden.
Damit komme ich erst mal klar ;-)