lumapu/ahoy #337: Bugs in Setup Seite (development Branch)

homeautomation2022, 2022-10-09:
1. Module Name auf 4 Zeichen begrenzt
2. Max Module Power (Wp) und Modulname Eingabefeld nach erstmaligen Speichern bzw. anlegen des Inv verschwunden(zumindest wenn man die Felder leer lässt, erst nach Reboot wieder Eingabefelder sichtbar, zum umkonfigurieren
3. nach Erase All sind auch die Standardkonfigs wie NTP und MQTT leer und Pinout (Wemos) und Radio (NRF24L01+) sind ohne Einagbefeld -> nach reboot behoben

homeautomation2022, 2022-10-11:
Wäre es so, dann hätte ich wohl den Bug nicht beobachten können? Bei Factory Reset wird ein Reboot gemacht, Erase All anscheinend nicht oder ist das bei dir so?, ich musste manuell rebooten, um dem Problem aus dem Weg zu gehen.

lumapu, 2022-10-11:
ok, ich habe es nur in Code heute verfolgt und nicht getestet, danke für den Hinweis dass nicht automatisch rebootet wird (das hatte ich erwartet).

DanielR92, 2022-10-11:
Habe es gerade mit diesem PR https://github.com/lumapu/ahoy/pull/341 getestet. 
Bei mir sieht alles normal aus, Felder sind vor ausgefüllt.

homeautomation2022, 2022-10-11:
OK, ich muss mich korrigieren, keine Ahnung warum es bei meinem Test nicht ging, aber habe nun nochmal mit einer anderen FW gestestet und er macht tatsächlich bei "ERASE SETTINGS (not WiFi)" ein reboot, somit alles i.O.

DanielR92, 2022-10-12:
Kurze Frage, geht es hier "Module Name auf 4 Zeichen begrenzt" darum das die PV-Module mit dem Namen gekürzt werden sollen? Wenn ja, wieso 4 Zeichen? Mit 8 wäre ich noch Einverstanden.

lumapu, 2022-10-12:
@DanielR92 ist doch schon gelöst. Es ging darum, dass die Input Felder nur 4 statt der 16 erlaubten Zeichen zugelassen haben

closed 2022-10-12 by lumapu