lumapu/ahoy #338: mqtt homeassistant Ordner

DennisXK, 2022-10-09:
Ich bekomme in meinem IOBroker mqtt Ordnerstruktur einen Ordner „homeassistant“ angelegt. Da ich ja den IOBroker nutze brauch ich keine mqtt daten für homeassistant! 

Kann man in dem nächsten Update evtl. das mit einer checkbox aktiveren/deaktivieren?!

stefan123t, 2022-10-10:
Es wäre vermutlich sogar noch besser die homeassistant Daten nur auf Knopfdruck im setup einmalig zu verschicken. Das muss mE nicht bei jedem Start der AhoyDTU verschickt werden ?

CBRocker, 2022-10-10:
@stefan123t 
Hey Stefan, wozu sind die homeassistant Daten?

fila612, 2022-10-11:
Der "Ordner" homeassistant dient der Autodiscover Funktion, damit ist der WR automatisch bzw. ohne viel Aufwand in Home  Assistant eingebunden.
Aktuell funktioniert das weitestgehend ganz gut und sollte m.M.n. nicht eingeschränkt werden, aber ich bin komplett bei Nutzern, welche gar kein HA nutzen, dass man das Autodiscover-Feature im Setup Togglebar macht.

stefan123t, 2022-10-11:
@fila612 ich benutze kein homeassistant aber wenn ich es richtig verstehe wird durch das autodiscover feature bei jedem reboot die discovery information an homeassistant geschickt. Warum ist das sinnvoll & notwendig ? Es müsste doch eigentlich ausreichend sein dies einmalig auf Knopfdruck aus dem /setup/ zu versenden. Wenn ich es durch Neustart des homeassistant oder löschen aller Konfiguration dort nochmal brauche, dann kann ich ja wieder auf den Knopf drücken.
Von mir aus zwar auch einmal wenn ich das MQTT setup ändere und speichere aber warum nur beim Neustart ?

lumapu, 2022-10-13:
@stefan123t es gab bisher keine einfache Möglichkeit irgendwas im System im Betrieb zu triggern, das klappt durch die Rest API jetzt sehr gut.
Bin auch für einen Button in Setup

DennisXK, 2022-10-16:
egal wie, Hauptsache es wird kein Homeassistant Ordner angelegt, wenn man ihn nicht benötigt!

lumapu, 2022-10-16:
Bitte nochmal testen ob durch den Button wirklich der `homeassistant` Ordner angelegt wird oder nicht.

commit a595950d (2022-10-16): fix #338, needs verifcation from homeassistant users
added download link for settings on setup.html and update.html
closed 2022-10-21 by lumapu