lumapu/ahoy #339: NRF24L01+ mit oder ohne Antenne?

VIEWSION, 2022-10-10:
Ich hab aktuell die Version ohne Antenne dran - vorsichtshalber aber noch eine Version mit Antenne bestellt. Ich bin direkt unter dem Dach unter der Anlage und die Verbindung scheint eigentlich recht stabil zu sein... "Receive fail" ist niedrig oder 0. Kann man sowas wie eine "Signalstärke" auslesen um das mal zu vergleichen? Wie ist Euere Erfahrung mit/ohne Antenne was die Reichweite angeht?

DanielR92, 2022-10-10:
Wenn du direkt am WR dran bist, reicht der kleine. :) 
Im Setup Bereich ist es hierbei Sinnvoll die Leistung so gering wie möglich zu halten (Low). 

Wenn du bis hierhin super Daten erhälst, wieso etwas daran ändern?

VIEWSION, 2022-10-10:
... ja - stimmt schon ;) Danke!

stefan123t, 2022-10-10:
Nein RSSI der nRF24 Module gibt es nicht. Lediglich die Information ob die Empfangsstärke besser/schlechter als 60/64dB ist.
Es gäbe auch noch ein GetLossRate DevInfo Kommando aber das haben wir mW noch nicht inplementiert. Damit könnte man die Paket-Fehlerrate des Wechselrichters auslesen. Du kannst das gerne implementieren wenn Du Zeit und Lust hast ?

VIEWSION, 2022-10-11:
gerne... dafür muss ich mir das aber erst mal genauer ansehen ;) Hatte bisher noch gar nix mit Arduino & Co am Hut und hier auch nur das fertige Image gezogen. Aber es interessiert mich.... mal sehen...

stefan123t, 2022-10-11:
@VIEWSION herzlich willkommen, das Kommando ist hier beispielhaft beschrieben:
https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#getlossrate--0x15

closed 2022-10-21 by lumapu