lumapu/ahoy #340: ESP-WROOM-32 , Meldung im Setup

mroenne2022, 2022-10-10:
Hallo, hab einen ESP32 heute mit 0.5.17 geflashed.
Konnte auch alles konfigurieren und ihn in mein WLAN integriert.
Ist auch erreichbar , allerdings meckert er im Setup wie folgt : 
WARNING! your NRF24 module can't be reached, check the wiring and pinout.
Hab einen mit Antenne und auch einen ohne Antenne versucht, Meldung bleibt.
Kabel hab ich mehrfach geprüft. Nun ist mein ESP32 einer mit 30 Pins und es gibt 
ja dieses Schema mit der Umsetzung der Pins von 38 auf 30 .
Meines Erachtens alles korrekt , hat jemand vielleicht eine Idee ?
Ob den WR erreiche kann ich erst morgen testen wenn er arbeitet.
Bin noch recht neu hier muss ich dazu sagen.
VG Michael R.

stefan123t, 2022-10-10:
Also ich habe bei mir in Anlehnung an die OpenDTU folgende Belegung am Lolin NodeMCU ESP32 board:

```c
[env:lolin32 lite]
board = lolin32_lite
build_flags = ${env.build_flags}
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
```

Die MISO, MOSI und SCLK sind mW mehr oder minder fest definiert. Die anderen drei IRQ, CE und CS kann man mE auch beliebig anders definieren / belegen.
Es gab auch schon nRF24 Boards bei denen sie Beschriftung der Pins (spiegel?) verkehrt war. Am Besten an dem auf der Oberseite durch ein kleines Quadrat eingekreisten Pin1 GND orientieren. Daneben ist Pin2 3.3V Vcc und so weiter aussen ungerade und näher zur Antenne sind die geraden Pins.

Schick doch mal ein Bild von Deinem Aufbau.

Wenn Du die Verkabelung richtig machst kannst Du auch gleich noch den 100uF Elko zwischen Pin1&2 des nRF24 Moduls löten.

CBRocker, 2022-10-10:
@stefan123t 
Hey, was genau bringt der 100uF Elko? Eine stabilere Spannung?
Grüße

mroenne2022, 2022-10-11:
Hallo stefan123t,
ich hab die Verkabelung mehrmals kontroliert, sieht eigentlich genau so aus wie bei deinem Bsp.
PINs benutzt am ESP-WROOM-32 lt. Umsetztabelle von 38 PIN Version:
------------------------------------------------------------------

Am ESP	Am ESP		Am
38 PIN	30 PIN		NRF24L01+
------       ------		---------
3,3V	       3V3		3V3
GND	GND	GND
GPIO23	D23		MOSI
GPIO19	D19		MISO
GPIO18	D18		SCK
GPIO05	D05		CSN
GPIO16	RX2		IRQ
GPIO01	D4		CE

Allerdings hat mein ESP nur 30 PIN , von daher einige PINs lt. der Umsetztabelle benutzt.
Ob das letztendlich so OK weiß ich nicht wirklich.
Das die 3 PINs  IRQ, CE und CS beliebig wählbar sind hab auch gelesen. 
Kann man sogar im Setup in der GUI konfigurieren hab ich gesehen.
Ich glaube da wird auch das Problem liegen.
Was mich wundert - offensichtlich arbeitet das Teil auch ohne NRF , ich erreich ihn ja auch im WLAN.
Ob allerdings der WR Hoym.600 erreicht wird weiß ich erst später.
Ich frag mich ob der NRF überhaupt nötig ist ( dazu fehtl mir aber die Ahnung :-(  )
Den Tip mit dem Kondensator hab ich auch schon gelesen, hab mal die Spannung am NRF gemessen , es kommen 3.26 V an.
VG Michael R.

CBRocker, 2022-10-11:
@mroenne2022 
Hey, ist es wirklich der NRF24L01+ (also mit dem Plus?). Ich hatte anfangs den falschen bestellt (bei ebay: Kabellos Modul NRF24L01+PA+LNA with Antenne 1000 Meters Long Distance), weil ich dachte, das Plus-Zeichen in der Beschreibung bezieht sich auf den nRF.

Das wäre auf jeden Fall der richtige:
https://www.az-delivery.de/products/3x-nrf24l01-mit-2-4-ghz?variant=37105715410&utm_source=google&utm_medium=cpc&utm_campaign=azd_de_google_performance-max_labelled-products&utm_content=&utm_term=&gclid=CjwKCAjwqJSaBhBUEiwAg5W9p6dWwuP6dOONFK8Qva7P2wQIx0uiHMz4F-fcah5w5j1Lj608yp0XTRoCVowQAvD_BwE

Und wie lautet die genaue Bezeichnung auf dem ESP?

mroenne2022, 2022-10-11:
Hallo , ich hab einen mit Antenne und genau das 3er Set ohne Antenne von Amazon (Az-Delivery) , also die 3 sind defintiv die mit dem "+" Zeichen ! Mit beiden bekomme ich diese Meldung das er keine Verb. zum NRF Modul bekommet.
DEn mit Antenne muss ich nochmal genau untersuchen .....
Die genaue Bezeichnung auf dem ESP lautet:
ESP-WROOM-32
211-161007
FCC ID 2A7C

stefan123t, 2022-10-11:
@mroenne2022 
Es gibt bei ESP8266 & ESP32 eine **Mehrdeutigkeit** bei der Zuordnung der Datenpin Bezeichnungen mit `D`.
Der ESP8266 und der ESP32 haben hier jeweils andere `GPIO`s zugeordnet.
Das kann sich evtl. sogar zwischen zwei ESP32 Boards oder zwei ESP8266 Boards unterscheiden.
Wichtig und richtig weil **eindeutig** sind hierbei **immer** die **GPIO** Bezeichnungen!

Laut dem folgenden NRF24 Issue sollten wir m.E. als Default den `VSPI` (aka SPI3) anstelle des `HSPI` (aka SPI2) verwenden:
**[Question] How to use ESP32 HSPI #722** 
https://github.com/nRF24/RF24/issues/722
> Is it possible to use ESP32's HSPI instead of VSPI to communicate with nRF24L01? I would like to use pins 12, 13, 14 insead of 19, 23, 18.
> It should be  possible but **not currently supported**. Requires a few modifications.

Ich habe mal auf die Schnelle folgendes Pinout für das 30 pin ESP32 Modul gefunden:
https://www.electronicshub.org/wp-content/uploads/2021/02/ESP32-Pinout-1.jpg
[image]

und hier ein nRF24L01+ Pinout
https://lastminuteengineers.b-cdn.net/wp-content/uploads/arduino/Pinout-nRF24L01-PA-LNA-External-Antenna-Wireless-Transceiver-Module.png
[image]

| Pin | ESP32 | NRF24L01+ | Pin |
| --- | --- | --- | --- |
| Pin 14/17 | GND | GND | Pin 1 |
| Pin 16 | +3.3V | +3.3V | Pin 2 |
| Pin 20 | GPIO4 D4 | CE | Pin 3 |
| Pin 23 | GPIO5 D5 | CSN | Pin 4 |
| Pin 24 | GPIO18 D18 | SCK | Pin 5 |
| Pin 30 | GPIO23 D23 | MOSI | Pin 6 |
| Pin 25 | GPIO19 D19 | MISO | Pin 7 |
| Pin 19 | GPIO2 D2 | IRQ | Pin 8 |

D.h. im /setup/ muß man die drei Pins auch eintragen:
CS: D1 (GPIO5)
CE: D2 (GPIO4)
IRQ: D4 (GPIO2)

Die drei Pins kannst Du auch alternativ in Deiner config_override.h z.B. so angeben:
```c
#undef DEF_RF24_CS_PIN
#undef DEF_RF24_CE_PIN
#undef DEF_RF24_IRQ_PIN

// ESP32 default pinout
#define DEF_RF24_CS_PIN         5
#define DEF_RF24_CE_PIN         4
#define DEF_RF24_IRQ_PIN        2

// ESP8266 default pinout
//#define DEF_RF24_CS_PIN         15
//#define DEF_RF24_CE_PIN         2
//#define DEF_RF24_IRQ_PIN        0
```

Die folgende config sollte also bei gleicher Verdrahtung auch mit OpenDTU, funktionieren:
```c
    -DHOYMILES_PIN_MISO=19
    -DHOYMILES_PIN_MOSI=23
    -DHOYMILES_PIN_SCLK=18
    -DHOYMILES_PIN_IRQ=16
    -DHOYMILES_PIN_CE=4
    -DHOYMILES_PIN_CS=5
```

mroenne2022, 2022-10-12:
Hallo Stefan ,
ich dank dir für deine Mühe und die Ausführliche Darstellung :-) Ich werde mich beizeiten mit dem ESP32 wieder beschäftigen und hab auch die 3 PINs in Verdacht, ich denke genau da liegt das Problem.
Weil ich den ESP32 zunächst nicht zum laufen bekam hab ich mir einen D1 Mini wp8266 nachbestellt - und den hab ich heute völlig problemlos zum laufen bekommen :-) Bei der Verkabelung stimmten die 3 PINs mit der Info im Setup der GUI kpl. überein ! Aber den ESP-WROOM-32 mit 30 Pin will ich unbedingt auch noch hinbekommen demnächst , ich werde dir hier berichten woran es letztendlich lag. Sicher ist nun, das die NRF24L01+ die richtigen mit dem "+" sind ! VG Michael

Kouichi-kun, 2022-10-13:
@mroenne2022 

ich habe das selbe Setup wie folgt zum laufen bekommen. (Sowohl mit der ext. Antennen Version als auch der int.)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

mroenne2022, 2022-10-13:
Hallo @Kouichi-kun  ,
das sieht gut aus , ich danke dir für deine Info. Ich denke das bekomm ich dann auch so zum laufen, werde mich am WE damit beschäftigen, melde mich hier dann wieder !
VG Michael

mroenne2022, 2022-10-15:
Hallo !  Ich hab mal eine Frage bzgl. 2 Module parallel schalten an einem WR HM-600 ! Ich hab heute  ein 3.Modul parallel zu einem der beiden vorhandenen geschaltet ( natürlich völlig andere Himmelsrichtung ausgerichtet ). Muss ich das im "AHOY" Setup noch eintragen? Hat das irgendeine Relevanz  oder sind die Einträge für die Eingänge 1+2 je 300Wp nur informativ ? Wäre toll wenn es jemand weiß , gibt doch garantiert hier jemand der schon Module parallel am laufen hat
[image]
. Vielen Dank und Grüße. Michael R.

lumapu, 2022-10-16:
@mroenne2022  Ja ist relavant für die `Irradiation` als der prozentuale Wert der maximal möglichen Einstrahlung. Es liegt bei dir hier die Summe der Maximalleistungen deiner parallel geschalteten Module oder einen anderen Wert einzutragen.

Ich denke hiermit ist der Issue abgearbeitet und kann geschlossen werden.

closed 2022-10-16 by lumapu