lumapu/ahoy #344: Interesse an PCB Boards?

CyberLine, 2022-10-13:
Ich habe in der Vergangenheit bereits für andere OS Projekte PCB Boards entworfen. Da die Anfertigung solcher Boards nicht ganz umsonst ist, würde ich anbieten, ein Board zu entwerfen, auf dem der Wemos und der NRF Platz finden, sowie einen USB Anschluss für die Stromversorgung platzieren. Bei genügend Interessenten könnte ich diese dann Bestellen und als Warensendung weiterleiten. Der Preis pro PCB dürfte je nach Volumen im Rahmen 5 bis 15€ liegen. Vielleicht findet sich dann noch jemand mit einem 3D Drucker der entsprechende Gehäuse anfertigen kann?

CyberLine, 2022-10-13:
OK, gibts schon. Kann zu.

CyberLine, 2022-10-13:
@pascaltippelt wäre es möglich die T3001 Datei als Eagle zu Exportieren?

pascaltippelt, 2022-10-13:
Sicher, ich würde aber ein paar Tage brauchen. Beruflich bedingt bin ich derzeit knapp bei Zeit.

CyberLine, 2022-10-14:
Keine Hektik. Ich seh ja dann im Repo wenns da is. Danke!

closed 2022-10-14 by CyberLine