lumapu/ahoy #349: RSSI-Anzeige auf Hauptseite - fehlendes Leerzeichen zwischen Zahlenwert und Einheit

roku133, 2022-10-14:
Der Entwicklungs-Build 46c3d85 hat seit dem Release v0.5.17 große Fortschritte gemacht - ein tolles Ergebnis nach viel Entwicklungsarbeit.

Eine Kleinigkeit auf der Hauptseite:
Bitte bei der RSSI-Anzeige zwischen Zahlenwert und Einheit "dBm" noch ein Leerzeichen einfügen.
Das müsste die Zeile 20 in der Datei index.html betreffen.

Archibald-Harry-Tuttle, 2022-10-14:
Wenn einer die index.html schon bearbeitet,bitte auch noch den Buchstabendreher in Zeile 70 beheben:
Dort steht "anwser" ´, richtig wäre answer

Bei mir wurden heute im "Total" die KW gesamt im Format 93,4899999 angezeigt.
die 99999 rechts waren dabei statisch.
Wobei es schon spät war und eventuell nicht beide Inverter genügend Pakete geliefert haben.

Ansonsten prima geworden, auch die nacheinander Erfassung der Inverter im Setup

lumapu, 2022-10-16:
@Archibald-Harry-Tuttle bzgl. der Kommastellen evtl. einen neuen Issue aufmachen, das wird mW. auch schon von @DanielR92 und homeautomatisation2022 bearbeteitet / diskutiert

commit 8bd75741 (2022-10-16): fix #349
closed 2022-10-16 by lumapu