lumapu/ahoy #350: MQTT P_AC All YieldDayAll YieldTotalAll

chrkan, 2022-10-15:
Hallo ist es irgendwo möglich die 3 Gesamt Daten P_AC All YieldDayAll YieldTotalAll auch mittels matt zu übertragen?
Das würde das zusammenrechnen in grafana erleichtern.

Danke

Kouichi-kun, 2022-10-16:
Es sind doch nur 2 kleine Querys und eine zusammensetzung:

Das Prinzip ist bei allen 3 Werten gleich:

Aktuelle Leistung / dein P_AC:
SELECT last("value") FROM "Power-WR1" WHERE $timeFilter GROUP BY time(30s) 
SELECT last("value") FROM "Power-WR2" WHERE $timeFilter GROUP BY time(30s) 

unten in der Zusammenrechnung dann    $A + $B

Tagesertrag / dein YieldDay:
SELECT last("value") FROM "Tagesertrag-WR1" WHERE $timeFilter GROUP BY time(30s) fill(linear)
SELECT last("value") FROM "Tagesertrag-WR2" WHERE $timeFilter GROUP BY time(30s) fill(linear)

unten in der Zusammenrechnung dann    $A + $B

Gesamtertrag / dein YieldTotal:
SELECT last("value") FROM "Gesamtertrag-WR1" WHERE $timeFilter GROUP BY time($__interval)
SELECT last("value") FROM "Gesamtertrag-WR2" WHERE $timeFilter GROUP BY time($__interval)

unten in der Zusammenrechnung dann    $A + $B

[image]

chrkan, 2022-10-16:
Top danke das mit dem math hatte ich nicht auf dem schirm.

DanielR92, 2022-10-19:
Hört sich als wäre es gelöst. Könnte man diesen Issue dann closen?

closed 2022-10-19 by chrkan