lumapu/ahoy #355: Github Actions Deprecation Warnings

lumapu, 2022-10-16:
Wir haben aktuell zwei Warnings, die wir beheben sollten:

```
Node.js 12 actions are deprecated. For more information see: https://github.blog/changelog/2022-09-22-github-actions-all-actions-will-begin-running-on-node16-instead-of-node12/. Please update the following actions to use Node.js 16: benjlevesque/short-sha, cschleiden/replace-tokens
```

und

```
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/
```

DanielR92, 2022-10-19:
Node.js: 
Im projekt bei actions sind keine using's bezüglich 'node' definiert. Daher sehe ich kein Punkt zu reagieren?

save-state/set-output: 
Wird im Projekt auch nicht aktiv genutzt. 

Wie soll hier verfahren werden? Sehe kein Handlungsbedarf.

lumapu, 2022-10-19:
@DanielR92 evtl. wir node in Container der build Action eingesetzt, ich habe sonst auch keine Idee und habe die Warnungen einfach mal kopiert

stefan123t, 2022-10-19:
Habe mal die Version einer unserer genutzten github actions angepaßt.
> upgraded https://github.com/benjlevesque/short-sha from v1.2 to v2
> this may fix the two issues mentioned in #355

Hier ist die Liste der Actions die wir verwenden und deren aktuelle Version bzw. letztes Update:
* v2.5.0 https://github.com/actions/checkout
* v2.0 https://github.com/benjlevesque/short-sha
* v3.0.11 https://github.com/actions/cache
* v4.3.0 https://github.com/actions/setup-python
* 2020 https://github.com/actions/create-release
* 2021 https://github.com/cschleiden/replace-tokens
* 2020 https://github.com/actions/upload-release-asset

DanielR92, 2022-10-20:
@stefan123t Du bist ein Gott! :D

stefan123t, 2022-10-25:
@DanielR92 nicht zu früh freuen, es klappt m.W. noch immer nicht alles. Ich sehe immer noch: 
Run benjlevesque/short-sha@v2.0
  with:
    length: 7
The `set-output` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2022-10-11-github-actions-deprecating-save-state-and-set-output-commands/

lumapu, 2022-10-25:
schade, ich schaue auch mal bei Gelegenheit

commit 0630e684 (2022-10-19): upgrade actions/cache@v

try to fix #355
commit f63977a7 (2022-10-19): update actions short-sha@v2 and setup-python@v4

fix #355 in dev03
commit 05f88a9d (2022-10-19): update action setup-python@v4 and short-sha@v2

fix #355 in dev03
closed 2022-10-30 by lumapu