lumapu/ahoy #360: rpi code in __init__.py throws exception

geronet1, 2022-10-19:
```
Poll inverter 116174802823
2022-10-19 18:26:03.510398 Payload: 27 20 07 e5 03 f4 06 aa 00 64 00 00 00 00 a1 7d
Exception catched: unpack requires a buffer of 2 bytes
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/__main__.py", line 346, in <module>
    main_loop(do_init)
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/__main__.py", line 29, in main_loop
    poll_inverter(inverter, do_init)
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/__main__.py", line 108, in poll_inverter
    data = result.__dict__()
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/decoders/__init__.py", line 165, in __dict__
    data['phases'] = self.phases
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/decoders/__init__.py", line 125, in phases
    if hasattr(self, prop):
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/decoders/__init__.py", line 640, in ac_voltage_0
    return self.unpack('>H', 46)[0]/10
  File "/home/kali/rf24/ahoy/tools/rpi/hoymiles/decoders/__init__.py", line 107, in unpack
    return struct.unpack(fmt, self.response[base:base+size])
struct.error: unpack requires a buffer of 2 bytes
```

geronet1, 2022-10-20:
Es sieht so aus daß die Antwort vom WR in self.response zu kurz ist und self.response[base:base+size] dadurch leer ist.

lumapu, 2022-10-21:
hat du einen fix für den issue, wäre toll wenn du einen PR machen könntest, dann haben sie etwas davon.

closed 2023-03-08 by lumapu