lumapu/ahoy #363: ESP8266 - Empfängt nur daten wenn Ser. Verbindung aktiv

linuxdep, 2022-10-20:
### Platform

ESP8266

### Model name

Wemos D1 mini

### nRF24L01+ Module

nRF24L01+ plus, round dot

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

Default connect.

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.17

### Github Hash

5402e9b

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Debug Serial Log output

```shell
---------------------------------------
I: Welcome to AHOY!
I: 
point your browser to http://192.168.123.213
I: to configure your device
I: ----------------------------------------

I: subscribe to hsw71/HM600/devcontrol/#
I: RF24 Amp Pwr: RF24_PA_I: MAX
I: Radio Config:
SPI Frequency= 1 Mhz
Channel= 3 (~ 2403 MHz)
Model= nRF24L01+
RF Data Rate= 250 KBPS
RF Power Amplifier= PA_MAX
RF Low Noise Amplifier= Enabled
CRC Length= 16 bits
Address Length= 5 bytes
Static Payload Length= 32 bytes
Auto Retry Delay= 250 microseconds
Auto Retry Attempts= 0 maximum
Packets lost on
    current channel= 0
Retry attempts made for
    last transmission= 0
Multicast= Disabled
Custom ACK Payload= Disabled
Dynamic Payloads= Enabled
Auto Acknowledgment= Disabled
Primary Mode= RX
TX address= 0xe7e7e7e7e7
pipe 0 (closed) bound= 0xe7e7e7e7e7
pipe 1 ( open ) bound= 0x1234567801
pipe 2 (closed) bound= 0xc3
pipe 3 (closed) bound= 0xc4
pipe 4 (closed) bound= 0xc5
pipe 5 (closed) bound= 0xc6
I: [NTP]: 2022-10-20 15:24:56
I: Requesting Inverter SN 114163905727
I: Requesting Inverter SN 114163905727
I: Requesting Inverter SN 114163905727
I: RX 27B Ch23 | 95 63 90 57 27 63 90 57 27 01 00 01 01 54 01 F3 06 9E 01 57 01 EF 06 A1 00 07 B2 
I: RX 27B Ch3 | 95 63 90 57 27 63 90 57 27 02 CE 69 00 07 B9 69 04 93 04 87 09 21 13 85 0C A6 E7 
I: RX 23B Ch3 | 95 63 90 57 27 63 90 57 27 83 00 03 00 8A 03 E8 01 0F 00 13 A5 5E 92 
I: Payload (42): 00 01 01 54 01 F3 06 9E 01 57 01 EF 06 A1 00 07 CE 69 00 07 B9 69 04 93 04 87 09 21 13 85 0C A6 00 03 00 8A 03 E8 01 0F 00 13 
I: enqueuedCmd: 17
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Inverter: 0
I: HM600-Hund/ch1/U_DC: 34.000 V
I: HM600-Hund/ch1/I_DC: 4.990 A
I: HM600-Hund/ch1/P_DC: 169.400 W
I: HM600-Hund/ch1/YieldDay: 1171.000 
I: HM600-Hund/ch1/YieldTotal: 511.593 k
I: HM600-Hund/ch1/Irradiation: 44.000 %
I: HM600-Hund/ch2/U_DC: 34.300 V
I: HM600-Hund/ch2/I_DC: 4.950 A
I: HM600-Hund/ch2/P_DC: 169.700 W
I: HM600-Hund/ch2/YieldDay: 1159.000 
I: HM600-Hund/ch2/YieldTotal: 506.217 k
I: HM600-Hund/ch2/Irradiation: 44.078 %
I: HM600-Hund/ch0/U_AC: 233.700 V
I: HM600-Hund/ch0/I_AC: 1.380 A
I: HM600-Hund/ch0/P_AC: 323.800 W
I: HM600-Hund/ch0/P_ACr: 0.300 VAr
I: HM600-Hund/ch0/Freq: 49.970 Hz
I: HM600-Hund/ch0/Temp: 27.100 °
I: HM600-Hund/ch0/ALARM_MES_ID: 19.000 
I: HM600-Hund/ch0/YieldDay: 2330.000 
I: HM600-Hund/ch0/YieldTotal: 1017.810 
I: HM600-Hund/ch0/P_DC: 339.100 W
I: HM600-Hund/ch0/Efficiency: 95.488 %
I: 
I: Requesting Inverter SN 114163905727
I: RX 27B Ch40 | 95 63 90 57 27 63 90 57 27 81 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00 5D EB 3C 
I: RX 27B Ch40 | 95 63 90 57 27 63 90 57 27 81 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00 5D EB 3C 
I: Payload (14): 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00
```

linuxdep, 2022-10-20:
Wenn ich die ser. Console nicht verbunden habe, sehe ich im MQTT nur "uptime" Meldungen. Öffne ich mit Putty die Verbindung, so empfängt er die Daten vom HM600 wieder...

stefan123t, 2022-10-20:
Danke und Hurra :tada: der erste Bug mit unserem neuen Issue Template. :partying_face: Gratuliere!
Du hast den nRF24L01+ mit PA_MAX laufen, ist das notwendig ?
Versuche mal Dich von unten mit PA_MIN oder PA_LOW heranzutasten.
Außerdem hast Du einen Wemos D1 mini, bei dem kommt es ab und an vor,
dass der GPIO0 mit dem IRQ manchmal nicht gut läuft. 
Alternativ kann man den an D1 / D2 anschließen und umkonfigurieren.
Aber prinzipiell gibt es bei der v0.5.17 auch das Problem, daß er bei schlechtem WiFi Empfang auf eine 
MQTT Verbindung wartet und ohne diese teilweise keine neuen Daten abfragt.

linuxdep, 2022-10-21:
Also mit der Version Git SHA: g5b6878d :: 0.5.20 geht nix, sendet immer nur die Time und sagt fehler im letzten Packet oder so. Sehe gerade nix mehr, ist dunkel.

stefan123t, 2022-10-22:
@linuxdep nach Sonnenuntergang ist idR keine Verbindung mehr mit dem Wechselrichter möglich. Bittr bei Tag nochmal probieren Dsnke

linuxdep, 2022-10-22:
> @linuxdep nach Sonnenuntergang ist idR keine Verbindung mehr mit dem Wechselrichter möglich. Bittr bei Tag nochmal probieren Dsnke

Das ist mir schon klar... hatte meine Nachricht erst geschrieben, als es dunkel war, Test war vorher, darum wusste ich auch nicht mehr wie der genaue Wortlaut war.
Kann aber glaube zu gemacht werden, warum dies scheinbar nur bei meiner Konfiguration so ist/war, keine Ahnung. habe mal openDTU übersetzt und aufgespielt, das rennt. Bei Gelegenheit werde ich es mal aus der Quelle selber übersetzen und schauen ob es dann geht, würde schon gern beim d1 Mini bleiben. Dann werde ich auch mal D1 oder D2 testen ob es damit besser wird wenn Interesse besteht.

stefan123t, 2022-10-23:
@linuxdep das hier ist auch seltsam:
```c
I: Requesting Inverter SN 114163905727
I: RX 27B Ch23 | 95 63 90 57 27 63 90 57 27 01 00 01 01 54 01 F3 06 9E 01 57 01 EF 06 A1 00 07 B2 
I: RX 27B Ch3 | 95 63 90 57 27 63 90 57 27 02 CE 69 00 07 B9 69 04 93 04 87 09 21 13 85 0C A6 E7 
I: RX 23B Ch3 | 95 63 90 57 27 63 90 57 27 83 00 03 00 8A 03 E8 01 0F 00 13 A5 5E 92 
I: Payload (42): 00 01 01 54 01 F3 06 9E 01 57 01 EF 06 A1 00 07 CE 69 00 07 B9 69 04 93 04 87 09 21 13 85 0C A6 00 03 00 8A 03 E8 01 0F 00 13 
```
Das sind zwar alle drei Frames 01, 02 und 83 aber die DTU_ID === WR_ID. Das sollte eigentlich nicht so sein.
Warum Du das vermutlich ersetzt hast erschließt sich mir auch nicht, da ja die SN des WR immer noch im Klartext da ist.
Der nächste Verbindungsversuch ist dann noch seltsamer:
```c
I: Requesting Inverter SN 114163905727
I: RX 27B Ch40 | 95 63 90 57 27 63 90 57 27 81 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00 5D EB 3C 
I: RX 27B Ch40 | 95 63 90 57 27 63 90 57 27 81 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00 5D EB 3C 
I: Payload (14): 27 18 07 E4 02 C6 04 EA 00 68 00 00 00 00
``` 
Keine Ahnung was das für eine Anfrage war.

@lumapu warum werden eigentlich die TX Pakete hier nicht ausgegeben ( das ist auch in development03 0.5.23 nicht aktiv !!) ?
Können wir das als Default im Setup aktivieren, da das für ein Debugging m.E. wichtig ist.
Die Klartext Interpretation der Werte ist mir persönlich nicht so wichtig, die kann man ja weiter optional mit System Config > send inverter data [ ] lassen.

linuxdep, 2022-10-23:
Ich habe nix geändert im Copy.
Aber weiß nicht ob das alles noch notwendig ist? Versuche gerade mal selber zu übersetzen, bei openDTU hat das alles problemlos geklappt, jetzt bin ich dabei das erst mal mit der hardware (esp32) mit ahoy zum laufen zu bringen.

lumapu, 2022-10-23:
@stefan123t gute Beobachtung, ich denke vor den Wochenende ging es noch.
Habe bei mir gerade auch nochmal geschaut, fehlt tatsächlich.
Während dem Schreiben habe ich schon eine Idee woran es liegen könnte, ich kümmere mich darum.

lumapu, 2022-10-23:
@stefan123t gefixt in 0.5.24

linuxdep, 2022-10-23:
habe ich übersetzt, mal schauen morgen, läuft erst mal.

stefan123t, 2022-10-23:
@lumapu was hast Du denn gefixt außer dem TX Paket anzeigen ?

lumapu, 2022-10-24:
nichts im Bezug auf diesen issue, eher UI.
Schau am besten in die Commit Messages

lumapu, 2022-10-31:
gibt es schon Neuigkeiten? Bitte mal das neue Release 0.5.28 testen

closed 2022-11-06 by lumapu