lumapu/ahoy #364: ESP32 GPIO settings, what this mean "For this wiring, set the 3 individual GPIOs under the /setup URL"

azuma82, 2022-10-22:
Hallo, 

ich war bisher nur auf der Raspberry unterwegs, nun habe ich mithilfe des NodeMCU PyFlasher auch die aktuelle Ahoi Software aufgespielt, jedoch erscheint der AP nicht wie es normal sein müßte. Der Punkt mit den GPIO Settings war aufgrund fehlender Kenntnisse nicht durchführbar von mir, wo finde ich diese Setup Datei/URL? Hierzu benötige ich bzw. die Anleitung mehr Information. 

Danke vorab und beste Grüße aus dem Frankenland 

## Hardware
  * [ ] ESP8266
  * [X ] ESP32
  * [ ] Raspberry Pi

Modelname: ESP-WROOM-32
Retailer URL:AZ Delivery

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ x] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

lumapu, 2022-10-22:
Das NRF24 Modul wird mit SPI und zusätzlich einen CE und IRQ Pin verbunden. Im Setup kann man auswählen, welchen Pin man für IRQ, CS und CE verwendet hat.
Ich hoffe dass beantwortet die Fragen.

azuma82, 2022-10-22:
Vielen Dank, wenn ich jetzt nach Anleitung diesen verbunden haben muss ich das tun oder eben nicht?
Bisher klappte das Schreiben jedoch finde ich den AP nicht.

Danke vielmals für die kurzfristige Antwort

lumapu, 2022-10-22:
ja dann stimmt das default pinout. Wie kann man 

> Bisher klappte das Schreiben jedoch finde ich den AP nicht.

verstehen?

azuma82, 2022-10-22:
Hallo, 

Danke, dann muss ich woanders ansetzen, bzgl. des nicht finden des Ap (Access Point) nach dem flashen mit nodemcu-pyflasher. 

Ich befolge die Anleitung Schritt für Schritt und nun muss ich den AP finden bzw. die Ursache das er sich nicht zeigt. https://github.com/lumapu/ahoy/tree/main/tools/esp8266, so war der obere Satz gemeint. Danke für eure Hilfe

stefan123t, 2022-10-23:
Der AP ist nach Neustart / Hard-Reset für 60 Sekunden aktiv.
Du musst das Passwort esp_8266 verwenden um den AP zu erreichen.
Auf der USB Serial Console kannst Du ggf. verfolgen was der ESP macht.
Ich hatte auch mit der 0.5.23 Probleme den AP zu erreichen.
Ich habe die WiFi SSID / Passwort in einer config_override.h eingetragen und er hat sich direkt mit dem WLAN verbunden.
Vielleicht probierst Du das mal ?
Welche Version / GitHub Hash von Ahoy DTU verwendest Du ?

lumapu, 2022-10-23:
@stefan123t ich denke ich habe den AP mode gefixt, indem die Wifi Netze der Umgebung jetzt erst nach Knopfdruck gescannt werden.

azuma82, 2022-10-25:
Hallo, habe OpenDTU aufgespielt das ging, jetzt habe ich einen älteren ESP8266-12F bestellt. Ich hatte 5.17 wie auch 5.23 versucht mit dem ESP32, aktuell möchte ich erst testen wenn die neuen Boards da sind um die bestehende Konfiguration nicht zu verlieren.

closed 2022-10-25 by azuma82