lumapu/ahoy #365: Feature Request: Blindleistung regeln, wie es die orig. DTU kann

bluespiano, 2022-10-22:
Vorab allen Beteiligten mein herzliches Dankeschön für den Projektfortschritt.
Ich kann zwar nicht programmieren, aber vielleicht durch meinen Hinweis das Produkt noch besser machen .. ?

In den Beschreibungen zu Wechselrichtern der HM-Serie und auch DTU steht: "Leistung- und Blindleistung regelbar (DTU benötigt)"
Über zB Ahoy wird bei einigen Usern ein Blindleistungswert von >5 bis zu 20 oder 30VAr angezeigt.
Die Blindleistung sollte immer mögl. niedrig sein, damit sie quasi zu Wirkleistung wird.
Die originale DTU bietet anscheinend genau hier eine manuelle Regelung, die ich bei Ahoy und auch openDTU vermisse.
Gibt es da schon Erkenntnisse?

stefan123t, 2022-10-23:
Ja das ist m.E. das Kommando Reactive Power Limit bzw. alternativ kann man es über sog. Grid Profile einstellen.
Die Grid Profile sind landesweit / national vorgegeben und an die muß sich Hoymiles bzw. der Wechselrichter halten.
Es gibt zwar die Möglichkeit diese Profile auszutauschen und evtl. sogar anzupassen, aber wir haben diesen Punkt noch nicht umgesetzt.
Warum willst Du den Wert für das Blindleistungs-Limit eigentlich anpassen ?
Bei den meisten ist der ziemlich nahe bei 0 var; ist das bei Dir anders ?

bluespiano, 2022-10-23:
Danke für Deine Antwort, ja mein Wert ist >20, kenne aber auch noch andere User, wo dies so ist, deshalb die Frage.
20VA entsprechen ja 20 Watt. Wenn diese ACseitig fehlen, schon doof, wenn die originale DTU dies verhindern könnte?? So lese ich es zumindest. Insofern wäre es interessant, ob es im Projekt auf einer ToDo steht? 
Ich bin mir auch unsicher, wie die Hoymiles arbeiten. Eigentlich hätte ich vorausgesetzt, wenn der Hoymiles so etwas misst, dass er dies dann autom. selbst optimiert und keine manuelle Interaktion erforderlich ist.

stefan123t, 2022-10-23:
@bluespiano eigentlich versucht der HM das gegen 0 var zu regeln.
Es kann in den Morgen oder Abendstunden oder bei Wolken dazu kommen, daß er kurzzeitig auch da hohe var Werte angibt.
Normal sollte das aber in den Hochlastbereichen, wenn er viel Leistung einspeist m.W. nicht der Fall sein.
Es gibt auch die Option dass der HM die Leistung als Blindleistung abgibt um die Frequenz von 50 Hertz zu stützen.
Das ist also durchaus auch ein Feature der Hoymiles Wechselrichter. 
Aber ohne die entsprechenden `GridOnProFilePara` bzw. `Type_ReactivePowerContr` und `DOWN_PRO` (Sub) Kommandos im AhoyDTU / OpenDTU wird das Feature aktuell (noch) nicht unterstützt.
Bitte schau mal im Discord, dort haben das bereits einige der Nutzer und Entwickler diskutiert.

stefan123t, 2022-10-24:
Hier ein Beispiel für ein GridProfile aus dem Discord:

Dateiname: DE_VDE4105_2018
-- 

SN | Name | Wert | Einheit | Bereich
--- | --- | --- | --- | ---
1 | **DE_VDE4105_2018** |   |   |  
2 | **H/LVRT**
3 | Nominal Voltage (NV) | 230 | V | ~
4 | Low Voltage 1 (LV1) | 184 | V | 160~195.5
5 | LV1 Maximum Trip Time (MTT) | 3 | s | 3~3
6 | High Voltage 1 (HV1) | 287.5 | V | 270~287.5
7 | HV1 Maximum Trip Time (MTT) | 0.1 | s | 0.1~0.1
8 | Low Voltage 2 (LV2) | 103.5 | V | 100~150
9 | LV2 Maximum Trip Time (MTT) | 0.3 | s | 0.3~0.3
10 | 10mins Average High Voltage (AHV) | 253 | V | 250~270
11 | **H/LFRT**
12 | Nominal Frequency | 50 | Hz | ~
13 | Low Frequency 1 (LF1) | 47.5 | Hz | 47.5~49.9
14 | LF1 Maximum Trip Time (MTT) | 0.1 | s | 0.1~0.1
15 | High Frequency 1 (HF1) | 51.5 | Hz | 50.1~51.5
16 | HF1 Maximum Trip time (MTT) | 0.1 | s | 0.1~0.1
17 | **Islanding Detection (ID)**
18 | ID Function Activated |   |   | 0~1
19 | **Reconnection (RT)**
20 | Reconnect Time (RT) | 60 | s | 10~300
21 | Reconnect High Voltage (RHV) | 253 | V | 240~253
22 | Reconnect Low Voltage (RLV) | 195.5 | V | 195.5~210
23 | Reconnect High Frequency (RHF) | 50.1 | s | 50.1~50.9
24 | Reconnect Low Frequency (RLF) | 47.5 | s | 47.5~49.9
25 | **Ramp Rates(RR)**
26 | Normal Ramp up Rate(RUR_NM) | 20 | Rated%/s | 5~100
27 | Soft Start Ramp up Rate (RUR_SS) | 0.16 | Rated%/s | 0.1~10
28 | **Frequency Watt (FW)**
29 | FW Function Activated | 1 |   | 0~1
30 | Start of Frequency Watt Droop (Fstart) | 50.2 | Hz | 50.2~52
31 | FW Droop Slope (Kpower_Freq) | 40 | Pn%/Hz | 16.7~100
32 | Recovery Ramp Rate (RRR) | 0.16 | Pn%/s | 0.1~50
33 | Recovery High Frequency (RVHF) | 50.2 | Hz | 50.1~52
34 | Recovery Low Frequency (RVLF) | 49.8 | Hz | 49~49.9
35 | **Active Power Control (APC)**
36 | APC Function Activated | 0 |   | 0~1
37 | Power Ramp Rate (PRR) | 100 | Pn%/s | 0.33~100
38 | **Volt Var (VV)**
39 | VV Function Activated | 0 |   | 0~1
40 | Voltage Set Point V1 | 213.9 | V | ~
41 | Reactive Set Point Q1 | 30 | %Pn | 0~100
42 | Voltage Set Point V2 | 223.1 | V | ~
43 | Voltage Set Point V3 | 236.9 | V | ~
44 | Voltage Set Point V4 | 246.1 | V | ~
45 | Reactive Set Point Q4 | 30 | %Pn | 0~100
46 | **Specified Power Factor (SPF)**
47 | SPF Function Activated | 0 |   | 0~1
48 | Power Factor (PF) | Voreilend |   | 0.9~1
49 | **Watt Power Factor (WPF)**
50 | WPF Function Activated | 0 |   | 0~1
51 | Start of Power of WPF (Pstart) | 50 | %Pn | ~
52 | Power Factor ar Rated Power (PFRP) |   |   | 0.8~1
53 | **Reactive Power Control (RPC)**
54 | RPC Function Activated |   |   | 0~1
55 | Reactive Power (VAR) | Voreilend | %Sn | 0~50

stefan123t, 2022-10-24:
Siehe auch #96 und #225

stefan123t, 2022-10-24:
Hier die Dokumente zum Thema von der E-Control dem österreichischen Pendant zur Bundesnetzagentur

[Technische und Organisatorische Regeln für Betreiber und Benutzer von Netzen (TOR)](https://www.e-control.at/marktteilnehmer/strom/marktregeln/tor)

[TOR Erzeuger Typ A (1,3 MB)](https://www.e-control.at/documents/1785851/1811582/TOR+Erzeuger+Typ+A+V1.2.pdf/0b3e3ab1-34c1-fc59-b422-514436713933)
Anschluss und Parallelbetrieb von Stromerzeugungsanlagen des Typs A und von Kleinsterzeugungsanlagen (Maximalkapazität < 250 kW und Nennspannung < 110 kV), Version 1.2, gültig ab 18.04.2022

lumapu, 2022-10-24:
@stefan123t danke für die Tabelle, da gibt's einen Eintrag den ich gerne für meinen Wechselrichter ändern würde - das ramp up ;-)

wenn man sich meinen Screenshot anschaut erhebt man, das evtl. die Berechnung von Q_AC falsch ist oder?

[image]

homeautomation2022, 2022-10-24:
@lumapu Da ist keine Berechnung drin und in diesem untersten Teillastbereich haben die meisten elektronischen Geräte/Netzteile oder umgekehrt Wechelrichter Probleme den PowerFactor bei 1 (und die Effizienz gut) zu halten und somit nimmt der Blindleistungsanteil deutlich zu, da die Bauteile darin für einen andere Idealleistungsbereich ausgelegt sind. Kannst du auch mit PC-Netzteilen in dieser Leistungsklasse vergleichen, selbst die allerbesten mit 80+ Platinum sind da nicht gut.
Aber ab ca. 50W sollten die Werte schon sehr viel besser sein und ab 100 dann ideal?

Was willst du denn an der Rampe ändern? NA-Schutz ist langsam mit ca. 1W/s, aber wenn trennt man denn schonmal tagsüber AC??? Danach kann er ja sehr schnell hoch und runterregeln.

Was ich noch komisch finde, diese "hohen" Blindleistungswerte und Abweichungen von PF 1,0 scheinen bisher hauptsächlich bei den 4 kanaligen WRs aufzufallen. Entweder liegt es an den Bauteilen darin oder er misst anders, villeicht sogar realistischer.

stefan123t, 2022-10-25:
Hier das offizielle Hoymiles Dokument zu den o.g. Parametern des Grid Profiles:
**Technical Note - How to set Hoymiles 3rd Gen Grid Profile ?**
Version 1.2 (March 2021)
https://thecapitalsolar.com.au/wp-content/uploads/2021/04/Technical-Note-How-to-set-Hoymiles-3Gen-Grid-Profile-V1.2.pdf

rwalli, 2022-10-28:
>  Aber ab ca. 50W sollten die Werte schon sehr viel besser sein und ab 100 dann ideal?

Der Wert ändert sich bei meinem HM-1200 kaum, auch bei extrem wenig oder viel Leistung bleibt er bei ca. 19 Var.
Sollte ich was unternehmen?
[image]

lumapu, 2022-10-28:
Interessant, ich habe auch einen HM1200 und bei mir ist es genauso.

tavoc, 2022-11-21:
Hallo,

ich habe 2 HM 1500.

Einer mit 3 Modulen und der andere nur zum Test dran.
Die Q_AC ist bei beiden ca. 19-20

In der Gesamtansicht wird der Wert aber addiert - ich denke das ist falsch. Hier müsste der Mittelwert gebildet werden, oder habe ich einen Denkfehler?

[image]

stefan123t, 2022-11-21:
@tavoc eigentlich der mit den aktuellen Leistungen gewichtete Mittelwert, oder ?

tavoc, 2022-11-21:
korrekt. Aber die Addition wie oben ist falsch.

lumapu, 2022-11-21:
ok, das lässt sich beheben. Danke für den Hinweis

stefan123t, 2022-11-21:
@lumapu Danke dann können wir hier wieder mit dem Update Grid Profile Kommando weitermachen :)

GHolli, 2022-11-26:
> ok, das lässt sich beheben. Danke für den Hinweis

Blindleistungen werden (wie Wirkleistungen) auch addiert. Induktive und kapazitive Blindleistungen werden subtrahiert. Eine Mittelwertbildung daher falsch. Induktive Verbraucher werden normalerweise mit Q > 0 angegeben.

GHolli, 2022-11-28:
Wenn ich das richtig verstanden habe, sollte dieses "issue" auch #96 abdecken, oder?
In dem Fall würde dies weitere Parameter als nur die Blindleistung umfassen und der Titel sollte daher nicht nur auf die Blindleistung abzielen, sondern auch auf die anderen Einstellungen.
Mein Vorschlag wäre daher z. B. "Feature Request: Netzparameter laden (Update GridProfile), analog zur orig. DTU".

Oder soll ich dafür einen eigenen Feature Request öffnen?

P.S.: Die Leistungsbegrenzung (#422) habe ich im Dokument "Technical Note - How to set Hoymiles 3rd Gen Grid Profile" gar nicht finden können. Oder habe ich den Parameter übersehen?

GHolli, 2022-11-28:
> Hier ein Beispiel für ein GridProfile aus dem Discord:
> ## Dateiname: DE_VDE4105_2018
> [...]

@stefan123t, danke für das Profil!
Mir ist aufgefallen, dass der Parameter "Volt Watt (VW)" fehlt (lt. Technical Note "There are twelve modules in the EN 50549-1:2019 grid profile"). Im Profil sind aber lediglich 11 gelistet.
Ein paar Werte für die Parameter fehlen auch, aber ich gehe davon aus dass das ein Copy & Paste-Problem ist.

stefan123t, 2022-11-30:
@GHolli bitte schau mal ins Wiki für das Hoymiles Protokoll. Dort habe ich noch ein paar Grid Profiles mehr hinterlegt.
https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet

Das habe ich alles abgetippt, also sind auch da Fehler möglich :grinning:

stefan123t, 2022-11-30:
@GHolli gut beobachtet Kapitel 3.7 Volt Watt (VW) ist offenbar nicht in DE_VDE4105_2018 sondern nur in AT_TOR_Erzeuger_default enthalten. Ich werde die Doku im Wiki um dieses Profil demnächst ergänzen ...

GHolli, 2022-12-06:
@stefan123t vielen Dank für die Profile, die sind spitze, weil man die ohne Cloud-Zugang nicht erhält!
Danke auch für den Hinweis aufs Wiki, das hatte ich noch gar nicht beachtet!

belzig, 2023-02-16:
> @stefan123t danke für die Tabelle, da gibt's einen Eintrag den ich gerne für meinen Wechselrichter ändern würde - das ramp up ;-)
> 
> wenn man sich meinen Screenshot anschaut erhebt man, das evtl. die Berechnung von Q_AC falsch ist oder?
> 
> [image]

> @stefan123t danke für die Tabelle, da gibt's einen Eintrag den ich gerne für meinen Wechselrichter ändern würde - das ramp up ;-)

Hast du da vlt eine Lösung gefunden, mein 300er braucht biszu 6 Minuten bis er volle leistung bringt, und ich würde den gern zu Lastspitzenglättung tagsüber an der Batterie einsetzen,
Wenn der nach 6 Minuten die Leistung erreicht hat ist, die Lastspitze i.d.R schon vorbeit, daher würde ich gerne 
Soft Start Ramp up Rate (RUR_SS) auf 10 setzen. Ist das mit Ahoy DTU möglich. (Die Hoymiles Pro DTU kann das angeblich)

hbarthel, 2023-03-29:
Gibt es Neuigkeiten bzgl. Setzen des Grid Profiles? Mein HM-600 schaltet auf 0 Watt, wenn die Spannung kurz über 253V geht. Das würde ich gern ändern.

ReneSpanl, 2023-05-27:
Wäre auch an einer Anpassung oder Austauschen des Gridfiles interessiert. Kioto/Sonnenkraft hatte damals BKWs mit 70% Begrenzung (was völlig unnötig ist) ausgeliefert, das gilt es jetzt auszuschalten. Hat jemand eine Idee dafür?

stefan123t, 2023-10-09:
Dafür sollten wir erstmal das Grid Profile auslesen können, bevor wir es drüber flashen können und wollen.

Siehe auch Issues 
https://github.com/tbnobody/OpenDTU/issues/1369, 
https://github.com/tbnobody/OpenDTU/issues/987, 
https://github.com/tbnobody/OpenDTU/issues/707, 
https://github.com/tbnobody/OpenDTU/issues/900, 
https://github.com/tbnobody/OpenDTU/issues/679 
sowie die Discussions 
https://github.com/tbnobody/OpenDTU/discussions/1394, 
https://github.com/tbnobody/OpenDTU/discussions/364, 
https://github.com/tbnobody/OpenDTU/discussions/550.

SN | Name | Value | Unit | Range
-- | -- | -- | -- | --
**46** | **Specified Power Factor (SPF) (?)** |   |   |  
47 | SPF Function Activated | 0 |   | 0~1
48 | Power Factor (PF) | v0.95 |   | 0.9~1
**49** | **Watt Power Factor (WPF) (?)** |   |   |  
50 | WPF Function Activated | 0 |   | 0~1
51 | Start of Power of WPF (Pstart) | 50 | %Pn | ~
52 | Power Factor ar Rated Power (RFRP) | 0.95 |   | 0.8~1
**53** | **Reactive Power Control (RPC) (?)** |   |   |  
54 | RPC Function Activated | 0 |   | 0~1
55 | Reactive Power (VAR) | v 0 | %Sn | 0~50

Was das Grid Profile an sich angeht wurden bereits diese "Topcis" hier dokumentiert:
https://github.com/lumapu/ahoy/wiki/Protocol#file-name-de_vde4105_2018

Von @lumapu und Flole habe ich ein GridProfile als Response auf eine  bekommen und ansatzweise ausgewertet.
Das kommt sogar einigermaßen hin mit den Topics. 
Ich bin nur noch nicht ganz sicher wie die Topic Überschriften (fett & gelb) zu interpretieren sind. 
Wie das ganze im Config-Speicher des TMS320F28034 von Texas Instruments abgelegt wird ist noch unbekannt.

[DE_VDE4105_2018_v2.0.0_20230924.xlsx](https://github.com/tbnobody/OpenDTU/files/12850517/DE_VDE4105_2018_v2.0.0_20230924.xlsx)

In Ahoy wurde das Thema bisher in https://github.com/lumapu/ahoy/issues/96 und vor allem hier https://github.com/lumapu/ahoy/issues/365 analysiert.

stefan123t, 2023-11-22:
@lumapu see the following gist for an almost feature complete Grid Profile parser in python by @noone2k.
https://gist.github.com/noone2k/0b3a116a6f35286abef7199b62a0777a

@tbnobody started to document most of the grid profiles on this wiki https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser

stefan123t, 2023-11-23:
@lumapu nachdem @tbnobody bereits eine große Zahl von Grid Profiles dokumentiert hat:
https://github.com/tbnobody/OpenDTU/wiki/Grid-Profile-Parser
und @noone2k den passenden Parser gebaut hat, der diese Grid Profile lesen kann, sollte es jetzt an der Zeit sein auch mal zu probieren ob wir A) ein Grid Profile Apply auf der Funkstrecke tracen können oder B) direkt eines der bekannten Grid Profiles (evtl. an einer Stelle und im CRC16 geändert) auf den Wechselrichter geflasht bekommen.

##### [Wie wird das DOWN_PRO (0x0E) / DOWN_DAT (0x0A) verwendet](https://github.com/lumapu/ahoy/wiki/Protocol#wie-wird-das-down_pro-0x0e--down_dat-0x0a-verwendet)

>   Type_Init (0xFF=255) Das SubCmd = Type_Init (0xFF) scheint nur bei MainCmd = DOWN_PRO (0x0E=14) oder DOWN_DAT (0x0A=10) verwendet zu werden.

Könnten wir das ebenfalls analog zum Request des GridProfile oben über die Web API ansteuern ?
Die Payload ist vermutlich einfach das Grid Profile inkl. der CRC16 Checksumme.
Das müsste die AhoyDTU / OpenDTU REST API eben entsprechend in Pakete aufteilen und mit CRC8 an den Wechselrichter schicken.

lumapu, 2023-12-28:
Anzeige des Gridprofiles ist jetzt möglich, ab Version 0.8.29

stefan123t, 2024-01-18:
Prima! Danke an nivadis für den Dump beim Grid Profile DOWN_DAT 0x0A Kommando:

Ich glaube das ist es:
```code
TX 6.316844960000000,0x7E 0A 80722536 74438179 01 03002000000A08FC0730001E0B3B0001 E0 7F,,
TX 6.366399360000000,0x7E 0A 80722536 74438179 02 040B001E09E210001388128E0001141E C0 7F,,
TX 6.417439200000000,0x7E 0A 80722536 74438179 03 000120000001300302580AC807A31392 89 7F,,
TX 6.467220000000000,0x7E 0A 80722536 74438179 04 128E400007D003E850080001139C0190 87 7F,,
TX 6.517172800000000,0x7E 0A 80722536 74438179 05 001001F6137470020001271080000000 65 7F,,
TX 6.567292640000000,0x7E 0A 80722536 74438179 06 085B012C08B70941099D012C90000000 82 7F,,
TX 6.628907360000000,0x7E 0A 80722536 74438179 07 FFA1B000000001F4005FA00200000000 C5 7F,,
TX 6.667958880000000,0x7E 0A 80722536 74438179 88 9C27 17 7F,,

TX 6.767887520000000,0x7E 0A 80722536 74438179 88 9C27 17 7F,,
... alle 100 ms wiederholt
TX 8.984026399999999,0x7E 0A 80722536 74438179 88 9C27 17 7F,,
RX 9.041319200000000,0x7E 8A 80722536 74438179 81 000003002000 9F19 80 7F,,
TX 9.107654080000000,0x7E 0A 80722536 74438179 88 9C27 17 7F,,
RX 9.113543840000000,0x7E 8A 80722536 74438179 81 000003002000 9F19 80 7F,,
... alle 100 ms wiederholt
TX 26.587171999999999,0x7E 0A 80722536 74438179 88 9C27 17 7F,,
RX 26.589525120000001,0x7E 8A 80722536 74438179 81 000003002000 9F19 80 7F,,
... die CRC16 Checksumme wird bis zu 20 Sekunden wiederholt ... dann sollte der Inverter das persistiert haben.

... kurze Pause ~120-130 ms
TX 26.717758400000001,0x7E 15 80722536 74438179 80 0B0065A9393B000000000000000039F6B1 7F,,
RX 26.724774239999999,0x7E 95 80722536 74438179 01 000100F9012C02E9000001A0000A099FB9 7F,,
```

Man sieht in der letzten TX Zeile des oberen Blocks die CRC16 über das Grid Profile.
Danach wiederholt die DTU brav das letzte Paket 0x88 mit der Checksumme 0x9C27 
bis 26.587 Sekunden, also insgesamt ca. 20 Sekunden lang, das sollte genügen, 
damit der WR das Grid Profile persistieren / flashen kann.

@lumapu @tbnobody vielleicht wollt Ihr das irgendwann demnächst mal ins Projekt einbauen ?

stefan123t, 2024-07-05:
@lumapu I have described the sequence for the DOWN_DAT 0x0A command which `@nivadis` traced on his HMS in more detail here.

https://github.com/tbnobody/OpenDTU/issues/900#issuecomment-2209546838

stefan123t, 2024-10-27:
@lumapu hat keiner Interesse das mal zu implementieren ?

lumapu, 2024-11-07:
weiß immer noch nicht, ob das Teil einer DTU sein sollte - wir bewegen uns damit auch in einem Bereich, der den Einsatz von Ahoy u.U. kritisch macht.

GHolli, 2025-01-27:
> > weiß immer noch nicht, ob das Teil einer DTU sein sollte - wir bewegen uns damit auch in einem Bereich, der den Einsatz von Ahoy u.U. kritisch macht.

Stimmt, aber: die meisten Benutzer dieser alternative Lösung werden (hoffentlich) einigermaßen wissen, was sie tun. Ein falsches Profil kann man schließlich auch mit der originalen DTU auswählen. Und ich finde es fast schade, diese Funktion nicht anzubieten, da viele Profile und Kommandos bereits analysiert und zusammengestellt wurden, soweit ich verstanden habe.
Ich würde es auch insofern nützlich finden, da die Wechselrichter noch einige Zeit am Markt sein werden (teilweise auch gebraucht). Wenn man nun seine Anlage erweitern möchte und sich noch so einen Wechselrichter holt und zu dem Zeitpunkt eine DTU (gerade) nicht mehr (zu einem einigermaßen vernünftigen Preis) verfügbar ist, ist das lästig (so sind die Profile von Deutschland und Österreich auch etwas unterschiedlich).
Und um das Risiko "was macht den dieser Knopf" zu reduzieren, könnte man auch ein
#ifdef DANGER_ZONE
oder so einbauen, damit nur nicht jeder, der das Projekt einfach schnell einmal ausprobiert, unachtsam den Wechselrichter mit nicht erlaubten Werten konfiguriert.

stefan123t, 2025-05-24:
Ich würde vorschlagen nur die bisher bekannten Hoymiles DTU Profile als Auswahl zu erlauben.

Wir haben auf der Wiki Seite alle bisher bekannten und m.E. relevanten [Grid Profile](https://github.com/lumapu/ahoy/wiki/Protocol#file-name-at_tor_erzeuger_default) dokumentiert.

Somit wäre eine Änderung z.B. von DE VDE4015_2018 auf AT TOR_Erzeuger_default möglich, wenn man dabei bestätigt, dass man den WR auch in dem Land für das gewünschte Zielprofil betreibt. Man könnte auch die Information zum Betreiberland der OpenDTU aus der NTP > Zeitzone verwenden um das Land zu bestimmen 😜 

Diego Forte wrote a small [Javascript](https://codepen.io/diego-fortes/pen/YzEPxYw) which translates timezones to country codes

```javascript
Intl.DateTimeFormat().resolvedOptions().timeZone
```

from: [How to get visitor's location (i.e. country) using geolocation?](https://stackoverflow.com/questions/3489460/how-to-get-visitors-location-i-e-country-using-geolocation)