lumapu/ahoy #367: No connection to Hoymiles HM-1200

Birndlmaier, 2022-10-22:
Hello,

I just can't connect to my inverter.
The funny thing is that I get the message on the home screen that it is connected, but only if I don't get any NTP time. As soon as NTP is available, the status changes back to not available. Only TX messages leave the ESP. RX Nothing arrives
Anyone have any ideas what else I could try? I've also switched between different D1 mini. No success

## Hardware
  * [ ] ESP8266
   -> V3.0.0

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Wie hier vorgschlagen

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

## Software
* [ ] AhoyDTU

### Build & Flash Method:
* [ ] ESP Home Flasher

### Desktop OS:
* [ ] Windows

### Debugging:
* [ ] USB Serial Log (attached)

[12:32:33]I: Welcome to AHOY!
[12:32:33]I: 
[12:32:33]point your browser to http://192.168.178.XX
[12:32:33]I: to configure your device
[12:32:33]I: ----------------------------------------
[12:32:33]
[12:32:33]I: RF24 Amp Pwr: RF24_PA_I: LOW
[12:32:33]I: Radio Config:
[12:32:33]SPI Frequency		= 1 Mhz
[12:32:33]Channel			= 3 (~ 2403 MHz)
[12:32:33]Model			= nRF24L01+
[12:32:33]RF Data Rate		= 250 KBPS
[12:32:33]RF Power Amplifier	= PA_LOW
[12:32:33]RF Low Noise Amplifier	= Enabled
[12:32:33]CRC Length		= 16 bits
[12:32:33]Address Length		= 5 bytes
[12:32:33]Static Payload Length	= 32 bytes
[12:32:33]Auto Retry Delay	= 250 microseconds
[12:32:33]Auto Retry Attempts	= 0 maximum
[12:32:33]Packets lost on
[12:32:33]    current channel	= 0
[12:32:33]Retry attempts made for
[12:32:33]    last transmission	= 15
[12:32:33]Multicast		= Disabled
[12:32:33]Custom ACK Payload	= Disabled
[12:32:33]Dynamic Payloads	= Enabled
[12:32:33]Auto Acknowledgment	= Disabled
[12:32:33]Primary Mode		= RX
[12:32:33]TX address		= 0xdeadbeef01
[12:32:33]pipe 0 (closed) bound	= 0xdeadbeef01
[12:32:33]pipe 1 ( open ) bound	= 0x1234567801
[12:32:33]pipe 2 (closed) bound	= 0xc3
[12:32:33]pipe 3 (closed) bound	= 0xc4
[12:32:33]pipe 4 (closed) bound	= 0xc5
[12:32:33]pipe 5 (closed) bound	= 0xc6
[12:32:39]I: [NTP]: 2022-10-22 12:32:35
[12:32:39]I: Requesting Inverter SN 1161612xxxxxx
[12:32:54]I: Requesting Inverter SN 1161612xxxxxx
[12:33:08]I: Power limit for inverter 0 set to 100%
[12:33:09]I: Requesting Inverter SN 1161612xxxxxx
[12:33:09]I: Devcontrol request 11 power limit 100
[12:33:09]I: TX 19B Ch75 | 51 61 21 13 63 78 56 34 12 80 0B 00 03 E8 01 01 40 41 08 
[12:33:09]I: enqueuedCmd: 5
[12:33:24]I: Requesting Inverter SN 1161612xxxxxx
[12:33:24]I: Devcontrol request 11 power limit 100
[12:33:24]I: TX 19B Ch3 | 51 61 21 13 63 78 56 34 12 80 0B 00 03 E8 01 01 40 41 08 
[12:33:24]I: enqueuedCmd: 5

lumapu, 2022-10-22:
can you try the development branch binary?

Birndlmaier, 2022-10-22:
@lumapu  > where can i find the file?

lumapu, 2022-10-22:
go to actions (available in Browser only). Make shure to be logged in.  
Click on the last successful action and scroll to the very bottom. there you can download a zip with the firmware

Birndlmaier, 2022-10-22:
Thank you 

13:14:23 W: while retrieving data: last frame missing: Request Retransmit
13:14:23 I: sendTimePacket
13:14:23 I: TX 27B Ch3 | 15 61 21 13 63 86 91 21 86 80 05 00 63 53 d0 86 00 00 00 00 00 00 00 00 0a 6a 16 
13:14:29 I: enqueued cmd failed/timeout
13:14:29 I: Inverter #0 I: no Payload received! (retransmits: 5)
13:14:29 I: resetPayload: id: 0
13:14:29 I: Requesting Inverter SN 1161612XXXX
13:14:29 I: enqueuedCmd: 11
13:14:29 I: enqueuedCmd: 1
13:14:29 I: enqueuedCmd: 5
13:14:29 I: sendTimePacket
13:14:29 I: TX 27B Ch23 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:30 W: while retrieving data: last frame missing: Request Retransmit
13:14:30 I: sendTimePacket
13:14:30 I: TX 27B Ch40 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:32 W: while retrieving data: last frame missing: Request Retransmit
13:14:32 I: sendTimePacket
13:14:32 I: TX 27B Ch61 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:34 W: while retrieving data: last frame missing: Request Retransmit
13:14:34 I: sendTimePacket
13:14:34 I: TX 27B Ch75 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:36 W: while retrieving data: last frame missing: Request Retransmit
13:14:36 I: sendTimePacket
13:14:36 I: TX 27B Ch3 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:38 W: while retrieving data: last frame missing: Request Retransmit
13:14:38 I: sendTimePacket
13:14:38 I: TX 27B Ch23 | 15 61 21 13 63 86 91 21 86 80 0b 00 63 53 d0 95 00 00 00 00 00 00 00 00 f4 bc 23 
13:14:44 I: enqueued cmd failed/timeout
13:14:44 I: Inverter #0 I: no Payload received! (retransmits: 5)
13:14:44 I: resetPayload: id: 0
13:14:44 I: Requesting Inverter SN 1161612XXXX
13:14:44 I: sendTimePacket
13:14:44 I: TX 27B Ch40 | 15 61 21 13 63 86 91 21 86 80 01 00 63 53 d0 a4 00 00 00 00 00 00 00 00 6f ef d0 
13:14:45 W: while retrieving data: last frame missing: Request Retransmit
13:14:45 I: sendTimePacket
13:14:45 I: TX 27B Ch61 | 15 61 21 13 63 86 91 21 86 80 01 00 63 53 d0 a4 00 00 00 00 00 00 00 00 6f ef d0 
13:14:47 W: while retrieving data: last frame missing: Request Retransmit
13:14:47 I: sendTimePacket
13:14:47 I: TX 27B Ch75 | 15 61 21 13 63 86 91 21 86 80 01 00 63 53 d0 a4 00 00 00 00 00 00 00 00 6f ef d0 
13:14:49 W: while retrieving data: last frame missing: Request Retransmit
13:14:49 I: sendTimePacket
13:14:49 I: TX 27B Ch3 | 15 61 21 13 63 86 91 21 86 80 01 00 63 53 d0 a4 00 00 00 00 00 00 00 00 6f ef d0

lumapu, 2022-10-22:
I can only see TX messages, but there should be some RX too

Birndlmaier, 2022-10-23:
Yes, thats my Problem. I wrote this in my first Post. I have no Idea what can i do.

lumapu, 2022-10-23:
Wie weit ist der Wechselrichter ca von ESP entfernt?
Könntest du mal versuchen den IRQ auf einen anderen Pin zu legen?

Birndlmaier, 2022-10-23:
Aktuell ca. 30 cm. Getrennt durch schindeldach.  Okay dann muss ich wohl die Platine bearbeiten.

lumapu, 2022-10-23:
30cm ist ja sehr nah. evtl kannst du die Distanz testweise mal auf zwei Meter erhöhen

stefan123t, 2022-10-23:
@Birndlmaier der Wemos D1 mini hat mW ab und an Probleme mit IRQ an GPIO0, probiere es mal an D1 / D2.
Das müsste erklären warum er keine RX Pakete signalisiert bekommt.

Birndlmaier, 2022-10-24:
Hallo zusammen, 

ich habe denn Pin jetzt auf D2 umgelötet. Die Sendeleistung auf LOW gestellt und bin jetzt ca. 2 bis 3 vom WR entfernt.

15:09:09 I: enqueued cmd failed/timeout
15:09:09 I: Inverter #0 I: no Payload received! (retransmits: 5)
15:09:09 I: resetPayload: id: 0
15:09:09 I: Requesting Inverter SN 116161XXXXX
15:09:09 I: sendTimePacket
15:09:09 I: TX 27B Ch40 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:10 W: while retrieving data: last frame missing: Request Retransmit
15:09:10 I: sendTimePacket
15:09:10 I: TX 27B Ch61 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:12 W: while retrieving data: last frame missing: Request Retransmit
15:09:12 I: sendTimePacket
15:09:12 I: TX 27B Ch75 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:14 W: while retrieving data: last frame missing: Request Retransmit
15:09:14 I: sendTimePacket
15:09:14 I: TX 27B Ch3 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:16 W: while retrieving data: last frame missing: Request Retransmit
15:09:16 I: sendTimePacket
15:09:16 I: TX 27B Ch23 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:18 W: while retrieving data: last frame missing: Request Retransmit
15:09:18 I: sendTimePacket
15:09:18 I: TX 27B Ch40 | 15 61 21 13 63 86 91 21 86 80 01 00 63 56 8e 75 00 00 00 00 00 00 00 00 42 0c 94 
15:09:24 I: enqueued cmd failed/timeout
15:09:24 I: Inverter #0 I: no Payload received! (retransmits: 5)
15:09:24 I: resetPayload: id: 0
15:09:24 I: Requesting Inverter SN 11616121lXXXX
15:09:24 I: sendTimePacket
15:09:24 I: TX 27B Ch61 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d 
15:09:25 W: while retrieving data: last frame missing: Request Retransmit
15:09:25 I: sendTimePacket
15:09:25 I: TX 27B Ch75 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d 
15:09:27 W: while retrieving data: last frame missing: Request Retransmit
15:09:27 I: sendTimePacket
15:09:27 I: TX 27B Ch3 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d 
15:09:29 W: while retrieving data: last frame missing: Request Retransmit
15:09:29 I: sendTimePacket
15:09:29 I: TX 27B Ch23 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d 
15:09:31 W: while retrieving data: last frame missing: Request Retransmit
15:09:31 I: sendTimePacket
15:09:31 I: TX 27B Ch40 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d 
15:09:33 W: while retrieving data: last frame missing: Request Retransmit
15:09:33 I: sendTimePacket
15:09:33 I: TX 27B Ch61 | 15 61 21 13 63 86 91 21 86 80 05 00 63 56 8e 84 00 00 00 00 00 00 00 00 12 00 3d

lumapu, 2022-10-24:
sieht nicht besser aus  Hat du den IRQ Pin im Setup geändert?
Stimmen die anderen beiden Pins auch mit dem Setup überein?

Birndlmaier, 2022-10-24:
@lumapu dies habe ich alles nochmals überprüft und auch durchgemessen. Wäre es möglich das der WR keine Daten schickt wenn ich nur 3 Platten laufen habe? Dies wäre jetzt noch das einzige was von der "Norm" abweicht. Anonsten gehe ich davon aus das der WR einfach einen an der klatsche hat.

lumapu, 2022-10-24:
nein es reicht ein Modul und 3W

lumapu, 2022-10-24:
ich würde eher das nrf24 Modul als Fehler betrachten, ich kann mir vorstellen, dass hoymiles die Wechselrichter testet während der Produktion

stefan123t, 2022-10-24:
@Birndlmaier what is your connection diagram. We have seen nRF24 Modules which had a wrong (mirrored) Pin Layout printed on the lower layer. The top layer should have a small square around Pin 1 GND.
Can you double check and add a picture of your connections ?

Birndlmaier, 2022-10-25:
I'll check that tonight.

Birndlmaier, 2022-10-25:
[image]

it seems i didn't get a good nrf24 module.  unfortunately my circuit board was pretty badly affected by all the handicraft work.  so I had to make some silver wire bridges.

lumapu, 2022-10-25:
was bedeutet das jetzt? Läuft es ist sind weniger Fehler im Aufbau?
Bist echt hartnäckig bei den steinigen Weg zu einem funktionierenden AhoyDTU

Birndlmaier, 2022-10-26:
Fehler im Aufbau gibt es keine. Alle Pins gehen da hin wo sie hingehören. 
Ich sehe jedoch kein aufgedrucktes Rechteck bei Ground vom nrf24. Wäre ein ESP32 besser geeignet? So einer würde auch noch rumliegen.

lumapu, 2022-10-31:
gibt's schon Neuigkeiten? Der ESP32 wäre auf jeden Fall eine Option.
Flashen sollte jetzt auch einfacher gehen:
https://ahoydtu.de/flash

Birndlmaier, 2022-11-05:
Kurzes Feedback:
Neue Platine, neues NRF24 Modul. Keine Verbindung. Ich habe jetzt 3 Module probiert.
Ich gebe mich damit jetzt zufrieden das der WR anscheinend in dieser Richtung einen Defekt hat oder ein chinesischer Mitarbeiter die Falschen Sr. Nr. Aufkleber draufgeklebt hat.  Issue kann geschlossen werden.   Finde das Projekt aber cool und ziehe vor dir den Hut sowas auf die Beine zu stellen.

lumapu, 2022-11-05:
schade zu hören. Bist du auch in Discord aktiv, wenn ja schreib mir doch einfach Mal per PM?

Ich habe gestern auch ein Modul aufbaut und noch keinen Empfang. Ich habe unser Default Pinout verwendet weiß aber noch nicht warum nichts kommt.

[image]

Birndlmaier, 2022-11-05:
UPDATE:
Ich habe den WR jetzt 2 Stunden komplett vom Netz getrennt und die Module abgesteckt. Dann wieder alles angesteckt und wieder am Netz angesteckt. Siehe da > Es kommen Daten. Warum auch immer.  Wahrscheinlich hätte es mit den 2 anderen Modulen auch schon funktioniert.  Egal.  Danke an das Team

closed 2022-11-05 by lumapu