lumapu/ahoy #368: Feature Request / Improvement: Add and display (Multi-) Sensor for Monitoring the ESP8266 (e.g. BME280)

andi-bar157, 2022-10-22:
## Hardware
x - ESP8266
 
Modelname: Wemos D1 mini 
Retailer URL: 

### nRF24L01+ Module
x -  nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
x circuit board

### Power Stabilization:
x  - 100uF Electrolytic Capacitor 

### Connection diagram:
n.A.

### Connection diagram I used:
n.a.

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
x - AhoyDTU

### Version / Git SHA: 
n.A.

### Build & Flash Method:
n.A.

### Desktop OS:
n.A.

### Debugging:
n.A.

lumapu, 2022-11-06:
Verstehe den Feature Request nicht, bitte nochmal detailierter beschreiben. Ist die Funktion für die Allgemeinheit nützlich?

stefan123t, 2022-11-06:
@lumapu BME280 ist ein Bosch Environment Sensor und mißt neben Luftdruck, Temperatur und Luftfeuchtigkeit. Wird m.W. per I2C Bus angeschloßen. Ich weiß jedoch nicht ob das für die DTU so interessant ist, wenn die im Gebäude aufgestellt ist und der WR bzw. die PV-Panele vor der Tür stehen. Wenn man die DTU aber auch draußen aufstellt und noch genügend Pins zur Verfügung hat, könnte man die gewünschten Werte ermitteln. ~~Eventuell reicht aber auch ein AM2302 oder ein DS18B20 für die Umgebungs-Temperatur. Obwohl die gibt der WR ja selbst aus ... Vielleicht kann man ja auch die Temperatur des ESP8266 aus einem Register abfragen~~ ?

closed 2023-09-16 by lumapu