lumapu/ahoy #369: RaspberryPi Exception while receiving data payload

McSplanish, 2022-10-23:
### Platform

RaspberryPi

### Model name

Raspberry Pi 4

### nRF24L01+ Module

nRF24L01+ plus, round dot

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.17

### Github Hash

5402e9b

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
2022-10-23 12:05:57.221957 Payload: 00 01 01 50 01 1f 03 c5 00 00 2c 9f 00 e6 08 c0 13 8a 03 9a 00 00 00 29 03 e8 00 f5 00 04 67 72
2022-10-23 12:05:57.221957 Decoded: temp=24.5 phase0=voltage:224.0, current:0.41, power:92.2, frequency:50.02 string0=voltage:33.6, current:2.87, power:96.5, total:11.423, daily:230
{'inverter_ser': 112182060088, 'inverter_name': None, 'dtu_ser': None, 'phases': [{'voltage': 224.0, 'current': 0.41, 'power': 92.2}], 'strings': [{'voltage': 33.6, 'current': 2.87, 'power': 96.5, 'energy_total': 11423, 'energy_daily': 230}], 'temperature': 24.5, 'frequency': 50.02, 'powerfactor': None, 'event_count': None, 'time': datetime.datetime(2022, 10, 23, 12, 5, 57, 222473)}
Exception catched: '<' not supported between instances of 'int' and 'NoneType'
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/csommerfeld/ahoy/ahoy/tools/rpi/hoymiles/__main__.py", line 317, in <module>
    main_loop(do_init)
  File "/home/csommerfeld/ahoy/ahoy/tools/rpi/hoymiles/__main__.py", line 29, in main_loop
    poll_inverter(inverter, do_init)
  File "/home/csommerfeld/ahoy/ahoy/tools/rpi/hoymiles/__main__.py", line 125, in poll_inverter
    if event_message_index[inv_str] < data['event_count']:
TypeError: '<' not supported between instances of 'int' and 'NoneType'
```

### Error description

First off all, im getting very rear data from my HM-300. I would say from 100 requests i'm getting 1 response. I'm working on it and i ask about that in the mikrocontroller forum already.
If i get an answer from my HM-300 i always getting an exception, with the following payload:

2022-10-23 12:05:57.221957 Payload: 00 01 01 50 01 1f 03 c5 00 00 2c 9f 00 e6 08 c0 13 8a 03 9a 00 00 00 29 03 e8 00 f5 00 04 67 72

please have a look at debug serial log output.
Any idea?

stefan123t, 2022-10-23:
@McSplanish sorry mit dem Python code kenne ich mich nicht aus. 
I do not know the python code well enough to help here.

I have seen that @geronet1 has opened a similar issue #360 recently.
Maybe you can figure it out together or ask @DanielR92 or @Sprinterfreak as they have been using it before ?

Sprinterfreak, 2022-10-23:
Bug introduced by fb6250a1098f371d96fe409849e011ce1b970f19
Comparison against undefined variables.

Sprinterfreak, 2022-10-23:
> @Sprinterfreak as they have been using it before ?

I'm no gonna merge anymore, because my base has diverted too much that it would make sense to pull in changes from here

stefan123t, 2022-10-24:
@Sprinterfreak I think this would be a good move forward as this would leave room to have the C++ implementations for ESP8266 / ESP32 in root and would alleviate some troubles as to the PlatformIO project files too. 
Do you want to create a separate repo or should we link to your ahoy Fork ?

@lumapu Should we drop the Python Code from the repo and replace it with a nice and friendly pointer to Jan-Jonas Repo ?
Maybe we can discuss the path forward on the Discord chat together with Jan-Jonas and the other users / contributors to the python code ?

McSplanish, 2022-10-24:
I really would like to participate as well. But I have to stabilize the communication between the pi and my HM-300 in the first place.
I would drop the python code to avoid any misunderstanding and link to the other repo.
On the other side it would be great to have all together.

Sprinterfreak, 2022-10-24:
The connection reliability also highly depends on Your environment. Since there is no such CSMA or a like, collisions are going to happen. If for instance, You have other appliances heavily using 2,4GHz lots of packet loss is to be expected.
Anyways, You going to get much more granular data compared to the Hoymiles cloud service.

McSplanish, 2022-10-24:
Sorry, I know it is something different but could it be a problem to place the nrf24 module next to my Wifi router?

stefan123t, 2022-10-24:
@McSplanish we have repeated requests for nRF24L01+ quality and settings.
Recommendations and Experience so far is to prefer the shielded modules with external antenna above unshielded or circuit board antenna. That is the LNA+PA modules with external antenna do have a higher yield.
I have compared two modules cheap with circuit board vs. another with LNA+PA and external antenna.
The one with ext. antenna received in MIN settings fine enough. The other I had to set to LOW / HIGH settings.
It is generally not advised to use MAX settings and to add an electrolytic capacitor to the GND and VDD Pins 1 & 2 of your nRF Module in order to stabilize voltage supply to the module during sending. We estimate some 100-300 mA may be needed.
I do not know if this is really a problem with the Raspberry Pi with ample power supply.
The antennas may be placed apart (31mm is ~1/4 of the wavelength) and twisted into different planes if possible.

closed 2023-03-08 by lumapu