lumapu/ahoy #375: [ESP32] WiFiUdp.cpp NTP problem?

DanielR92, 2022-10-26:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_4 CE    |
| Pin 4 CSN     | GPIO_5 CS    |
| Pin 5 SCK     | 18 SCLK  |
| Pin 6 MOSI    | 23 MOSI  |
| Pin 7 MISO    | 19 MISO  |
| Pin 8 IRQ     | GPIO_17 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.22

### Github Hash

gec83406

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Windows

### Setup

{"system":{"ssid":"Daniels Home","device_name":"AHOY-DTU","version":"0.5.22","build":"gec83406","ts_uptime":331,"ts_now":0,"ts_sunrise":0,"ts_sunset":0,"ts_sun_upd":0,"wifi_rssi":-81,"disclaimer":false,"esp_type":"ESP32"},"inverter":{"inverter":[{"id":0,"name":"Solarkraftwerk","serial":"116174403329","channels":4,"version":"0","ch_max_power":[0,0,0,0],"ch_name":["","","",""]}],"interval":"30","retries":"5","max_num_inverters":4},"mqtt":{"broker":"192.168.10.30","port":"1883","user":"","pwd":"","topic":"inverter"},"ntp":{"addr":"pool.ntp.org","port":"123"},"sun":{"lat":"","lon":"","disnightcom":false},"pinout":{"cs":5,"ce":4,"irq":17},"radio":{"power_level":1},"serial":{"interval":5,"show_live_data":false,"debug":false}}

### Debug Serial Log output

```shell
...
21:09:49.365 > Custom ACK Payload       = Disabled
21:09:49.368 > Dynamic Payloads = Enabled
21:09:49.371 > Auto Acknowledgment      = Disabled
21:09:49.373 > Primary Mode             = RX
21:09:49.376 > TX address               = 0xdeadbeef01
21:09:49.379 > pipe 0 (closed) bound    = 0xdeadbeef01
21:09:49.382 > pipe 1 ( open ) bound    = 0x4410608101
21:09:49.385 > pipe 2 (closed) bound    = 0xc3
21:09:49.387 > pipe 3 (closed) bound    = 0xc4
21:09:49.391 > pipe 4 (closed) bound    = 0xc5
21:09:49.393 > pipe 5 (closed) bound    = 0xc6

21:09:49.396 > [ 16355][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9

21:09:51.401 > I: connect to network 'Daniels Home' ...
21:09:51.404 > .

21:10:01.957 > I: [NTP]: n/a UTC

21:10:04.683 > I: subscribe to inverter/devcontrol/#
```

### Error description

1) Reboot
2) Console put that out: [ 16355][E][WiFiUdp.cpp:219] parsePacket(): could not receive data: 9

Should we track this bug, can somebody check that to?

Thanks

lumapu, 2022-10-26:
platform ist ESP32 oder, da oben ESP8266 steht

DanielR92, 2022-10-27:
Hups, ja ESP32

DanielR92, 2022-11-16:
Kann geschlossen werden. Passt nun alles wieder... :)

closed 2022-11-16 by DanielR92

stefan123t, 2022-11-17:
@DanielR92 was war denn das Problem, damit wir was draus lernen können ?