lumapu/ahoy #376: [ESP8266] No connection to HM1500

mipek, 2022-10-27:
### Platform

ESP8266

### Model name

Wemos D1 mini and nRF24L01+

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO5 / D1    |
| Pin 4 CSN     | GPIO15 / D8    |
| Pin 5 SCK     | GPIO14 / D5 SCLK  |
| Pin 6 MOSI    | GPIO13 / D7 MOSI  |
| Pin 7 MISO    | GPIO12 / D6 MISO  |
| Pin 8 IRQ     | GPIO4 / D2    |

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.25

### Github Hash

f33cd93

### Build & Flash Method

Platform IO (build & flash)

### Desktop

Linux

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### Inverter
#### Inverter 0
    - Address: 11618110xxxx
    - Name: HM-1500
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 300, 300, 300, 300
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D1 (GPIO5)
    - IRQ: D2 (GPIO4)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
11:00:43.854 > ----------------------------------------
11:00:43.857 > I: Welcome to AHOY!
11:00:43.860 > I: 
11:00:43.860 > point your browser to http://192.168.0.33
11:00:43.863 > I: to configure your device
11:00:43.866 > I: ----------------------------------------
11:00:43.870 > 
11:00:43.870 > I: RF24 Amp Pwr: RF24_PA_I: MIN
11:00:43.873 > I: Radio Config:
11:00:43.876 > SPI Frequency            = 1 Mhz
11:00:43.879 > Channel                  = 3 (~ 2403 MHz)
11:00:43.881 > Model                    = nRF24L01+
11:00:43.881 > RF Data Rate             = 250 KBPS
11:00:43.884 > RF Power Amplifier       = PA_MIN
11:00:43.886 > RF Low Noise Amplifier   = Enabled
11:00:43.890 > CRC Length               = 16 bits
11:00:43.892 > Address Length           = 5 bytes
11:00:43.895 > Static Payload Length    = 32 bytes
11:00:43.898 > Auto Retry Delay = 250 microseconds
11:00:43.901 > Auto Retry Attempts      = 0 maximum
11:00:43.903 > Packets lost on
11:00:43.903 >     current channel      = 0
11:00:43.906 > Retry attempts made for
11:00:43.909 >     last transmission    = 0
11:00:43.912 > Multicast                = Disabled
11:00:43.912 > Custom ACK Payload       = Disabled
11:00:43.915 > Dynamic Payloads = Enabled
11:00:43.917 > Auto Acknowledgment      = Disabled
11:00:43.920 > Primary Mode             = RX
11:00:43.923 > TX address               = 0xdeadbeef01
11:00:43.926 > pipe 0 (closed) bound    = 0xdeadbeef01
11:00:43.929 > pipe 1 ( open ) bound    = 0x707xxxxx01
11:00:43.932 > pipe 2 (closed) bound    = 0xc3
11:00:43.935 > pipe 3 (closed) bound    = 0xc4
11:00:43.937 > pipe 4 (closed) bound    = 0xc5
11:00:43.940 > pipe 5 (closed) bound    = 0xc6
11:00:49.992 > I: [NTP]: 2022-10-27 09:00:43 UTC
11:00:49.995 > I: enqueued cmd failed/timeout
11:00:49.998 > I: Inverter #0 I: no Payload received! (retransmits: 0)
11:00:50.000 > I: resetPayload: id: 0
11:00:50.003 > I: Requesting Inverter SN 11618110xxxx
11:00:50.006 > I: enqueuedCmd: 11
11:00:50.008 > I: enqueuedCmd: 1
11:00:50.012 > I: enqueuedCmd: 5
11:00:50.012 > I: sendTimePacket
11:00:50.014 > I: TX 27B Ch40 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:00:51.608 > W: while retrieving data: last frame missing: Request Retransmit
11:00:51.613 > I: sendTimePacket
11:00:51.613 > I: TX 27B Ch61 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:00:53.703 > W: while retrieving data: last frame missing: Request Retransmit
11:00:53.709 > I: sendTimePacket
11:00:53.709 > I: TX 27B Ch75 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:00:55.799 > W: while retrieving data: last frame missing: Request Retransmit
11:00:55.805 > I: sendTimePacket
11:00:55.805 > I: TX 27B Ch3 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:00:57.894 > W: while retrieving data: last frame missing: Request Retransmit
11:00:57.899 > I: sendTimePacket
11:00:57.899 > I: TX 27B Ch23 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:00:59.990 > W: while retrieving data: last frame missing: Request Retransmit
11:00:59.995 > I: sendTimePacket
11:00:59.995 > I: TX 27B Ch40 | 15 81 10 04 44 84 36 72 70 80 0B 00 63 5A 48 BB 00 00 00 00 00 00 00 00 5E C0 AB 
11:01:20.095 > I: enqueued cmd failed/timeout
11:01:20.098 > I: Inverter #0 I: no Payload received! (retransmits: 5)
11:01:20.101 > I: resetPayload: id: 0
11:01:20.103 > I: Requesting Inverter SN 11618110xxxx
11:01:20.106 > I: sendTimePacket
11:01:20.109 > I: TX 27B Ch61 | 15 81 10 04 44 84 36 72 70 80 01 00 63 5A 48 E3 00 00 00 00 00 00 00 00 97 51 A1 
11:01:21.703 > W: while retrieving data: last frame missing: Request Retransmit
11:01:21.708 > I: sendTimePacket
11:01:21.708 > I: TX 27B Ch75 | 15 81 10 04 44 84 36 72 70 80 01 00 63 5A 48 E3 00 00 00 00 00 00 00 00 97 51 A1
```

### Error description

I never receive a response from the HM1500 (no RX as seen in the serial log). I've tested different distance between the Inverter and Ahoy (from up to 10m to as close as about 1m).

I started with the stock wiring/settings but changed it because it didn't work and I've read there are some problems with IRQ on GPIO0 with D1 mini.

Maybe I've a faulty nRF24, is there any way I can check?

Thanks!

RaveRebel, 2022-10-27:
Hi, i update to 0.5.26 and the communication to HM-1500 is still ok. Maybe the connection problem is based on your hardware. 
@mipek: Is it possible to connect with another Version?

lumapu, 2022-10-27:
These are my settings on a D1 mini. Today I changed the direction of the antenna and increased the success rate incredible.

[image]

mipek, 2022-10-27:
@RaveRebel I've tried with the latest version from the main branch but there' was difference. 

I had the Ahoy-DTU running the entire day and no frame was ever received. It's really weird, the interrupt is never triggered.
I've ordered another nRF24L01+ maybe the one I'm using is indeed faulty.

RaveRebel, 2022-10-27:
@mipek I have built 7 AHOY-DTU so far and all with this configuration.  Try the standard again.
[image]

mipek, 2022-10-27:
Yeah I'll try the default configuration again. I've another Ahoy-DTU with a D1 mini that's using the standard configuration and it's working fine (using a older version of ahoy but I don't think that matters).

RaveRebel, 2022-10-27:
If you have a working nRF24L01+ go back to standard configuration and put it on the not working DTU. So you can check the nRF24L01+. Or put the faulty nRF24L01+ on the working DTU.
 It's always good to have multiple DTUs.

mipek, 2022-10-31:
It's working fine with the other nRF24L01+ so my other module was probably faulty. I'll play around with my faulty module a little bit, maybe it's possible to check if they're not working correctly and show a warning.

Thanks everyone!

closed 2022-10-31 by mipek