lumapu/ahoy #377: Auto Neustart in der Nacht

DennisXK, 2022-10-27:
Ist es möglich in dem nächsten update, ein Autorestart in der Nacht machen zu lassen? Habe festgestellt, dass sich der ESP ab und zu wenn auch selten aufhängt und keinen Daten mehr sendet. Er ist dann auch nichtmehr im Netzwerk verfügbar.

Da ich die Werte im IoBroker mit logge, ist das natürlich blöd!

Mir ist auch aufgefallen das die Uptime auf der Startseite unregelmässig bei 0 Days wieder anfängt. Was soll mir das sagen?

turbolift, 2022-10-27:
Ich habe das gleiche Problem ... Die Uptime ist mit jeden Release schlechter geworden. Im Vergleich dazu läuft OpenDTU bislang stabil 4+ Tage. Ich bin deshalb gewechselt, weil sich ahoy manchmal komplett aufhängt und ich dann die Stromversorgung kappen muss. Ich gebe aber ahoy noch nicht auf, warte die nächsten releases ab und versuchte es dann noch einmal. Das was hier geleistet wird ist auf jeden Fall super.

lumapu, 2022-10-27:
Hallo zusammen,
generell bin ich gegen Neustarts da ich gerne eine unendliche Laufzeit haben würde.
Könntet ihr bitte die Version dazu schreiben?

Wir arbeiten quasi seit Beginn an der Stabilität und sind noch nicht da wo wir hin wollen. Es gibt auch noch verschiedene Ideen, die noch nicht getestet wurden, um die Stabilität weiter zu verbessern.

Wesentlich zur Stabilität trägt eine externe Powerversorgung, Kondensatoren am NRF Modul sowie ein kleines Amplifier Power level.

Der ESP32 (und damit OpenDTU hat den entscheidenden Vorteil von zwei Cores in Gegensatz zum ESP8266), was die bessere Stabilität erklären lässt. Ahoy ist übrigens auch auf den ESP32 lauffähig, ist aber nicht unser Hauptfokusm

Chrischi-, 2022-10-27:
@DennisXK @turbolift 

Warum behelft ihr euch nicht einfach, solange es keine Lösung bzw. die gewünschte Stabilität seitens Ahoy gibt mittels

`http://hier.ip.eures.ahoy/reboot`

also im Grunde mittels `/reboot`

Dies alles natürlich in der Annahme, dass ihr sowieso noch irgendeinen rPi oder ein System wie HomeAssistant etc. pp am Laufen habt. Dann dort einfach jede Nacht um Uhrzeit x oder alle 12h den Reboot triggern.

lumapu, 2022-10-27:
@Chrischi- sehr guter Ansatz für diesen individuellen Workaround

DennisXK, 2022-10-29:
ja die idee wäre für mich ok, da ich den iobroker laufen habe.

Also einen elko hab ich am nrf, aber der ist ja nicht entscheidet das sich der Esp aufhängt!?

Und habe die Version 5.17 am laufen

reserve85, 2022-10-29:
Die 0.5.17 hängt sich bei mir auch alle paar Tage mal auf. Ich habe das mit einem Reset um 00:00 Uhr gelöst wie @Chrischi- vorgeschlagen hat (mit einem raspi). Das ist für mich ausreichend! Seit zwei Tagen habe ich testweise die 0.5.26 laufen und bisher ist kein Absturz aufgetreten.

closed 2022-10-31 by lumapu