lumapu/ahoy #380: Leistungsbegrenzung in Prozent

Matze7779, 2022-11-02:
Bei der 0.5.28 funktioniert die Leistungsbegrenzung in Prozent nicht mehr.

Ahoy 0.5.28
HM 1200
Unter "Serial Console"

Angaben in Watt werden korrekt angenommen. In Prozent nicht.

reserve85, 2022-11-02:
kann ich bestätigen, mit einem HM-1500 das gleiche. Es funktioniert nur die Einstellung in "Watt".

DD0CW, 2022-11-02:
stimmt, und im Text "Every xx seconds the values are updated" wird der eingegebene Wert ( z.B. 20 Sekunden ) zwar entspr. ausgeführt aber immer nur xx = 30 angezeigt.

DanielR92, 2022-11-04:
PullRequest ist raus. Bug für das setzen eines Wertes in % wurde behoben. Danke.

lumapu, 2022-11-05:
@Matze7779 @reserve85 bitte um Feedback vom test mit aktueller Development Version

reserve85, 2022-11-06:
funktioniert bei mir (Watt + %), vielen Dank!

closed 2022-11-06 by lumapu