lumapu/ahoy #382: Bekomme keine Verbindung zum HM400 Wechselrichter

ThomasKlaus-ai, 2022-11-03:
Hallo an die Runde,
folgende Info sehe ich in Putty:

 Requesting Inverter SN 112183320731
I: Devcontrol request 11 power limit 100
I: TX 19B Ch75 | 51 83 32 07 31 78 56 34 12 80 0B 00 03 E8 01 00 80 80 BF
I: enqueuedCmd: 5

Kann mir Jemand eine Info geben, was ich jetzt machen kann ?

lumapu, 2022-11-03:
Du hast scheinbar keinen Empfang von Wechselrichter.
Bitte prüfen/testen:
- Pinout, speziell IRQ (mit Settings vergleichen)
- anderen Pin für IRQ verwenden
- Ausrichtung vom NRF zum Wechselrichter ändern (in allen Dimensionen)
- Amplifier Power auf low stellen
- Serial number überprüfen
- nicht gleich auf Power Limit gehen, erst mal Grundfunktionen testen
- Stromversorgung prüfen, evtl durch Elko nahe am NRF verbessern

Ansonsten mehr Infos, Version, Aufbau, Entfernung, gemachte Schritte mitteilen.

Bitte auch die FAQ und User Manual durchlesen, da stehen diese Schritte auch drin.

h1ghrise, 2022-11-05:
Hi,
ich hatte das gleiche Problem mit meinem HM-800. Bei mir lag es wohl an einem Problem mit dem Chip Select Pin (CS), welcher Initial auf *D4* liegt (Wemos D1 mini). Auf dem GUI sah alles gut aus, aber ich habe keinerlei Daten vom WR empfangen. Nachdem ich den Pin auf D8 gelegt habe funktioniert alles problemlos.

stefan123t, 2022-11-06:
@h1ghrise danke für die Rückmeldung! Wemos D1 mini macht immer mal wieder Probleme bei der Default Pin Belegung. IdR. ließ sich das durch umstecken des IRQ von D3 auf D1/D2 lösen. Daher ist die Info mit CS von D4 -> D8 neu. Kannst Du feststellen welche Version des Wemos D1 mini Du hast: v1/v3 ? 

@ThomasKlaus-ai hast Du die von @lumapu vorgeschlagenen Punkte geklärt. Bitte fülle unser Issue Template vollständig aus, damit wir nicht alle Details einzeln abfragen müssen. Danke!

roku133, 2022-11-07:
Bei meinem Wemos D1 mini v3 funktioniert es mit folgender Einstellung:
CS -> D8
CE -> D3
IRQ -> D4
CE und IRQ müssen bei mir im Vergleich zur Default-Konfiguration vertauscht werden.
Die Kommunikation mit meinem HM-400 läuft damit dann über viele Tage (bis zum nächsten Versions-Upgrade) sehr stabil.

stefan123t, 2022-11-07:
@roku133 vielen Dank für die positive Rückmeldung!
Der Original Poster @ThomasKlaus-ai hat sich ja leider verabschiedet, daher würde ich das Issue damit zu machen.

closed 2022-11-07 by stefan123t