lumapu/ahoy #384: Werte bitte auf 2 Dezimalstellen begrenzen

Waldo56, 2022-11-03:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.28

### Github Hash

2e08ee0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

[image]

lumapu, 2022-11-03:
Ist bekannt, danke für den issue hier als Reminder

lumapu, 2022-11-05:
@Waldo56 bitte nochmal mit aktueller Develoment version testen

heldel56, 2022-11-12:
Gibt es denn eine Lösung zur Begrenzung der Dezimalstellen? Wenn JA , welche Einstellungen müssen geändert werden ? 
Danke im Voraus für ein Feedback .

Waldo56, 2022-11-13:
> ### Platform
> ESP8266
>
> ### Model name
> _No response_
>
> ### nRF24L01+ Module
> _No response_
>
> ### Antenna
> circuit board
>
> ### Power Stabilization
> ~100uF Elko
>
> ### Connection diagram
> ## Connection diagram I [...]

Waldo56, 2022-11-13:
Ok mit Version 0.5.32.und.0.5.35

lumapu, 2022-11-13:
hmm, dann starte ich demnächst einen neuen Versuch 🙈

Waldo56, 2022-11-13:
Mit ok meinte ich, dass der Fehler bisher nicht mehr aufgetreten ist.

stefan123t, 2022-11-14:
Problem laut @Waldo56 in 0.5.35 gelöst. Danke für die Rückmeldung

commit 74f4a114 (2022-11-04): improved GUI password handling
fix serial console uptime
fix #384 round total values
closed 2022-11-14 by stefan123t