lumapu/ahoy #386: Feature Request: Broadcast / Suche nach Invertern

FFW-Scripter, 2022-11-04:
**Problem:**

Viele wollen ahoy nutzen und haben Ihren Inverter schon verbaut, oftmals auch an Stellen wo man nicht leicht rankommt um die Seriennummer abzuschreiben oder abzufotografieren. Teilweise ist es auch so, das die Aufkleber durch verwitterung nicht mehr richtig lesbar sind und/oder nur Teile der Seriennummer verfügbar sind.

**Lösungsansätze:**

Daniel92 und Ich hatten schon einmal über einen Auto-Discovery Modus nachgedacht. Man könnte anhand einer Range einfach die Limit-Abfrage oder ähnliches nutzen und schauen ob ein Inverter darauf anspringt. Diese Range durchzugehen kann bzw. wird mitunter sehr lange dauern. Daher kam die Idee das man auch Platzhalter im Falle von "beschädigten" Seriennummer verwenden könnte und somit die Range einschränkt.

Vorstellen könnte ich mir das in etwa so: 1161123**67*1

Für die Voll-Suche müsste dann leider 1161******** abgesucht werden - dauert sicherlich, würde aber manchen das abmontieren des Inverters sparen.

Ein zweiter vielleicht effektiverer Ansatz von stefant wäre der Broadcast 0x02 von Hoymiles selbst:
https://github.com/stefan123t/ahoy/wiki/FAQ-Frequently-Asked-Questions#broadcast-0x02

Dazu müsste wohl aber die Baudrate geändert werden.

Danke noch einmal an dieses Coole Projekt :+1: 
PS: Über eine parallele Implementierung in openDTU würde ich mich freuen :-)

reserve85, 2022-11-04:
ich wäre eigentlich froh wenn es sowas nicht gibt. Dann verstellt der nette Nachbar nämlich nicht den Inverter anderer. Ist eh schon unsicher genug.

DanielR92, 2022-11-04:
Ich könnte mir dies sehr gut Vorstellen. Jedoch mit einer Einschränkung, das die Abfrage nur im Low-Modus möglich ist. 
Um die Reichweite ein zu krenzen.

Wenn jedoch ein Broadcast möglich ist (ohne BruteForce Attacke), wäre es natührlich angenehmer.

lumapu, 2022-11-04:
wäre allerdings ein Sicherheitsmanko der Hoymiles Wechselrichter oder? Ich könnte ja dann theoretisch dem Nachbarn sein WR ausschalten

DanielR92, 2022-11-16:
Ja da gebe ich dir recht.

stefan123t, 2022-11-17:
Ja die Hoymiles WR sind jetzt nicht rock solid was die Sicherheit angeht. So was wie Passwörter die nur auf der DTU selbst abgefragt werden und beim WR einfach mit 0xFFFFFFFF überschrieben werden können ist jetzt nicht gerade Vertrauen erweckend.
Aber bisher wollte auch noch niemand seinen WR mit Lock/Unlock Passwort Codes versehen bzw aus Versehen bricken.
Davon abgesehen könnten wir die ersten 5-7 Ziffern der WR Serial ID aus dem Modell, Baujahr und KW relativ gut erraten. Es blieben pro KW noch 99999 Seriennummern zu prüfen. Jede Seriennummer dauert ca. 1-3 Sekunden das wären dann pro KW 27.7-83.3 Stunden. Also für alle 52 KW eines Baujahrs 60-180 Tage.

Also mE wäre es sinnvoller und zielführender das BROADCAST 0x02 Kommando zu implementieren. Wie gesagt könnte es sein dass die RX pupe irgendwie auf 0x010LL gesetzt wird bzw mit 2MBps eine höhere Datenrate bzw niedrigere Reichweite für diesen Einsatzzweck von der DTU verwendet wird.
Müsste man mal im Source Code nachlesen...

stefan123t, 2023-01-15:
Siehe auch #301

Thomas-PV-Tryer, 2023-04-01:
Schade - ich hatte das gleiche "Urproblem nicht lesbare Seriennummer"! Habe mich dann an die Hoymiles Hotline gewendet - dort gibt es ja die Scanfunktion und die Hotline verweist auch drauf.
Wäre toll, wenn sie doch jemand tragen und dann implementieren könnte ...
Mir sind die Hoymiles DTU's zu teuer ....

stefan123t, 2024-10-27:
@nivadis @broth-itk könnte einer von Euch beiden das mal Tracen ?
Gibt es die Option in der S-Miles Cloud tatsächlich noch für HM / MI Wechselrichter ?

broth-itk, 2024-10-29:
Habe gerade in der S-Miles Cloud nachgesehen, finde diese Funktion aber nicht auf die Schnelle.
Wo soll sich diese denn verstecken?
Tracen wäre kein Problem...

stefan123t, 2024-10-29:
@broth-itk ich habe mal in einer alten Doku nachgesehen, dort wird es wie folgt beschrieben:

1. Click O&M and enter Micro Toolkit.
2. Click “Power generation and control” button
3. Press the “Add Device” button to add the microinverter to the list. (The microinverter added here
is only used for on-site debugging, and it will not be uploaded to the server, nor can it replace the
power station creation on S-Miles Cloud.)
4. You can click “Automatic Search” to add microinverter, or you can type in / scan the microinverter
ID.

Das ist also offenbar eine Funktion der S-Miles Installer App vermutlich Zusammenspiel mit einer DTU Pro / Lite.

| (1) | (2) | (3) | (4) |
| --- | --- | --- | --- |
| [image] | 
[image] | 
[image] | 
[image] |

Die o.a. Dokumentation heißt bei mir User-Manual_DTU-Lite-S_Global_EN_V202202.pdf Seiten 11-14.

broth-itk, 2024-10-29:
An die App habe ich nicht gedacht - wird heute Abend mal ausprobiert und mitgeschnitten.

broth-itk, 2024-10-29:
Die App zeigt mir partout kein "Auto Search" Button in der Geräteliste an.
Es macht keinen Unterschied ob die Liste leer ist oder nicht.
Manuell per Eingabe der Seriennummer oder durch Scan eines QR Codes wäre möglich.

stefan123t, 2024-10-30:
Okay, vielen Dank fürs Ausprobieren!
Dann müssen wir den Feature Request für das BROADCAST 0x02 wohl als Can't Reproduce / Won't fix schließen.

broth-itk, 2024-10-31:
@stefan123t Ich habe eine DTU Pro S und drei HMS Wechselrichter.

stefan123t, 2024-11-02:
@broth-itk danke, ich weiß nicht ob es das Broadcast 0x02 bei einer DTU Pro/Lite S bzw. HMS/HMT-Modell mit CMT2300A überhaupt noch gibt / braucht. Ich habe es aber auch in der aktuellen S-Miles Installer App nicht mehr gefunden.

Evtl. könnte das aber jemand mit einer alten DTU Pro / WLite und einem HM- / MI-Modell nochmal überprüfen...