lumapu/ahoy #389: Verbindungsversuche zu YOUR_WIFI_SSID

HFarryranz, 2022-11-04:
### Platform

ESP32

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.31

### Github Hash

921419d

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

 ohne ausgeführtes  Setup

### Error description

 Ein frisch geflashter Chip ohne Config öffnet für 60 Sekunden sein WLAN AHOY-DTU.  Falls sich kein Client verbindet wird versucht sich mit dem WLAN  "YOUR_WIFI_SSID"  zu verbinden.
Nach dem die Verbindung scheitert, steht dann erst wieder nach x Sekunden das WLAN AHOY-DTU wieder zur Verfügung.

Ein Verbindungsversuch zu diesem WLAN YOUR_WIFI_SSID sollte nicht durchgeführt werden.

lumapu, 2022-11-04:
Die Anschauung ist nachvollziehbar, allerdings sieht die Implementierung vor, dass man sein eigenes WLAN hart codiert.
Dadurch wird es etwas schwieriger wenn man das unterbinden will, und vor allem erschließt sich mir der Vorteil nicht.

HFarryranz, 2022-11-04:
Ok nicht bedacht, dass das WLAN hart codiert sein kann, Aber dennoch könnte man definieren, dass wenn es YOUR_WIFI_SSID lautet, er die Verbindung nicht versuchen soll. Ok es könnte User geben, deren WIFI SSID wirklich so lautet...

lumapu, 2022-11-05:
habs jetzt lösen können, insgesamt war es eine gute Idee hier reinzuschauen.

stefan123t, 2022-11-17:
@lumapu mE sollte YOUR_WIFI_SSID und YOUR_WIFI_PASSWORD nicht im Code gesetzt werden. Lediglich ein Beispiel in der example_config_override.h und die Angaben als placeholder attribute der (beiden) Input Felder sollten mE genügen.

DanielR92, 2022-11-18:
Soweit ich weis kann man die WiFi klasse schon dazu bringen nur ein AP auf zu machen... und erst wenn die config hinterlegt ist, als normaler Wifi CLient zu fungieren. 

Sobald die Config fehl schlägt (corrupt), oder nach 3x Verbindungsversuch kein WLAN verbindung aufbauen konnte, sollte dann ein AP aufgemacht werden.  - Sollte umsetzbar sein.

commit ea991a77 (2022-11-05): fix #389 prevent connect to default Wifi 'YOUR_WIFI_SSID'
implemented #388 add inverter Id to the front of serial print
fix inverter serial number length in input field (api.js)
closed 2022-11-19 by lumapu