lumapu/ahoy #390: ESP32 Sunrise / sunset and Serial Console  Interval not saved after reboot

Kouichi-kun, 2022-11-05:
### Platform

ESP32

### nRF24L01+ Module

nRF24L01+ plus, square dot

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.31

### Github Hash

921419d

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 114174607XXX
    - Name: HM-600-Links
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 365, 365
    - Module Name: links, rechts
#### Inverter 1
    - Address: 114174610178XXX
    - Name: HM-600-Rechts
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 365, 365
    - Module Name: links, rechts

### General
    - Interval [s]: 60
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP:  192.168.xxx.xxx
    - Port: 1883
    - Username (optional): xyz
    - Password (optional): password
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: GPIO5
    - CE: GPIO4
    - IRQ: GPIO1
#### Radio (NRF24L01+)
    - Amplifier Power Level: MIN
#### Serial Console
    - print inverter data: []
    - Serial Debug: []
    - Interval [s]: 10

- Reboot device after successful save: [x]
- SAVE

### Error description

System does not save config for sunrise and sunset and the Serial Console Interval [s] after a reboot.

lumapu, 2022-11-05:
issue detected in code, will be fixed with next commit and version greater than `0.5.31`

Kouichi-kun, 2022-11-05:
super, danke :) ich mach mal zu

closed 2022-11-05 by Kouichi-kun

homeautomation2022, 2022-11-05:
Die beiden Koordinaten müssen gleichzeitig gesetzt werden, ging eigentlich mal.
Nur dann kann man auch die Inv Kommunikation nachts deaktivieren, denn nur so kann er Sonnenauf- und untergang kennen.

Kouichi-kun, 2022-11-05:
Ich setz die immer zusammen und mach den Haken rein. Also ich schließe einen Bedienerfehler aus. Ging ja mal mit der 0.5.28 und glaub noch mit der 0.5.29

commit 6fc5fb75 (2022-11-05): fix #390 geo location was not saved - wrong length of config struct was used while storing values