lumapu/ahoy #391: Ahoy Version 0.5.28 -> Different Info between GUI and MQTT -> available and (not) producing

haselchen, 2022-11-05:
### Platform

ESP8266

### Model name

Wemos D1 mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

board

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.28

### Github Hash

2e08ee0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141XXXXXXXXX
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: Panel 1, Panel 2
### General
    - Interval [s]: 20
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

### Error description

This information is displayed in the Ahoy GUI:

Inverter #0: HM-600 (v10010) is available and is producing

In MQTT in Iobroker is displayed this:

[image]

lumapu, 2022-11-05:
@haselchen bitte nochmal mit der aktuellen Developmentversion `0.5.32` testen

haselchen, 2022-11-06:
> @haselchen bitte nochmal mit der aktuellen Developmentversion `0.5.32` testen

a)  Wo finde ich die
b) kann ich die OTA einspielen

Wäre ein Riesenaufwand , das Konstrukt abzubauen zum Flashen.

lumapu, 2022-11-06:
ok, das ist den Aufwand aktuell nicht wert. Warte best auf das nächste Release, ich hoffe ich kann es bald veröffentlichen.

Um deine Fragen zu beantworten:

- zu finden unter Actions hier in GitHub
- ja OTA geht, nur Vorsicht: das Memorylayout hat sich verändert, daher müssen nach dem Update alle Daten neu eingegeben werden

roku133, 2022-11-08:
Discrepancy still exists in 0.5.32, GUI status is correct:

GUI:
... is available and is producing
MQTT:
available_text, ... (27 bytes): available and not producing
available, ... (1 bytes): 2

roku133, 2022-11-08:
Further conditions in addition to my comment above:

Case 2
GUI:
... is available and is not producing
MQTT:
available_text, ... (27 bytes): available and not producing
available, ... (1 bytes): 1
-->  ok

Case 3
GUI:
 ... is not yet available
MQTT:
no information except uptime
--> available_text, available intentionally skipped?

stefan123t, 2022-11-09:
@lumapu we are building the UI answer using several String concatenations and a complex conditional structure.
Wouldn't it be easier to simply declare the Strings for `MQTT_STATUS_AVAIL_PROD`, etc. as an `const enum char[]` / `String` in Flashmem and use the index from `available` to translate all responses `available_text` and UI ?
That could go into a common outputLibrary which can be called by both outputMQTT and outputJSON by he REST API.

lumapu, 2022-11-09:
maybe that's a good approach to share strings between MQTT and JSON.

roku133, 2022-11-14:
Inconsistency between GUI and MQTT now fixed in 0.5.35, 
no MQTT messages during unavailability of inverter, already addressed in #416.

stefan123t, 2022-11-15:
@haselchen would you mind to close the issue or does @lumapu want to reference and close this with the (hopefully) next release ?

lumapu, 2022-11-15:
I will close it with the next release, otherwise we will have duplicates here

haselchen, 2022-11-18:
> @haselchen would you mind to close the issue or does @lumapu want to reference and close this with the (hopefully) next release ?

Ich warte auf das nächste stable Release und , wenn ok, dann kann hier zu.

commit c467eead (2022-11-05): potential fix #391 MQTT avail and producing info inconsistent compared to web GUI
closed 2022-11-19 by lumapu