lumapu/ahoy #393: MQTT totals are occasionally corrupted and report twice the actual value

Argafal, 2022-11-05:
I run ahoy [Git SHA: 2e08ee0 :: 0.5.28](https://github.com/lumapu/ahoy/commits/2e08ee0) with three inverters: Two HM-1500, one HM-600. 

The total reported on MQTT is occasionally corrupted: I expect the reported total to be around 2600 kWh every time. Every now and then exactly twice that value sneaks into the MQTT messages. This is also true for the other total values reported, see below for a snapshot from a mqtt listener.

In the live view on the webserver I seem to always see the correct total value, never the double.

```
~ mosquitto_sub -h localhost -v -t 'inverter/total/#'
inverter/total/P_AC 2.900
inverter/total/YieldTotal 2672.321
inverter/total/YieldDay 4310.000
inverter/total/P_DC 6.800
inverter/total/P_AC 5.800
inverter/total/YieldTotal 5344.642
inverter/total/YieldDay 8620.000
inverter/total/P_DC 13.600

```

lumapu, 2022-11-05:
thank you for this bug report. I looked to the code and understand the problem.
There is a queue of MQTT messages to send out. Sometimes there are more than one message in queue, but the total values are only reset in the beginning of the function. I'll try to fix that and come back to you.

lumapu, 2022-11-05:
@Argafal can you test the new debug firmware `0.5.32`?

Argafal, 2022-11-06:
No more doubles in 0.5.32. Fabulous, thanks for your quick reaction!

commit 1620b633 (2022-11-05): potential fix #393 corrupted MQTT total values
closed 2022-11-06 by lumapu