lumapu/ahoy #394: Limiter per Rest-API

gaborfri, 2022-11-05:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: latest
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

ich habe den Ahoy erfolgreich mit dem Hoymiles HM300 verbunden. Die Daten werden korrekt ausgelesen. Nun möchte ich diesen als Einspeisewechselrichter verwenden. Das manuelle Eintragen des Limits von z.B. 100w hat funktioniert und es wird auch nicht mehr Leistung aus dem Akku gezogen.
Nun wollte ich mich an die Steuerung machen um das Limit dynamisch an den aktuellen Verbrauch anzupassen.  Leider funktioniert mein Testaufruf nicht wie erwarten:

ich habe mir das Bsp aus der Doku genommen und ausgeführt.

POST http://192.168.xxx.xxx/api
content-type: application/json

{
    "inverter":0,
    "tx_request": 81,
    "cmd": 11,
    "payload": 200,
    "payload2": 0
}

das Ergebnis ist ein 200 mit Success OK, aber parallel einer Meldung mit wrong Path. Ich habe nun schon einiges versucht. Bislang allerdings ohne Erfolg.
Irgendwas scheint nicht zu passen, ich finde aber einfach nicht meinen DenkFehler. 
Muss anstelle des /api-Endpunktes /api/setup aufgerufen werden?

lumapu, 2022-11-05:
es könnte ab der geänderten API liegen. Wir gingen mit der Dokumentation noch nach.
Der korrekte Endpunkt für deine Anfrage müsste `/api1` sein.

gaborfri, 2022-11-22:
ich habe eine Lösung gefunden. Der korrekte API-Endpunkt lautet [ip]/api/ctrl
und der Request muss wie folgt aussehen:

{
    "inverter":0,
    "tx_request": 81,
    "cmd": 11,
    "payload": [50, 1]
}

stefan123t, 2022-11-23:
@DanielR92 wir sollten mE die Doku in User_Manual.md anpassen. Danke!

lumapu, 2022-11-24:
der gestrige Commit ändert hier die commands.
Ich habe alles im Usermanual dokumentiert. Jetzt kann man über nahezu identische (selbstsprechende) Commands über MQTT und REST API die DTU steuern

closed 2022-12-30 by lumapu