lumapu/ahoy #395: [ESP8266] MQTT packets not sent

h1ghrise, 2022-11-06:
### Platform

ESP8266

### Model name

Wemos D1 Mini

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_4 (D1) CE  *default* |
| Pin 4 CSN     | GPIO_15 (D8) CS   |
| Pin 5 SCK     | GPIO14/18 SCLK  *default*|
| Pin 6 MOSI    | GPIO13/23 MOSI  *default*|
| Pin 7 MISO    | GPIO12/19 MISO  *default*|
| Pin 8 IRQ     | GPIO_5 (D1) IRQ   |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.28

### Github Hash

2e08ee0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Mac OS

### Setup

### Device Host Name
    - Device Name: IOT-SolarPanel
### WiFi
    - SSID: IoT
    - Password: [*redacted*]
### Inverter
#### Inverter 0
    - Address: 11418***
    - Name: Balkonkraftwerk
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 405, 405
    - Module Name: Panel 1, Panel 2
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 10.10.0.2
    - Port: 1883
    - Username (optional): mqtt
    - Password (optional): *redacted*
    - Topic: solar
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D2 (GPIO4)
    - IRQ: D1 (GPIO5)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
- no MQTT specific serial output
Status Page
WARN: MQTT is not connected
```

### Error description

1) Go to http://10.10.20.2/setup
2) configure above settings
3) Reboot

**ER (Expected result)**:
- MQTT connects to broker.
- MQTT packets are send to Home assistant instance

**AR (Actual result)**:
- no MQTT packets are leaving the system

** Network Info **:
- Ahoy DTU is located in a seperate Wifi Network (`IoT`) which is tagged as VLAN 20 (Interface `eth1.20`)
- MQTT Broker is located in `VLAN 1` via LAN
- Routing is performed via an OpenWRT box.
- `IoT` VLAN and `LAN` VLAN are separate firewall zones
- Firewall rules explicitly allow traffic passing from IoT VLAN to LAN VLAN on tcp/1883 (mqtt)

** Troubleshooting**:
- ran tcpdump from OpenWRT on IoT Interface (where traffic originates)
`tcpdump -i eth1.20 host 10.10.20.2`:
```tcpdump: listening on eth1.20, link-type EN10MB (Ethernet), capture size 262144 bytes
<no packets captured>
```
- tried to send discovery config (HA)
- no packets captured
- Broker logs also do not indicate any packets reaching HA.

I have other MQTT devices in that VLAN, which are communicating fine.
Do I miss something?

h1ghrise, 2022-11-06:
Solved: i was too stupid to tick the box "reboot after successful save"
Sorry for the inconvenience caused.

stefan123t, 2022-11-17:
@lumapu as long as we require a reboot to load the config anew in AhoyDTU we should flag the [x] reboot after save as default IMHO.

closed 2022-12-30 by lumapu