lumapu/ahoy #396: MQTT totals should only be reported once all inverters are available

Argafal, 2022-11-06:
Running 0.5.32, 64fb587cf3042a02422028fdf4d9d4abb6db702b, with three inverters.

This issue concerns the reported totals on MQTT: Totals are reported before all inverters are available. This happens at start up as the various inverters start replying. See below for a snapshot. 

I propose that totals on MQTT should only be sent once all inverters are available. MQTT listeners such as domoticz expect strictly increasing counters and have no way of knowing that the total is based on an incomplete number of inverters - this causes various issue with the downstream statistics.

`$ mosquitto_sub -h localhost -v -t 'inverter/total/#'

inverter/total/P_AC 104.000
inverter/total/YieldTotal 1056.386
inverter/total/YieldDay 175.000
inverter/total/P_DC 109.400

[a minute later the second inverter comes online]

inverter/total/P_AC 147.700
inverter/total/YieldTotal 1635.361
inverter/total/YieldDay 250.000
inverter/total/P_DC 155.200

[a few minutes, now the third and final inverter comes online. This means the first actually correct total is the one below.]

inverter/total/P_AC 308.700
inverter/total/YieldTotal 2672.769
inverter/total/YieldDay 448.000
inverter/total/P_DC 324.700
`

Argafal, 2022-11-06:
Proposed patch below. Seems to resolve the issue for me.

```
diff --git a/tools/esp8266/app.cpp b/tools/esp8266/app.cpp
index 52f22f4..87bd0e8 100644
--- a/tools/esp8266/app.cpp
+++ b/tools/esp8266/app.cpp
@@ -585,6 +585,7 @@ void app::sendMqtt(void) {
     char topic[32 + MAX_NAME_LENGTH], val[32];
     float total[4];
     bool sendTotal = false;
+       bool totalIncomplete = false;
     snprintf(val, 32, "%ld", millis() / 1000);
 
     mMqtt.sendMsg("uptime", val);
@@ -604,8 +605,10 @@ void app::sendMqtt(void) {
             if(mMqttSendList.front() == RealTimeRunData_Debug) {
                 // inverter status
                 uint8_t status = MQTT_STATUS_AVAIL_PROD;
-                if (!iv->isAvailable(mUtcTimestamp, rec))
+                if (!iv->isAvailable(mUtcTimestamp, rec)) {
                     status = MQTT_STATUS_NOT_AVAIL_NOT_PROD;
+                                       totalIncomplete = true;
+                               }
                 else if (!iv->isProducing(mUtcTimestamp, rec)) {
                     if (MQTT_STATUS_AVAIL_PROD == status)
                         status = MQTT_STATUS_AVAIL_NOT_PROD;
@@ -660,7 +665,7 @@ void app::sendMqtt(void) {
 
         mMqttSendList.pop(); // remove from list once all inverters were processed
 
-        if (true == sendTotal) {
+        if (true == sendTotal && false == totalIncomplete) {
             uint8_t fieldId;
             for (uint8_t i = 0; i < 4; i++) {
                 switch (i) {
```

stefan123t, 2022-11-07:
@lumapu I thought about such an incomplete Total calculation / validation earlier too. The question here remains as this can now catch and prevent only collection rounds where one of the inverters does not answer exceeding the retransmit and retry limits we have set. Only option to fix this is to manually remove the faulty or not responding inverter from the setup to make it work again. 
I would expect we remove an inverter after several periods / rounds to query a single inverter have failed from the total calculation. Otherwise the Total may not be calculated until the faulty inverter (e.g. wrong serial) is removed from the setup again, so at least a warning that and why the total has not been calculated in the past period should be given in the UI and/or a new mqtt topic ?

Especially with east- / west-oriented inverters there might be some hours before the west-oriented inverters get enough sunlight to respond in the morning or east-oriented may cease responding earlier in the evening. Or is the global irradiation generally enough to counter this effect ?

lumapu, 2022-11-07:
> so at least a warning that and why the total has not been calculated in the past period should be given in the UI and/or a new mqtt topic ?

this would be a good aproach. I don't think it makes sense to stop communicating with inverters dynamically. In most of the setups the settings should be static an the inverters are reachable.

> Or is the global irradiation generally enough to counter this effect ?

Yes I think it's enough. We are talking about some minutes of difference between east and west.

stefan123t, 2022-11-07:
@lumapu I will leave this issue open to add a warning on the UI / in a special MQTT status topic that the Total has not been calculated in the past round of inverter queries.

Argafal, 2022-11-10:
Yes, I agree, a warning is a good idea.

commit 28267262 (2022-11-06): fix #396 thx @Argafal
closed 2022-11-19 by lumapu