lumapu/ahoy #398: Remove "not producing" from inverter status when an inverter is not available

Argafal, 2022-11-06:
When an inverter is not yet available, a message is displayed stating that the inverter "is not available and not producing". However, the latter is not true: my electricity meter clearly shows the inverter producing. ;)

If an inverter is not available, I believe no statement can be made whether it is producing or not. It can simply not be known until the the inverter becomes available -> Schrödinger's production message. I propose to change the message to "is not (yet?) available."

haselchen, 2022-11-06:
> When an inverter is not yet available, a message is displayed stating that the inverter "is not available and not producing". However, the latter is not true: my electricity meter clearly shows the [...]

Look at this Issue

https://github.com/lumapu/ahoy/issues/391

The Same?

Argafal, 2022-11-06:
No, subtle difference. The issue you reference is about the production status when an inverter is available, and an inconsistency between gui and mqtt. 

My issue is about an inverter that is _not_ available (because it has not answered yet). In that case we simply have no knowledge whether its producing or not.

lumapu, 2022-11-06:
dev build should fix that

closed 2022-11-20 by lumapu