lumapu/ahoy #402: [ESP8266/ESP32/RaspberryPi] Problem Description / Beschreibung

rbike, 2022-11-07:
### Platform

ESP8266

### Model name

AZDelivery D1 Mini NodeMcu

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO2    |
| Pin 4 CSN     | GPIO15    |
| Pin 5 SCK     | GPIO14  |
| Pin 6 MOSI    | GPIO13  |
| Pin 7 MISO    | GPIO12  |
| Pin 8 IRQ     | GPIO4    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.28

### Github Hash

2e08ee0

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: hhha02
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: xxxxxxxxxxxx
    - Name: HM-600
    - Max Module Power (Wp): 420,420
    - Module Name: ML,MR
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: de.pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: xxx.xxx.xxx.xxx
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D2 (GPIO4)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Debug Serial Log output

```shell
----------------------------------------
I: Welcome to AHOY!
I: 
point your browser to http://192.168.20.23
I: to configure your device
I: ----------------------------------------

I: subscribe to HM600/devcontrol/#
I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 250 microseconds
Auto Retry Attempts	= 0 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Disabled
Primary Mode		= RX
TX address		= 0xe7e7e7e7e7
pipe 0 (closed) bound	= 0xe7e7e7e7e7
pipe 1 ( open ) bound	= 0x5056438401
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: [NTP]: n/a UTC
W: Time not set or it is night time, therefore no communication to the inverter!
```

### Error description

It is not possible to set the ESP time, neither per NTP (various NTP server) nor per Browser. 
It always shows on the entry page: ESP-Time: NTP timeserver unreachable.
Therefore also a connection to the inverter cannot be established

lumapu, 2022-11-07:
Thank you for your issue report. This issue will be fixed in the next release. Until then you can test the most recent development version `0.5.32`. This can be found in the `Actions` section here in Github or without login on our website https://ahoydtu.de/getting_started in the 3rd section *Firmware flashen*

rbike, 2022-11-07:
Thank you for the pointer to the development version. With this version setting the time via Browser works, but still not via NTP.

closed 2022-12-01 by lumapu

stefan123t, 2022-12-02:
I have seen that it takes about 1 minute (or more?) after connection to the WiFi to get an automatic NTP sync.
@rbike did you wait patiently and the ntp servers are reachable from the ESP.
@lumapu maybe we could add a ping or some other verification that NTP can be reached / synced ?

lumapu, 2022-12-02:
@stefan123t yes I already think about how to solve it.
My next idea will be the same trigger as I use inside MQTT. There is a wifi connect event, once that is valid we can sync the time.