lumapu/ahoy #403: Mqtt 64 characters?

Sinned42, 2022-11-07:
Could you give the usename for mqtt at least 64 characters? Flespi.io uses tokens with 64 characters as usename. 

Thanks in advance. 

## Hardware
  * [ ] ESP8266

Modelname: Wemos D1 Arduino Uno

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [x] **square dot** indicates original Nordic Semicon chip 

### Antenna:
* [x] circuit board

DanielR92, 2022-11-17:
First fix is available, but I cant test the change. I dont have the envirement here.

stefan123t, 2022-11-18:
@DanielR92 you closed the PR with String implementation and want to do it with increased size of the char[] array.
Does this issue also align with the 158 characters for Azure IoT Cloud we discussed recently on Discord ?

closed 2023-03-08 by lumapu