lumapu/ahoy #405: what bin-file to use

gent99, 2022-11-08:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

There are two bin-files for ESP8266:

221030_ahoy_0.5.28_esp8266_1m_2e08ee0.bin
221030_ahoy_0.5.28_esp8266_2e08ee0.bin

what file has to be used for what?

Kind regards

knickohr, 2022-11-08:
…1m… is for the smaller 8266 with only 1MB RAM.

stefan123t, 2022-11-09:
@lumapu we should rename the 1m to 8285 and add 4m to the esp8266 binary name to make the purpose / platform clear.

lumapu, 2022-11-09:
@stefan123t is it really needed? Can't we flash the 'normal' ESP8266 binary to a 1M ESP? I know someone who has an ESP07 with 1M flash and the 'normal' firmware works great - expect OTA

stefan123t, 2022-11-11:
@lumapu I do not know what was the conclusion about AsyncWeb and the 1MB vs 4MB flash requirement ?
No OTA will probably not work for ESP8285, maybe not even if we reduce MAX_INVERTER to 1 *sic*

commit 9c68ad14 (2022-11-12): moved payload related stuff to other file
fix line break for inverter list on index
update release workflow action
fix #405 renamed binary for ESP8285 (platformio.ini)
closed 2022-11-19 by lumapu