lumapu/ahoy #406: MQTT Topic

gent99, 2022-11-08:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266/32 GPIO |
| ------------- | --------------- |
| Pin 1 GND []  |  GND            |
| Pin 2 +3.3V   |  +3.3V          |
| Pin 3 CE      | GPIO_2/_4 CE    |
| Pin 4 CSN     | GPIO15/_5 CS    |
| Pin 5 SCK     | GPIO14/18 SCLK  |
| Pin 6 MOSI    | GPIO13/23 MOSI  |
| Pin 7 MISO    | GPIO12/19 MISO  |
| Pin 8 IRQ     | GPIO_0/0 IRQ    |

Note: [] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

After updating from 0.5.17 to 0.5.28 i've problems to figure out my topic.

I used to have

stefan123t, 2022-11-09:
@gent99 can you post your current settings and add the USB Serial Log of your 0.5.28 version ?
Would you consider to try development release 0.5.32 ?

gent99, 2022-11-10:
Unfortunately i'm not able to connect the DTU tu USB, but I'd like to test the development version.

closed 2023-02-07 by lumapu