lumapu/ahoy #407: Seit 0.5.28 wieder falsche MQTT Nachrichten

Matze7779, 2022-11-10:
Hallo,

seit der 28 hab ich wieder verstümmelte MQTT Nachrichten.
Das war früher auch schon mal der Fall und war bei der Version 17 komplett verschwunden. Mit der 28 taucht es wieder auf.
Anbei mal ein Screenshot aus IOBroker.

[image]

lumapu, 2022-11-10:
danke für den Report und die Versionen. Damit kann man den Fehler besser eingrenzen.

stefan123t, 2022-11-10:
@lumapu meiner Meinung nach ein Buffer overflow oder Problem beim String kopieren ?
Vor allem daß es wirklich überall steht: Solaranlage, total, 0% etc. 
Vielleicht wird irgendwo bei einer Kopie des Wertes das begrenzende Null Byte bzw. `sprintnf` vergessen.

h1ghrise, 2022-11-11:
Ich hatte sowas ähnliches mal in einem C Projekt. (Zwar ein ESP32, aber auch Mikrocontrollerprogrammierung).
Bei mir lag es an einer fehlenden `static` Deklarierung der Variablen.

Skusi, 2022-11-12:
Bei mir sind die Readings die in Fhem ankommen ok. Ich teste gerade mit 6 invertern HM-300. Nur funktioniert das auseinander schlüsseln der einzelnen Inverter nicht per autocreate. Das hab ich dann von Hand gemacht.
Stören tut mich aber die "Gesprächigkeit" per MQTT. In 50 min hat die DTU 17134 Messages gesendet.
Das ist doch nicht so gewollt !?
Meine 0.4.17 ist da ruhiger.

roku133, 2022-11-12:
Bis v0.5.32 sind es bei mir mit einem Wechselrichter und einem Solarmodul ca. 50.000 MQTT-Messages innerhalb von zwei Tagen.
Ab v0.5.33 gibt es immer wieder Message-Bursts, die Gesamtanzahl der Nachrichten hat sich dadurch drastisch erhöht.

roku133, 2022-11-13:
@lumapu: mit v0.5.35 sind die MQTT-Message-Bursts wieder verschwunden - super und vielen Dank für den schnellen Fix.

stefan123t, 2022-11-14:
Problem laut @roku133 in 0.5.35 gelöst. Danke für die Rückmeldung.

closed 2022-11-14 by stefan123t