lumapu/ahoy #410: Interface for thirdparty apps

womoak75, 2022-11-12:
An interface which allows easy integration of additional user applications into a custom Ahoi build would be a nice feature.

Unfortunately my knowledge of C++ is very 'minimalistic' (java ist my favourite), but i tried my very best to provide a simple example.
https://github.com/lumapu/ahoy/compare/main...womoak75:ahoy:main

| pros | cons |
|-------|---------|
|minimal impact on existing code base | no 'nice' sw design||
|esp8862/esp32 is in idle mode most of the time anyway| |
|minimize merge effort on future custom builds| |

please feel free to agree/disagree or add additional topics.

stefan123t, 2022-11-14:
@womoak75 this looks like a example template for building and adding custom plugins into the AhoyDTU.
Thanks, let us see who will make use of this blueprint for whatever app-plugin they want to provide for AhoyDTU.
Did you already have a specific purpose in mind ?

womoak75, 2022-11-14:
Yes, i do.
Want to connect my power meter and implement some "Nulleinspeisung".

I also added a mqtt callback to my example, to enable input/control data for a thirdparty app.
e.g. publishing to topic 'inverter/thirdparty/{data}' which will be ignored by the ahoi app.
Should be sufficient for a simple start and enables user to add their own code, 
without "hacking" into the ahoi code base.

A more sophisticated approach would be adding hooks for all major functions (configuration, web, api, etc.),
but 'a journey of a thousand miles begins with a single step' 

 If desired, I could add a pull request for simplicity reasons. :)

stefan123t, 2022-11-15:
@womoak75 could you add a #define to disable your plugin by default and some example in example_overwrite_config.h before you PR.
@lumapu did you take a quick look at this sample plugin or should @womoak75 wait with his PR till you have refactored a couple of things which you are busy with ?

womoak75, 2022-11-15:
Will add a PR at your request.

lumapu, 2022-11-15:
@stefan123t will do

womoak75, 2022-11-16:
also tried to add web / webApi / gui support

stefan123t, 2022-11-17:
A possible example scenario for this plugin adapter could be the Thingspeak integration in #185 or the Prometheus support in #190 and PR #356
Maybe the Azure IOT Cloud OAuth also could qualify for such a flexible plugin approach ?

womoak75, 2022-11-18:
https://github.com/lumapu/ahoy/compare/main...womoak75:ahoy:plugins
plugin hacker version v0.0.1 :)

stefan123t, 2024-10-27:
@lumapu können wir das als Beispiel für ein Plugin aktzeptieren und mergen ?
@womoak75 hat einiges an Doku geschrieben und man müsste mal prüfen ob es noch aktuell / mergebar ist ?