lumapu/ahoy #411: Suche Hilfe zum "inkludePath" Problem

daja64, 2022-11-12:
Hallo, jetzt wollte ich mal mit meiner Anzeigegeschichte weitermachen und  nun dieses Problem. Mußt auf einen neues NB umziehen, beide haben Win10 64bit. VScode gestartet, Clone Git Projekt, bis hier ok. Wenn ich jetzt die z.B. die main.cpp anklicke, kommt dieser Fehler. Auf dem alten NB war das nicht der Fall, weil ich da schon mal damit hantiert habe ohne diese Meldung. Leider find ich außer "die Pfade händisch eintragen" keine gescheite Hilfe. Warum nun händisch obwohl es beim alten NB automatisch ging ?

lumapu, 2022-11-12:
evtl. liegt es an einer der Dateien im `.vscode` Ordner.
Ansonsten mal clean Projekt (Mülltonne unten in der Statuszeile) ausprobieren.

daja64, 2022-11-12:
Bitte mal genau beschreiben wie das mit Vscode geht, also den Ablauf vom Start des Programm, bis dahin wo ich kann dann z.B. die app.cpp bearbeiten kann.

daja64, 2022-11-13:
Mann bin ich manchmal doof. Es war nur der falsche Projektordner. Also ist noch einiges an Einarbeitung nötig. Kann dann geschlossen werden. Sorry für die Aufregung.

lumapu, 2022-11-13:
alles gut, hat immer seine Berechtigung ;-)

closed 2022-11-13 by lumapu