lumapu/ahoy #413: JSON zum Import für Node-Red

knickohr, 2022-11-13:
Ich habe für Node-Red einen Flow erstellt um die Ausgabe von Ahoy ein wenig grafisch darzustellen.

Hintergrund: 
Mich haben schon einige danach gefragt ob ich ihnen das zur Verfügung stellen kann. Offensichtlich ist Node-Red nicht so einfach wie es für mich erscheint 😇

Voraussetzungen:
- es muß ein MQTT-Broker (ich nutze Mosquitto auf einem Raspberry Pi 4B mit 1GB RAM und einer 16GB SD-Karte) installiert sein
- Node-Red muß installiert sein 😅
- Das Node „ node-red-dashboard“ muß in der Palette installiert sein

ToDo:
- Ahoy am Broker anmelden (unter Settings, MQTT)
- Überprüfen ob Ahoy auch publihed
- Das JSON im Attachment importieren
- In den Eigenschaften eines jeden MQTT-Node den Broker anpassen, username und password nicht vergessen !
- den Flow deployen (roter Button oben rechts)
- Das Dasboard im Browser aufrufen und sich erfreuen 😉

Viel Spaß damit!  Sollte noch was unklar sein, so kann man hier im Issue fragen.

[image]
[image]
[image]
[image]
[flows.json 2.txt](https://github.com/lumapu/ahoy/files/9996911/flows.json.2.txt)

knickohr, 2022-11-13:
Nicht vergessen den User, Password, und die IP im MQTT-Node anzupassen. Ebenso wie den Topic identisch zu dem im Ahoy. Also alles was bei mir in den Violetten Nodes AhoyDTU heißt muß geändert werden in eurem Topic im Ahoy.

lumapu, 2022-11-13:
das wäre doch was für unsere Website ahoydtu.de
Ich kann es gerne einstellen, und hierfür eine neue Rubrik schaffen.

knickohr, 2022-11-13:
Problem ist, das es halt von jedem der es nutzen möchte angepaßt werden muß.

Nicht nur wegen User, Password, iP und Topik, sondern auch in der Struktur.

Ich verwende momentan 2 WR mit je einem Strang, bei einem WR mit 2 Strängen oder noch mehr, wird es nur wenige Daten ausgeben.

Es ist halt auch ein bißchen Eigeninitiative notwendig. Aber man kann aus dem Flow lernen und selbst umstricken und erweitern.

lumapu, 2022-11-13:
ja genau, mein Gedanke war, dass man dadurch schon ein Grundgerüst stehen hat auf das man aufbauen kann. Ich kenne es von mir selbst, so eine Visualisierung wird optimiert solange sie existiert und ist nie fertig.

stefan123t, 2022-11-14:
@knickohr könntest Du das Template mit inverter anpassen, damit es bereits out-of-the-box funktioniert ?

@lumapu, evtl. kann man dafür auch den @womoak75 in #410 zur Verfügung gestellten app Callback nutzen ?
Damit ließen sich ggf. die notwendigen Informationen (vor allem das aktuelle MQTT Topic) aus AhoyDTU holen.

stefan123t, 2022-11-14:
Wenn die uptime in Sekunden angegeben ist einfach uptime/(60*60) bzw. uptime/3600 dann hast Du einen Wert in Stunden.
Oder willst Du einen String anstelle eines Zahlenwertes ausgeben ?
Schau doch mal hier im Node-Red Forum: https://discourse.nodered.org/t/date-time-formatting/1138

knickohr, 2022-11-14:
Nein, das Problem war, die Ausgabe ist ein String, ich mußte ihn einfach in eine Zahl umwandeln 🤪

knickohr, 2022-12-02:
Update 😎

Ich habe es wieder ein bißchen angepaßt. Ist zwar sehr speziell für meine Zwecke gestrickt, aber man kann es mit etwas Hirnschmalz an die eigenen Gegebenheiten anpassen.

Hintergrund:
Ich verwende 6 Inverter mit jeweils Max. 300Wp. Rein theoretisch könnte ich auch 3x 600er mit 2 Strängen nehmen. Die Panels sind nur leider zu weit auseinander, so würde es eine zu lange DC-Zuleitung werden. Da ich 3 Stockwerke mit jeweils eigenem Zähler habe, kann ich natürlich auch 3x BKWs betreiben.

Der Flow ist noch nicht sehr schön, z.B. die vielen MQTT-Nodes, das kann man verbessern. Ihr müßte auf jedem Fall die Topics in den Nodes an eure Konfiguration anpassen, also „AhoyDTU“ muß wahrscheinlich in „inverter“ (Standardwert) umgetauft werde.

[image]

[flows.json 2.txt](https://github.com/lumapu/ahoy/files/10142455/flows.json.2.txt)

knickohr, 2022-12-05:
Ich habe den Flow nochmal angepaßt und optimiert. Jetzt muß nur noch an einer Stelle das MQTT-Topic geändert werden. Allerdings sind noch feie Inverternamen in dem großen gelben Split „Topic“ enthalten. Weiß noch nicht wie ich das dynamisch lösen kann. Ebenso die Anzahl der Inverter und Stränge sind hart kodiert. Also ohne Hand anlegen wird das nicht „out of the box“ funktionieren.

[image]

[flows.json 2.txt](https://github.com/lumapu/ahoy/files/10157718/flows.json.2.txt)

Quickiman, 2022-12-10:
Hallo Leute, ich habe die Vorlage von Knickohr genommen und sie an meinen HM-1500 mit 2 Strings a 2 Module angepasst.
Ich bin ein totaler Nub deswegen sind da bestimmt noch Fehler in den Flows aber es läuft soweit alles wie ich es mir vorgestellt habe. Das einzige was ich noch nicht weiss, ist wie man die Position der Fenster im UI verändern kann.
[flows-Quickiman.json.txt](https://github.com/lumapu/ahoy/files/10200000/flows-Quickiman.json.txt)
[image]

knickohr, 2022-12-10:
Hast einen guten Lehrer 😉

Quickiman, 2022-12-10:
Das kann man wohl sagen!

stefan123t, 2022-12-11:
@knickohr & @Quickiman dann kann man das ja jetzt so allgemein formulieren und ablegen, dass man /DTU1/WR1..n/CH1..4 als Topics vorgibt und die kann dann jeder durch seine persönlich verwendeten Namen ersetzen (search&replace) bzw. so kann man auch einfach einen zusätzlichen WRn anhängen ?

Ich habe ja immer noch das Zusatzprojekt "Harmonisierung der OpenDTU und AhoyDTU MQTT Topic Hierarchie" auf dem Radar, aber das sprengt Euren Fokus hier für NodeRed und sollte mE in ein eigenes Issue ...

knickohr, 2022-12-11:
Ich muß mal schauen ob man das in eine/zwei Variablen ablegen kann, so das man nur noch diese Variablen einmalig anpassen muß und der Arrest stimmt dann.

Dummerweise gibt es zig Konfigurationen für die Inverterm
- Inverter mit einem Strin
- Inverter mit 2
- Inverter mit 4 …
- Ein Inverter
- Zwei Inverter …
- Und alle Mischkonfigurationen
- Und zu allem Überfluß auch noch unterschiedliche Typen durcheinander 😲

Im Prinzip kann man nur ein Gerüst vorgeben, so das man sich das jeweils zusammen puzzeln muß.

Quickiman, 2022-12-13:
Ich habe das Layout ein wenig angepasst, damit es auf dem handy richtig skaliert ist. Die Widgets haben jetzt alle die gleiche breite und füllen das ganze display aus.

[flows-Quickiman.json.txt](https://github.com/lumapu/ahoy/files/10217528/flows-Quickiman.json.txt)

teletapi, 2023-01-02:
Da ich das ganze mit der Nodered Geschichte nicht so ganz kapiere hab ich mir den available Text etwas anders dargestellt und den jeweiligen text den LED zugeordnet anders bekomme ich das leider nicht hin.

[image]

[Testboard FW5.67.json.txt](https://github.com/lumapu/ahoy/files/10329716/Testboard.FW5.67.json.txt)

knickohr, 2023-01-04:
Update :  das Dashboard hat sich weiter entwickelt

- Neue Statusanzeige der Wechselrichter
- Status wurde komplett überarbeitet, die Zustände werden jetzt aus den Stati der einzelnen WRs selber berechnet
- Die Uhren wurden teilweise durch andere Gauges ersetzt und erweitert
- Jeder WR hat jetzt einen eigenen auswählbaren Tab (vorbereitet auf 6 Kanäle)
- Die Datenrate des Grafen wurde reduziert um RAM zu sparen
- Nodes wurden in funktionelle Blöcke zusammen gefaßt

[image]

Viel Spaß beim abgucken und anpassen. Topic und Invertername muß mit ziemlicher Sicherheit in den Nodes von euch geändert werden. Ich habe noch keine vernünftige Möglichkeit gefunden das variabel zu machen.

Mit ziemlicher Sicherheit müßte ihr auch Nodes in Eurer Palette nachinstallieren !!!

[flows.json 2.txt](https://github.com/lumapu/ahoy/files/10346850/flows.json.2.txt)

stefan123t, 2023-01-05:
@knickohr Du kannst ggf. auch den ersten Post editieren und dort eine Zusammenfassung (TL;DR) posten mit den aktuellen flows.

stefan123t, 2023-01-12:
@knickohr willst Du das ggf. auf dem Wiki dokumentieren ?

knickohr, 2023-01-12:
Kann ich machen.

Leider ist es kein Universalrezept da ich die Namen nicht variabel machen kann und somit ist selbst Hand anlegen gefordert. Sind zwar nur 2 Nodes die bearbeitet werden müssen, alter trotzdem. Das Modell mit den Strängen und Invertern muß man ja auch anpassen.

knickohr, 2023-03-04:
Mal ein Update (damit’s nicht langweilig wird 🤪)

Der Flow wurde angepaßt auf 6 HM-300 Inverter, jeweils 2 sind zu einem BKW zusammengefaßt, also max. 600W. Außerdem habe ich den Flow etwas übersichtlicher gestaltet.

[image]

[AhoyDTU.json.txt](https://github.com/lumapu/ahoy/files/10888951/AhoyDTU.json.txt)

knickohr, 2023-03-31:
Update 😉

Jetzt mit 9 Invertern und übersichtlicheren Flow.

[Ahoy.json.txt](https://github.com/lumapu/ahoy/files/11123771/Ahoy.json.txt)

[image]

knickohr, 2023-07-23:
Hier auch mal wieder ein Update 😅

- Jetzt mit 13 Invertern
- Etwas anderes Design
- Die einzelnen WRs werden jetzt beim Durchscannen am Dashboard angezeigt
- Manuelles auswählen der WRs ebenfalls möglich (springt nach 10s wieder in den Automatikbetrieb zurück
- Anzeige des Free Heap und des möglichen Tagesertrages aus einem Forecast
- Diagramme aus dem Solar Forecast
- Diagramme für Power Limit, Power und Temperatur der einzelnen WRs
- Balkendiagramm zur Anzeige der eingespeisten Leistung aller WRs nebeneinander
- Optimierung der Flows
- Und und und …. Eine Fundgrube für den der selber was basteln will oder den Flow anpassen möchte

(Achtung, der api-key für den Solar Forecast wurde raus genommen, hier muß sich jeder seinen Key selbst eintragen oder weglassen. Dann werden die Forecastdaten aber nur jede Stunde aktualisiert. Siehe Flow Solar Forecast, die beiden http-request Nodes.)

[image]

[image]

[image]

[image]

[image]

[image]

Zwischen den einzelnen Dashboards wechselt man mit Wischen nach links oder rechts. Diese Version ist nur für Tablets oder größere Smartphones geeignet. Am PC muß man die Optionen vom Dashboard verändern.

[flows.json.txt](https://github.com/lumapu/ahoy/files/12139187/flows.json.txt)

lumapu, 2023-07-23:
sehr schöne Grafiken @knickohr 😃

Starfoxfs, 2023-10-26:
Hi,
mal ne Frage dazu:
Bei AhoyDTU werden die Werte in einzelnen Topics und als Nummer von MQTT übermittelt.

Kann man das nicht so wie beim Shelly senden in einem zusammengefassten JSON Objekt ?

Beim Shelly mit Tasmota kommt es so in Node-Red an:

`{
  "Time": "2023-10-26T08:59:34",
  "ENERGY": {
    "TotalStartTime": "2023-01-28T00:00:00",
    "Total": 411.679,
    "Yesterday": 0.522,
    "Today": 0.034,
    "TodaySumImport": 0.034,
    "TodaySumExport": 0,
    "ExportActive": [
      0,
      0
    ],
    "Period": [
      1,
      0
    ],
    "Power": [
      34,
      0
    ],
    "ApparentPower": [
      63,
      0
    ],
    "ReactivePower": [
      18,
      0
    ],
    "Factor": [
      0.54,
      0
    ],
    "Frequency": 50,
    "Voltage": 242,
    "Current": [
      0.238,
      0
    ]
  }
}

tele/tasmota/SENSOR
`

Bei der AhoyDTU muss man im Prinzip wie oben beschrieben für jedes Topic ein MQTT Verbindung erstellen.

z.b. 
ch0/YieldDay Wert = 38
ch0/YieldTotal Wert = xyz

usw usw

knickohr, 2023-10-26:
Siehe Issue #1199

knickohr, 2024-01-07:
Wieder mal ein paar Updates und neue Fows :

- Endausbau auf 16 Inverter
- Anzeige einiger Informationen über Popup-Menüs
- Ausgaben der verschiedenen Stati der DTU und Inverter
- Ausgabe der Anzahl der erfolgreichen Datenpakete pro Inverter
- Diagramm für Yield Day (Werte werden lokal gespeichert)
- Radiostatistiken (hilfreich zur optimalen Platzierung der Inverter)
- Flows optimiert und in verschiedene Gruppen zusammen gefaßt
- Performanceverbesserung

[image]

[image]

[image]

[image]

[image]

[image]

stefan123t, 2024-10-27:
@knickohr wäre es nicht sinnvoll Deine Node-Red Flows unter tools/Node-Red im Repository zu hinterlegen ?
Vielleicht noch eine kleine aktuelle Anleitung als Readme.md dazu wie man das in Node-Red importiert / aufsetzt.

knickohr, 2024-10-28:
Uiuiiiuiiii !

Ich muß erst mal schauen ob „das Alte“ überhaupt noch kompatibel mit Ahoy, bzw. jSON-Payload ist 😲