lumapu/ahoy #414: YieldDay Reset

usekevin1988, 2022-11-13:
Bin mir nicht sicher, ob es so gewollt ist oder ein BUG, daher sicherheitshalber mal festhalten.

Wenn ich mein Ahoy neustarte, wird "YieldDay" resettet. "YieldTotal" bleibt erhalten.

Nutze Version 0.5.35

Kann man den Wert eigentlich Manuell anpassen irgendwie?

homeautomation2022, 2022-11-13:
Beide Werte kommen direkt vom Wechselrichter, dieser resettet YieldDay nur beim Ausgehen zum Sonnenuntergang. Wenn du tagsüber Ahoy neustartest, kommt der YieldDay wieder, sowie auch der Yield Total. In der Nacht kommt da nichts, da der Wechselrichter da nicht erreichbar ist.

Manuell kann man da nichts anpassen. Einzig YieldDay wird zurückgesetzt, wenn du das Restart Command sendest.
Wenn du nur ESP Reboot anklickst holt er sich die Werte vom WR danach wieder.

usekevin1988, 2022-11-13:
ok, dann lags am neustarten des WR.. dann hat sich das erledigt :)

closed 2022-11-13 by usekevin1988