lumapu/ahoy #417: Feature Request | "Sunrise" "Sunset" via MQTT übertragen

ambivalent11, 2022-11-13:
Ist es möglich die beiden Datenpunkte ebenfalls via MQTT zu übertragen?

stefan123t, 2022-11-14:
Möglich ja, die Frage ist, wieso brauchst Du die explizit und vor allem wie oft ?
Sollte ja dann einmal am Tag genügen ggf. mit einem retain Flag von 12 / 24h ?

@lumapu wie oft, bzw. wann werden die nochmal berechnet ?
Nur ein / zweimal am Tag um 0:00 und 12:00 Uhr ?

lumapu, 2022-11-14:
@stefan123t ich glaube 1x, muss ich prüfen. 

im Grunde spricht nichts dagegen, diese beiden Infos auch zu übertragen.

knickohr, 2022-11-14:
Wenn wir schon dabei sind, dann bitte auch WiFi RSSI mit übertragen 😉

lumapu, 2022-11-14:
wird die noch nicht übertragen?

knickohr, 2022-11-14:
Ich habe keine. MQTT-Topic gefunden 😉

haselchen, 2022-11-14:
@knickohr 

Über die API kannste den Sunrise und Sunset abrufen und dann umwandeln in die Uhrzeit. Hab das per Iobrokers Blockly gemacht. Über MQTT gleich den Wert zu bekommen ist natürlich komfortabler :)

DanielR92, 2022-11-15:
Ok, beduetet dann:

- Sunrise + Sunset
- WiFi RSSi

Wenn ich Zuhause bin, kann ich mir das mal anschauen. Dann ist es vom Tisch.

lumapu, 2022-11-16:
@DanielR92 ist implementiert oder?

closed 2022-11-21 by lumapu