lumapu/ahoy #420: AHOY-DTU erzeugter AP keine Verbindung möglich

Sebel13, 2022-11-14:
## Hardware
  * [X ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [X ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D2 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D1 |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ X] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ X] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [X ] Windows
* [ ] Mac OS

### Debugging:
* [ X] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo,

ich bin leider neu auf dem Gebiet und habe nun angefangen mich einzulesen etc. Ich habe einen HM600 und habe mir nach der wirkluchen guten Anleitung ein DTU zusammengebaut. Das eigene WIFI AHOY-DTU wird erzeugt. Mit dem Passwort esp_8266 bekomme ich aber unter meinem Win10 Rehcner immer die MEldung Keine Verbindung zu diesem Netzwerk möglich.
Mit meinem Iphone habe ich das gleiche Problem. Ich habe die Firmware 05.28 mit dem Nodemcu geflasht.

reserve85, 2022-11-14:
Ich hatte ein ähnliches Problem auch schonmal - da kam allerdings die Meldung Passwort falsch. 
Nachdem ich den ESP-Flash vorab gelöscht und anschließend erneut geflasht hatte, funktionierte es.

lumapu, 2022-11-14:
Hallo, es kann sein, dass die Geräte (PC / Smartphone) versuchen über das Wifi eine Internetverbindung aufzubauen. Das kann nicht gegen und könnte der Grund für die Meldung sein.
Trotz der Meldung einfach versuchen auf die IP http://192.168.1.1

oder für die Development Version
http://192.168.4.1 

zuzugreifen.

Sebel13, 2022-11-14:
Hallo,

danke für die Hilfe. Er ist dann aber nicht verbunden.
Das funktioniert leider nicht. Er verbindet einfach nicht: keine Verbindung zu diesem Netzwerk möglich.

Sebel13, 2022-11-14:
Gibt es eine Möglichkeit sich anders zu verbinden?

Sebel13, 2022-11-14:
Mit einem alten PC wo ich alle Netzwerke vorher auslösche funktioniert es. Kann mir das jemand erklären?

HFarryranz, 2022-11-14:
Dein Heim-Wifi hat den gleichen IP- Adressbereich wie der AP des ESP?
Versuch s mal außerhalb deines Home-Wifi.

DukeWolle, 2022-11-15:
Versuch mal bitte
http://192.168.1.1/setup
oder 
http://192.168.4.1/setup
Je nach Version

stefan123t, 2022-11-15:
> Mit einem alten PC wo ich alle Netzwerke vorher auslösche funktioniert es. Kann mir das jemand erklären?

Hallo @Sebel13 vermutlich hat Deine Fritz Box die IP Adresse 192.168.1.1 (bzw. eigentlich Default 192.168.178.1). Wenn sich Dein PC jetzt mit dem WLAN AccessPoint der AhoyDTU verbindet gibt es eben zwei Endgeräte mit der 192.168.1.1 in Deinem Netzwerk. Er weiß dann nicht wohin er die Pakete schicken soll und entscheidet sich aufgrund der bestehenden Einträge im ARP Table für die MAC der Fritz!Box.
Daher haben wir auch die neue IP 192.168.4.1 bereits in der Development Version hinterlegt. Da kommt es vorraussichtlich noch seltener zu Kollisionen beim Routing 1x1.
So also jetzt klappts dann darfst Du das Issue als gelöst schließen. Danke!

Sebel13, 2022-11-15:
Danke funktioniert

closed 2022-11-15 by Sebel13