lumapu/ahoy #422: Leistungsbegrenzung

Nobelschroeder, 2022-11-14:
## Hardware
  * [X ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hallo zusammen,

erst einmal vilen Dank an die Entwickler der Firmaware. Da stecken ja schon ein paar Jahre Arbeit drin wenn ich das richtig verfolgt habe.

Ich habe mir jetzt die Hardware gekauft und die Firmeware V0.5.28 drauf geflasht. Hat alles super funktioniert und nach einem Neustart des Wechselrichters ist auch die Verbindung zu diesem zustande gekommen. Irgendwie vermisse ich nur die beschrieben Möglichkeit zu Einstellung der Leistung. Nach dem Durchstöbern der Anfragen bin ich darauf gestoßen, dass dies bei V0.5.14 noch möglich ist. Dem ist auch so. Jetzt meine Frage: Wurde diese Feature aus der Firmeware V0.5.28 raus genommen oder habe ich irgendetwas falsch gemacht?

lumapu, 2022-11-14:
wir sind seit Anfang dieses Jahres so richtig durchgestartet.
Unter dem Menüpunkt 'Serial Console' gibt es die Bedienung der Leistungsbegrenzung.

Nobelschroeder, 2022-11-14:
Alles klar, jetzt hab ich's auch gefunden. Vielen Dank noch mal und größten Respekt davor was ihr hier auf die Beine gestellt habt.

stefan123t, 2022-11-15:
@lumapu sollte man eventuell anders benennen Remote Control oder so.
Eventuell auch die Serial Console nochmal auskoppeln aber gerade die braucht man ja zur „Kontrolle“ ob es schon alles richtig klappt :)

Nobelschroeder, 2022-11-16:
Guten Morgen,

ich muss das Thema noch mal aufmachen. Hintergrund ist der, dass ein Freund letztes Jahr ein Balkonkraftwerk über eine österreichische Firma gekauft hat. Nachdem wir dieses Jahr dann einen Leistungsmesser angeschlossen haben, ist uns aufgefallen, dass die Anlage höchstens 424W peak produziert obwohl sie aus zwei 325Wp-Modulen mit jeweils einem HM-300 besteht. Auf Nachfrage beim Lieferanten kam raus, dass sie da wohl eine Zeit lang auf 70% gedrosselte Anlagen verkauft haben. Nun habe ich mit der Ahoy DTU die Leistungsbegrenzung auf 300W gestellt worauf in der Liveansicht 142,8% standen, die Anlage aber immer noch nur 424W produziert. Beim zurückrechnen auf 100% (210W) ist uns dann aufgefallen, dass das genau 70% von 300W sind. Auf ereute Nachfrage wurde uns mitgeteilt, dass wir die Länderkennung mittels der DTU auf Österreich umstellen müssen. Das könnten sie auch aus der Ferne machen, dafür bräuchten wir aber eine echte DTU und es würde 2 Tage dauern. Nun zu meiner Frage: Gibt es bei der Ahoy DTU schon die Möglichkeit die Länderkennung zu ändern oder habt ihr einen Wert ausgelesen mit dem ihr noch nichts anfangen könnt der evtl. genau das ist?

DanielR92, 2022-11-17:
Dieser Punkt sollte möglich sein. Jedoch haben wir damals diesen Punkt nicht für wichtig erachtet. 
Ich sehe konkret nur ein Problem, auch wenn man den WR in DE betreibt und diesen auf Österreich konfiguriert ist es keine saubere Arbeit. Ich kann mir sehr gut vorstellen das durch die Änderung von DE auf AUT hier womoglich im Hintergrund noch andere parameter dadurch beinflusst werden.

Normalerweise wäre es sauberer ein neues GridProfile auf den WR zu laden die den neuen Regelung von DE entsprechen. 
Auch das wäre theoretisch möglich. 

Nur sind wir hier auch noch nicht soweit, da eine Modifizierte Firmware (am besten eine Originale von Hoymiles) Garantie erlischt + das verhalten des WR beinflussen könnten die wir noch nicht genauer kennen.

Wenn es soweit kommen würde, das der Hersteller euch für die Umstellung eine DTU ausleihen würde und ihr das selbst ändert. 
Wären Infos super, da wir hier noch kein Fortschritte gemacht haben. :)

@lumapu oder @stefan123t gern bitte ergänzen. 👍🏻

Nobelschroeder, 2022-11-17:
OK danke erst mal für die Antwort. Sollte es dazu kommen, werde ich berichten. Mein Freund ist allerdings drauf und dran sich vom Verkäufer (Zwischenhändler hier bei uns im Ort) neue Wechselrichter im Austausch geben zu lassen weil er keinen Bock drauf hat fremde Leute 2 Tage lang über sein Heimnetzwerk daran rum basteln zu lassen.

stefan123t, 2022-11-17:
Das ist eigentlich das GridProfile TOR-Thema in #96 und #365 und hat mit der Leistungsbegrenzung aus diesem Issue erstmal nichts zu tun.

Bitte dort nachlesen, UpdateProfile reengineeren und implementieren und dieses issue hier wieder schliessen. Danke! 

Wenn Ihr schon das Angebot bekommt fragt doch mal ob ihr die DTU dazu bekommt, dann könnt Ihr die TX/RX Daten im der DTU auslesen und seht was die Jungs da wirklich machen und wir würden auch was lernen: GridProfile updaten 😋 

Also @klahus1 hat auch so einen (seiner zwei) auf 70% abgeregelten / gedrosselten HM-300.

closed 2022-11-19 by Nobelschroeder