lumapu/ahoy #428: Serial bugs with 0.5.37

Waldo56, 2022-11-16:
### Platform

ESP8266

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

~100uF Elko

### Connection diagram

## Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.5.37

### Github Hash

da04d3f

### Build & Flash Method

ESP Tools (flash)

### Desktop

Windows

### Setup

### Device Host Name
    - Device Name: AHOY-DTU
### WiFi
    - SSID: YOUR_WIFI_SSID *don't paste here*
    - Password: YOUR_WIFI_PWD *don't paste here*
### Inverter
#### Inverter 0
    - Address: 1141752123456
    - Name: HM-600
    - Active Power Limit: 65535
    - Active Power Limit Control Type: no powerlimit
    - Max Module Power (Wp): 375, 375
    - Module Name: link, rech
### General
    - Interval [s]: 30
    - Max retries per Payload: 5
### NTP Server
    - NTP Server / IP: pool.ntp.org
    - NTP Port: 123
### MQTT
    - Broker / Server IP: 
    - Port: 1883
    - Username (optional): 
    - Password (optional): 
    - Topic: inverter
### System Config
#### Pinout (Wemos)
    - CS: D8 (GPIO15)
    - CE: D4 (GPIO2)
    - IRQ: D3 (GPIO0)
#### Radio (NRF24L01+)
    - Amplifier Power Level: LOW
#### Serial Console
    - print inverter data: [x]
    - Serial Debug: [x]
    - Interval [s]: 5

- Reboot device after successful save: [x]
- SAVE

### Error description

Im Menü Serial Console erscheinen keine Invertereinträge mehr.
Wenn man in diesem Submenü ist kann man nicht mehr über die Auswahl links oben das Serial Condole Menü verlassen. Das geht nur mehr durch Klick auf den Titel AhoyDTU.

lumapu, 2022-11-16:
danke, habe ich heute auch festgestellt

DanielR92, 2022-11-17:
Das kann ich später Zuhause prüfen. :)

lumapu, 2022-11-17:
habe es heute wieder gesehen

roku133, 2022-11-17:
funktioniert mit v0.5.40 (0305c11) wieder - super 👍

lumapu, 2022-11-17:
danke für das schnelle Feedback

Waldo56, 2022-11-18:
Kann es leider aktuell nicht testen. Die DTU ist nach dem Update auf 0.5.40 nicht mehr ansprechbar. Auch nicht via DTU WLAN.

Waldo56, 2022-11-18:
Hab nochmal geflasht.  Funktioniert mit 0.5.40

commit 033d9f04 (2022-11-17): fix #428 (two XHR requests at the same time results in strange behavior)
closed 2022-11-18 by stefan123t