lumapu/ahoy #431: Einspeise-Steuerung

Merlin8888, 2022-11-17:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |

| nRF24L01+ Pin | ESP32 GPIO      |
| ------------- | --------------- |
| Pin 1 GND [*] | GND             |
| Pin 2 +3.3V   | +3.3V           |
| Pin 3 CE      | GPIO4  CE   D4  |
| Pin 4 CSN     | GPIO5  CS   D5  |
| Pin 5 SCK     | GPIO18 SCLK D18 |
| Pin 6 MOSI    | GPIO23 MOSI D23 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [ ] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] Arduino
* [ ] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [ ] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Merlin8888, 2022-11-17:
Hallo, ist es möglich die Netzeinspeisung live zu steuern (z.B. auf 600W drosseln oder Null-Einspeisung) - also wie bei der originalen Hoymiles DTU, indem man die Daten von einem Smartmeter im Zählerkasten über eine R485 Leitung zur Verfügung stellt?
Vielen Dank

DanielR92, 2022-11-17:
Hallo Merlin,

theoretisch wäre dies möglich ja. 
Es fehlt aktuell noch die Software Einbindung in dem vorhanden Code. 

Theoretisch würde jedes Energiemessgerät gehen, das eine Schnittstelle anbietet. Die Umsetzung muss jedoch der Nutzer selbst machen. Damit ist gemeint, wenn du ein SmartMeter hast, der nur TTL kann, wir aber aktuell MQTT anbieten. Müsstest du TTL konvertieren. 

Wie auch immer... Mein Ziel ist es, das die Ahoy Software auch direkt mit dem Zähler in Kontakt tretten kann und die Daten abfragen könnte. Damit es für jeden einfach zu realisieren ist. 

Jedoch stecken wir hier noch am Anfang.

stefan123t, 2022-11-18:
@DanielR92 der Ansatz mit RS485 ist vermutlich der Erfolgversprechendste/Professionellste! 

Was ist mit den anderen Ansätzen:
* MQTT Topic zB von einem Shelly Device (entweder Steckdoseanadapter an der Waschmaschine oder Shelly 3EM am Hauptanschlusskasten)
* TTL Signal zB von einem Hichi / Volkszähler am vorhandenen SmartMeter
* Anbindung von RS485-fähigen SmartMeter wie bereits in der Original DTU Pro.

DanielR92, 2022-11-20:
Ja das wäre auch alles noch was man einbauen könnte. Geb dir hier recht.

Leviathan09, 2022-11-20:
Also über eine Umsetzung mit dem Shelly 3EM würde ich mich sehr freuen.
Evtl lässt sich das ja auch mit der API von Shelly realisieren, das würde den Schritt über MQTT sparen

BernhardLinz, 2022-11-21:
Also wenn man einen MQTT-Datenpunkt angeben könnte der abonniert wird mit der aktuellen Bezug- oder Einspeiseleistung wäre das doch schon ein Anfang. Und dann wird - im Rahmen von Parametern - versucht den Wert auf 0 zu halten. Ist natürlich komplizierter, je nachdem wie die Werte vorliegen (Ein Wert mit Positiv für Bezug und Negativ für Einspeisung? Getrennte Werte? Nur ein Wert für Einspeisung?)

lumapu, 2022-11-21:
es wird aber beliebig komplex mit mehreren Wechselrichtern. Ich denke man sollte für sowas eher eine zentrale betreiben, aka NodeRed oä.

Ich kann mir schwer vorstellen, dass es eine generische Lösung für mehrere Anwender gibt.

stefan123t, 2022-11-21:
@lumapu naja eine Zentrale ist für viele Nutzer bereits Overkill :) wie gesagt mir würde evtl sogar ein einzelnes Shelly 1PM genügen. Das Shelly schliesse ich zB an meine Waschmaschine (zieht aktuell XXX Watt) an und sobald genug Sonne da ist, soll der WR auch die 600W + XXX aus den PV Modulen lokal einspeisen können. Bei nem Shelly 3EM oder einem anderen per Modbus RS485 auslesbaren SmartMeter ist es idR die saldierte Summe aller drei Phasen die man abboniert.

Klar kann man das dann noch beliebig kompliziert machen und es auch auf alle WR der DTU verteilen, das macht ja die ZeroControl der DTU Pro indem sie jeweils in 30 Watt Häppchen jeden angeschlossenen WR sukszessive der Reihe nach nach oben regelt.
Ggf sogar nach Phase A/B/C geordnet/getrennt. Das wäre dann evtl sogar für die HMS/HMT WR brauchbar.

abcdeef, 2022-11-23:
Hallo,
ich nutze V 5.28 mit einen HM-600.
Ich wollte eine Nulleinspeisung umsetzen. Dazu habe ich einen esp, der den Modbus eines 4Phasenzählers auswertet.

Beim setzen des Topics "inverter/devcontrol/0/11/1" in MQTT reagiert der Hoymiles erst nach ca. 10s.
Ist das normal?
Liegt das am Wechselrichter oder AhoyDTU?
Ist es geplant das schneller zu machen (auf einen Versatz von ca. 1s)?

Leviathan09, 2022-11-23:
> Hallo, ich nutze V 5.28 mit einen HM-600. Ich wollte eine Nulleinspeisung umsetzen. Dazu habe ich einen esp, der den Modbus eines 3Phasenzählers auswertet.
> 
> Beim setzen des Topics "inverter/devcontrol/0/11/1" in MQTT reagiert der Hoymiles erst nach ca. 10s. Ist das normal? Liegt das am Wechselrichter oder AhoyDTU (bzw. esp8266 zu langsam)? Ist es geplant das schneller zu machen (auf einen Versatz von ca. 1s)?

Also bei mir ist das auch so
Teilweise sind es auch mal mehr wie 10s (hier und da mal 30 oder mehr) und dann kommt es immer mal wieder zu phasen wo mir Ahoy meldet "Inverter #0: Balkon_PV (v10010) is not yet available" und dann bleibt er beim zu letzt eingestellten Wert.
Laut Serial Log wird zwar das Limit per MQTT weiter akzeptiert es passiert aber nichts.

Ich vermute das liegt eher am Wechselrichter als an Ahoy oder dem esp
Oder es ist die Kombination aus allem

stefan123t, 2022-11-23:
Also @tbnobody hat auch schon 3-4 Minuten gemessen bevor man das aktuell zuvor im WR gesetzte PowerLimit fehlerfrei wieder auslesen kann.

https://github.com/tbnobody/OpenDTU/issues/35

abcdeef, 2022-11-24:
das Powerlimit ist das einzige, dasz er noch liefert. Andere Werte von CH0,1,2 werden nicht mehr angezeigt bzw. an den MQTT geliefert.

stefan123t, 2022-11-24:
@abcdeef das klingt nach einem Bedien- oder Verständnisfehler. Du hast vermutlich ein Intervall eingestellt bzw. setzt das PowerLimit zu häufig. Die AhoyDTU braucht 2-3 Sekunden um ein Kommando auszuführen bzw. den Wechselrichter nach einem Datensatz zu fragen. Der Wechselrichter selber benötigt auch einige Sekunden um das gewünschte PowerLimit nachzuregeln bzw. anzusteuern. Wenn Du nun sehr / zu häufig das PowerLimit über das MQTT Topic regelst, dann kommt die AhoyDTU nicht mehr dazu die Daten abzufragen. Wenn Du das PowerLimit aus Versehen permanent setzen solltest, kannst Du damit auch das EEPROM im Wechselrichter zu Tode flashen!
Bitte mit den vorgegebenen Einstellungen von 30 Sekunden Abfrage Intervall und alle 5 Minuten das PowerLimit per MQTT setzen, testen und ggf. ein eigenes Issue aufmachen. Danke!

Dieses Issue hier geht um eine Einspeise-Steuerung durch die DTU selbst, d.h. AhoyDTU soll selbständig über MQTT, TTL Signal oder Modbus RS485 ein SmartMeter ablesen und darauf das Limit mit einer einstellbaren, maximalen Einspeisung von 0 / 600 / 800 VA (Türkei, Deutschland / Österreich) selbst regeln.

Screambear, 2023-01-10:
Hallo, ich würde mich über so eine Anbindung auch freuen!

Momentan wäre mein Wunsch auch ein MQTT Topic zu abonnieren und danach die Export Einstellung zu machen. 
Als Einstellung dann das MQTT Topic angeben, die maximal einzuspeisende Leistung und ggf. auch den das Updateintervall.
Eine Aufteilung der einzelnen Inverter je nach Phase wäre auch ein sehr gutes Feature welches ich in Zukunft gebrauchen könnte.

Für die Zukunft oder eben für andere Nutzer wäre ein Anbindung per RS485 interessant. 
Als Einstellung hier müsste doch die Adress- sowie Busdaten reichen. Dazu könnte man dann die Register für die Gesamtleistung oder die der einzelnen Phasen (Inverter eine Phase zuweisen) angeben. So wäre man frei in der Wahl des Smartmeters. Vorausgesetzt die Smartmeter geben die Daten in den Registern alle gleich an oder man kann dieser in der DTU konvertieren.

Nett wäre auch wenn man eine Prioritätenreihenfolge der Inverter angeben könnte. Die aussagt in welcher Reihenfolge die Inverter hoch oder runter gefahren werden sollen. Da müsste man dann aber auch wieder die DC Leistung mit der AC Leistung abgleichen ob überhaupt genug DC Leistung zum weiteren hochregeln vorhanden ist.

Ist das alles überhaupt von der Rechenleistung eines ESP darstellbar? Oder sollte man diese dinge dann tatsächlich lieber in eine Smarthome Zentrale auslagern?

Wenn ja, könnte man ja hier im Github Projekt ggf. für z.B. ioBroker Beispielprogramme (Blockly, Javascript) bereitspellen.

Danke!

Adminius, 2023-01-30:
ich verlinke mal diese Lib:
https://github.com/reaper7/SDM_Energy_Meter

Mit dieser Lib ist sehr einfach SDM* Zähler auszulesen, man braucht nur einen RS485 Adapter und 2 freie Pins am ESP (geht sogar über SoftwareSerial).
SDM72 liefert sogar 2 passende Leistungen:
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L203-L204

lumapu, 2023-01-30:
mit dem Power-Factor glaube ich irrst du dich. Ich denke der gibt an wie du prozentual stehst zu 100% Wirkleistung

Adminius, 2023-01-30:
Was Power Factor ist, weiß ich, wobei richtig verstehen tue ich die Blindleistung immer noch nicht :D 
Das ist aber aus dem DB:
https://stromzähler.eu/media/pdf/93/17/d7/SDM72DM-V2.pdf
[image]
[image]

lumapu, 2023-01-30:
ok, ich dachte der hat da evtl. auch Import und Export Felder.
Mich darfst du wegen Blindleistung nicht fragen 😉

Adminius, 2023-02-06:
also, SDM72D-M-2 liegt auf dem Tisch. um die Einspeisung zu simmulieren habe ich das Netz und Last falschherum angeschlussen. Folgendes kommt aus den Reginstern:

14:11:26.963 -> System Power:     -89.80 W (SDM_TOTAL_SYSTEM_POWER)
14:11:27.487 -> 1ph Power:     -89.74 W (SDM_PHASE_1_POWER)
14:11:28.044 -> 2ph Power:     0.00 W (SDM_PHASE_2_POWER)
14:11:28.587 -> 3ph Power:     0.00 W (SDM_PHASE_3_POWER)

D.h. bei 3 Phasigen Stromzähler reich ein einziger Register aus: SDM_TOTAL_SYSTEM_POWER.
Die Leistung ist negativ wenn eingespeist wird und positiv wenn verbraucht wird.
https://github.com/reaper7/SDM_Energy_Meter/blob/master/SDM.h#L132

Leider ist dieser Register nicht bei 1ph Zählern vorhanden. Da müsse man auf SDM_PHASE_1_POWER gehen.

Es wäre spannend zu wissen was die 1ph Zähler sagen, wenn man SDM_PHASE_2_POWER und SDM_PHASE_3_POWER abfragt? Die gibt es natürlich nicht bei den 1ph Zählern. 
Ich glaube am einfachsten ist eine Umschaltung für 1ph und 3ph einzubauen.

KG3RK3N, 2023-02-09:
Über eine Möglichkeit der direkten Kommunikation zwischen Zähler (in meinem Fall Shelly 3EM mit Tasmota) und ahoy wäre ich auch interessiert da ich eine Nulleinspeisung fahre.
Aktuell mach ich das mit Home Assistant und einer Automatisierung die das Limit dann an ahoy per mqtt sendet. Aber gerade erst gehabt das dann nach einem kurzen Stromausfall das System nicht hoch kam und erst ein weiteres Mal neu gestartet werden musste damit die Automatisierung wieder läuft oder auch die WiFi Verbindung im Zählerschrank nicht immer ideal ist.
Deshalb wäre mir da eine direkte Kommunikation doch etwas lieber. 

Hab ebenfalls schon in Richtung Modbus gedacht auch weil es vermutlich für einen größeren Teil an Nutzern interessant sein könnte.

stefan123t, 2024-10-27:
Siehe ein PR für OpenDTU-OnBattery hier: https://github.com/hoylabs/OpenDTU-OnBattery/pull/1150