lumapu/ahoy #433: Fehler auf der AhoyDTU Webseite

StillerBastler, 2022-11-17:
### Platform

ESP8266

### Antenna

circuit board

### Power Stabilization

nothing

### Connection diagram

nicht verwendet - Fehler auf WebSite

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

Aktuelle Website ahoydtu.de

### Github Hash

n/a

### Build & Flash Method

ESP Tools (flash)

### Desktop

Linux

### Setup

n/a

### Error description

Auf der Seite https://ahoydtu.de/getting_started/ wird als Zieladresse für die Einrichtung des AHOY-DTU http://192.168.4.1/ angegeben.

Korrekt wäre http://192.168.1.1/ (was auch auf der FAQ-Seite angegeben wird.

Die auf .4.1 endende Adresse wird von ESPEasy verwendet, eventuell ist das hier ein Überbleibsel, das den einen oder anderen User verwirrt.

Bei dieser Gelegenheit: Ich danke sehr für die Arbeit, die ihr hier reinsteckt!

reserve85, 2022-11-17:
Das ist schon richtig, die neue Dev-Version ist bereits umgestellt auf 192.168.4.1. 
Für die aktuelle "MAIN" gilt allerdings noch 192.168.1.1

stefan123t, 2022-11-18:
Ja die Umstellung auf 192.168.4.1 in Anlehnung an ESP Easy, Tasmota und OpenDTU hat den Vorteil dass die meisten zu Hause ein 192.168.1.1/24 bzw. das FritzBox Default 192.168.178.1/24 Netzwerk haben.
Hier kann es dann mit unserem alten Default 182.168.1.1 zu Konflikten mit dem Router/DNS Server kommen und der Access Point ist so nur erreichbar wenn man auch das LAN Kabel aussteckt bzw. deaktiviert.
Siehe #420 kann also mE auch hier geschlossen werden

closed 2022-11-21 by lumapu