lumapu/ahoy #434: WLAN Passwort ungültig?

MisterBJ, 2022-11-17:
## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: az-delivery d1 mini
Retailer URL: https://www.az-delivery.de/products/d1-mini

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip
? Habe diesen (auf dem Chip ist ein +):
https://www.amazon.de/DollaTek-NRF24L01-Funk-Transceiver-Modul-antistatischem-kompatibel/dp/B07PQPFTWZ

### Antenna:
* [ ] circuit board
* [x] external antenna

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Connection diagram:
* [ ] Image of the your wiring attached

### Connection diagram I used:
| nRF24L01+ Pin | ESP8266 GPIO   |
| ------------- | -------------- |
| Pin 1 GND [*] | GND            |
| Pin 2 +3.3V   | +3.3V          |
| Pin 3 CE      | GPIO2  CE   D4 |
| Pin 4 CSN     | GPIO15 CS   D8 |
| Pin 5 SCK     | GPIO14 SCLK D5 |
| Pin 6 MOSI    | GPIO13 MOSI D7 |
| Pin 7 MISO    | GPIO12 MISO D6 |
| Pin 8 IRQ     | GPIO0  IRQ  D3 |
| Pin 7 MISO    | GPIO19 MISO D19 |
| Pin 8 IRQ     | GPIO0  IRQ  D0  |

Note: [*] GND Pin 1 has a square mark on the nRF24L01+ module

## Software
* [x] AhoyDTU
* [ ] OpenDTU

### Version / Git SHA: 
Version: 0.5.28
Github Hash: 2e08ee0

### Build & Flash Method:
* [ ] Arduino
* [x] ESP Tools
* [ ] Platform IO

### Desktop OS:
* [ ] Linux
* [ ] Windows
* [x] Mac OS

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)
Ich habe es gegoogelt aber nicht gefunden wie man ein Log erstellt.
Bis auf WLAN Daten und Hostname habe ich nix geändert, wollte die Ahoy-DTU erstmal in meinen WLAN haben.

Hallo,
bin auch endlich dazu gekommen mir eine DTU aufzubauen. Tolle Arbeit von euch!
Mein Problem ist das sich die DTU nicht mit meiner Fritz!Box über das 2,4GHz verbindet.
Da ich schon öfters Probleme mit ESP8266 Geräten hatte gehe ich davon aus das es an mein Passwort liegt.
Es besteht aus:
- 31 Zeichen
- Groß-/Kleinbuchstaben
- Sonderzeichen wie [/,;
- Zahlen
Wlan 2,4GHz mit WPA2 wird verwendet und habe ich im Ahoy-DTU eingestellt.

Vollständigkeitshalber:
- 2,5GHz und 5GHz aktiv 
- WPA2/WPA3 Verschlüsselung aktiv
- SSID aktiviert (Weil sonst mein Saug-Robi das WLAN nicht findet)
- Mac-Adressenfilter aktiv (Mac-Adresse vom ESP8266 aus dem NodeMCU PyFlasher übernommen...kann man die vielleicht in die Weboberfläche vom Ahoy-DTU einfügen, z.B. unter System?)

Vielen Dnk im vorraus,
Björn

DanielR92, 2022-11-17:
Guten Abend Björn,

laut dem Code ist folgendes hinterlegt:

settings.h:Z:32
// wifi
char stationSsid[SSID_LEN]; // SSID_LEN = 32
char stationPwd[PWD_LEN]; // PWD_LEN = 64

Das bedeutet für mich, das es eigentlich klappen sollte. 
Wenn du dein ESP direkt am PC verbunden hast, kommst du mit einem Terminal drauf?
Wenn ja, was spuckt es bei dir alles aus?

MisterBJ, 2022-11-17:
Hallo Daniel,

danke für die schnelle Antwort.
Momenaten steht das Gerät neben mir und ist mit USB Verunden.
Mein Problem ist das ich noch nicht herausgefunden habe wie ich mit einem Terminal verbinde.
habe mit gerade PlatformIO unter VS Codes heruntergeladen und suche dort noch die Optionen.

lumapu, 2022-11-17:
deine Einstellungen sollten aus meiner Sicht kein Problem darstellen. Das Feature MAC Adresse anzeigen nehme ich gerne auf!

MisterBJ, 2022-11-17:
Habe es mit dem Terminal gerade herausgefunden:
`I: connect to network 'BJ24' ...
...........
........................................
........................................
........................................
I: 
   ---------
            AP MODE
                   SSID: AHOY-DTU
                                 PWD: esp_8266
                                              Active for: 60 seconds
                                                                    ---------`

DanielR92, 2022-11-17:
Laut deinen Angaben in deinem ersten Post nutzt du einen Mac... da ich keins Besitze ist es schwierig hier zu Unterstützen. 
Falls du einen Chrome Browser installiert hast, gibt es eine einfache Möglichkeit:

Gehe auf diese Seite: https://ahoydtu.de/getting_started/
Im mittleren Feld "Firmware flashen", klickst du auf Connect. 
Dort solltest du die Möglichkeit haben dein ESP32 auszuwählen. 
Danach kannst du "Logs & Console" auswählen.

MisterBJ, 2022-11-17:
Er versucht sich alle 60 Sekunden mit meiner Fritz-Box zu verbinden aber es gibt keine Fehlermeldung... er macht sofort den AP wieder auf.

Kein SMA auf dem D1 Mini.

Was mir gerade auffällt das die NudeMCU PyFlasher eine andere Mac-Adresse anzeigt als mein Netztwerk-Tool für die 192.168.1.1 wenn die Ahoy-DTU im AP-Mode läuft.
MCU: BC:.....
Tool: BE:.....
Beide Mac-Adressen in Whitelist gesetzt aber keine Besserung

https://ahoydtu.de/getting_started/:
`File Not Found

URL: /getting_started/
Method: GET
Arguments: 0`

Passwort wird aus meiner Passwortverwaltung kopiert daher gehe ich von keinen Tippfehler aus.

Habe eben mal den MAc-Filter deaktiviert und das gleiche. versucht sich zu connecten aber ohne Fehler wird der AP-Mode gestartet.

DanielR92, 2022-11-17:
Hallo Björn,
wegen der Seite sollte es bei dir so aussehen ('`File Not Found' sollte nicht kommen):
[image]

Bitte probiere folgendes aus:
1.) Lass dein ESP am PC verbunden und mittels Terminal auch.
2.) Drücke auf dem ESP die Reset-Taste
3.) Hier müsste relativ früh stehen "connect to network"... 
Bitte dies wie unten zu sehen prüfen.
[image]

MisterBJ, 2022-11-17:
Ups... da war ich wohl mit dem AP anstatt der FB verbunden.

Nach Reset bekomme ich folgendes...aber eigentlich alles wie gehabt:
`b��nn'l�l`�nn$`nr���nrr�`p�nr�bbn�nb��nn'l`�nn$`nr���nrl`r��nrl`��$bl`�n�`I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
I: resetPayload: id: 0
I: connect to network 'BJ24' ...
...........
........................................
........................................
........................................
I: 

   ----------------------------------------
I: Welcome to AHOY!
I: 
   point your browser to http://192.168.1.1
I: to configure your device
I: ----------------------------------------

I: RF24 Amp Pwr: RF24_PA_I: LOW
I: Radio Config:
SPI Frequency           = 1 Mhz
                               Channel                  = 3 (~ 2403 MHz)
Model                   = nRF24L01+
RF Data Rate            = 250 KBPS
RF Power Amplifier      = PA_LOW
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 250 microseconds
Auto Retry Attempts     = 0 maximum
Packets lost on
                   current channel      = 0
Retry attempts made for
                           last transmission    = 0
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Enabled
Auto Acknowledgment     = Disabled
Primary Mode            = RX
TX address              = 0xe7e7e7e7e7
pipe 0 (closed) bound   = 0xe7e7e7e7e7
pipe 1 ( open ) bound   = 0x1936638101
pipe 2 (closed) bound   = 0xc3
pipe 3 (closed) bound   = 0xc4
pipe 4 (closed) bound   = 0xc5
pipe 5 (closed) bound   = 0xc6
I: connect to network 'BJ24' ...
...........
........................................
........................................
........................................
I: 
   ---------
            AP MODE
                   SSID: AHOY-DTU
                                 PWD: esp_8266
                                              Active for: 60 seconds
                                                                    ---------

W: Time not set or it is night time, therefore no communication to the inverter!
I: AP will be closed in 49 seconds
I: AP will be closed in 39 seconds`

lumapu, 2022-11-17:
habe gerade die MAC hinzugefügt, schaue doch in unserem `development03` Branch bzw. unter Github Actions vorbei. Ohne Anmeldung auch unter https://ahoydtu.de verfügbar.

lumapu, 2022-11-17:
deine letzte Antwort bestätigt die erste Annahme: das Passwort ist falsch, warum auch immer

stefan123t, 2022-11-18:
Also das sieht verdächtig danach aus dass er mit dem WLAN keine Verbindung aufbauen kann.
Dann macht er halt den Access Point wieder auf. Dirt kannst Du Dich ja verbinden und das Setup ggf vervollständigen, speichern und neu booten. Nur dann werden die Einstellugen auchvaktiv geladen.

Obwohl ich ja auch zur Vorsicht neige, verstehe ich den Aufwand für das komplizierte WLAN Passwort ehrlich gesagt nicht. Bei mir geht das mach dem K.I.S.S. 💋 Prinzip :)
Nur Ziffern, Buchstaben und einige sorgfältig ausgewählte Sonderzeichen. Aber von dem allem natürlich eine ganze Menge (32), Länge läuft halt.

Das mit den MAC Adressfiltern ist schön für die Sicherheit im Netz, würde ich aber erst aktivieren nachdem es läuft. Woher weisst Du dass der ESP die MAC verwendet die er im ESP/ PyFlasher anzeigt ?

Man kann die beim ESP32 mit ESP.getEfuseMac() bzw beim ESP8266  ESP.getChipId() auslesen.
Vielleicht kann man die auch auf irgendeinen anderen Wert setzen ?

MisterBJ, 2022-11-18:
> enthält es zufällig \0? Das würde einiges erklären
Nein, tut es nicht.

> schaue doch in unserem development03 Branch...
In der Version [0305c11](https://github.com/lumapu/ahoy/commit/0305c114cfa83eb03c75d2913868e8beb78df090) steht zwar in der Konsole das der AP-Mode nach der Installation/Neustart gestartet ist aber der Webserver (192.168.1.1) startet wohl nicht. Connecten zum AP funktioniert nicht.
Hatte zur Vorsicht mehre Installationen gemacht NodeMCU PyFlasher und Online-EPSTool.
Ich dachte ich bin zu Blöd und da habe ich nochmal die 0.5.28 installiert und der Webserver war wieder da. Da dachte ich mir "ach komm" und habe nochmal meine Router-Daten eingegeben, neugestartet, auf einmal macht er eine Verbindung und ist im Heimischen Netz...versteh das noch einer.

> verstehe ich den Aufwand für das komplizierte WLAN Passwort ehrlich gesagt nicht.
Heute weis ich das auch...es kommt auf die Länge an ;)
Ich habe über 30 WLAN-Geräte Shelly, Sonoff (Tasmota), usw. Teilweise unter Putz verbaut wie die shelly i3, ich mag es nicht darauf ankommen lassen das die nicht mehr erreichbar sind und ich alles wieder zerrupfen muss ;)

> Das mit den MAC Adressfiltern ist schön für die Sicherheit im Netz, würde ich aber erst aktivieren nachdem es läuft. Woher weisst Du dass der ESP die MAC verwendet die er im ESP/ PyFlasher anzeigt ?
- Bin davon ausgegangen das die Tools den Chip irgendwie abfragen können.
   In diesen Fall war die MAC richtig und komisch ist das im Netzwerktool nachdem ich neu gescannt habe seit die DTU im Netz ist auch die korrekte MAC anzeigt.
- Mit den Filtern gebe ich dir Recht aber wenn ich in der FB sehe das es acht bis zehn  unbekannte Zugriffversuche gibt versuche ich mich erstmal auf die Tools zu verlassen ;)

closed 2022-11-18 by lumapu